jgyates/genmon #938: SNMP

nickabocker171, 2023-08-23:
Did snmp change in V1.18.18? its enabled in the interface but not responding to any requests

jgyates, 2023-08-23:
Yes. The way the SNMP schema is built was changed to allow for more controller types. the change log is here: https://github.com/jgyates/genmon/blob/master/changelog.md

Also, if you are having issues you can submit your logs or look at /var/log/gensnmp.log for any errors.

It is working for me, but your settings my be different.

nickabocker171, 2023-08-24:
log is filled with this any thoughts

`2023-08-23 12:27:54,824 : Error in SnmpCallbackFunction: <lambda>() takes 3 positional arguments but 4 were given : gensnmp.py:559
2023-08-23 12:27:54,826 : Error in SnmpCallbackFunction: <lambda>() takes 3 positional arguments but 4 were given : gensnmp.py:559
2023-08-23 12:27:54,827 : Error in SnmpCallbackFunction: <lambda>() takes 3 positional arguments but 4 were given : gensnmp.py:559
2023-08-23 12:27:54,828 : Error in SnmpCallbackFunction: <lambda>() takes 3 positional arguments but 4 were given : gensnmp.py:559
2023-08-23 12:27:54,833 : Error in SnmpCallbackFunction: <lambda>() takes 3 positional arguments but 4 were given : gensnmp.py:559
2023-08-23 12:27:54,834 : Error in SnmpCallbackFunction: <lambda>() takes 3 positional arguments but 4 were given : gensnmp.py:559
2023-08-23 12:27:54,835 : Error in SnmpCallbackFunction: <lambda>() takes 3 positional arguments but 4 were given : gensnmp.py:559
2023-08-23 12:27:54,836 : Error in SnmpCallbackFunction: <lambda>() takes 3 positional arguments but 4 were given : gensnmp.py:559
2023-08-23 12:27:54,837 : Error in SnmpCallbackFunction: <lambda>() takes 3 positional arguments but 4 were given : gensnmp.py:559
2023-08-23 12:27:54,837 : Error in SnmpCallbackFunction: <lambda>() takes 3 positional arguments but 4 were given : gensnmp.py:559
2023-08-23 12:27:54,839 : Error in SnmpCallbackFunction: <lambda>() takes 3 positional arguments but 4 were given : gensnmp.py:559
2023-08-23 12:27:54,840 : Error in SnmpCallbackFunction: <lambda>() takes 3 positional arguments but 4 were given`

jgyates, 2023-08-24:
Unfortunately no. That error is taking place inside the python snmp library. Here are some things to consider:

1) Are you using python 2.x or 3.x. Version 3.5 or higher is required 
2) Is your software up to date? This includes the OS software as well as the genmon libraries.

jgyates, 2023-08-24:
You can use this URL: 

  http://192.168.11.10:8000/cmd/support_data_json

to see what version of python genmon is using. Replace the IP address with the IP of your pi.

To update the Raspberry Pi operating system, type sudo apt-get update followed by sudo apt-get upgrade and then restart.

To update the libaries genmon uses, log in via ssh then type:

     cd genmon
     ./genmonmaint.sh -r

The -r option will update the installed libaries to the latest version.

jgyates, 2023-08-24:
I found this:

https://github.com/etingof/pysnmp/issues/440

Looks like some others are having a similar problem with the pysnmp library (this is the library genmon is using as well). The thread above offers some solutions, one is to update the pyasn1 library. The abvoe command (genmonmaint.sh -r) will do that.

jgyates, 2023-08-24:
i am going to close this issue as it looks like this is a solution. Let me know how it goes.

closed 2023-08-24 by jgyates

nickabocker171, 2023-09-06:
Just wanted to let you know changing the version didnt help im still digging into why

jgyates, 2023-09-06:
did you update the python libaries with this command: 

     cd genmon
    ./genmonmaint.sh -r

If you did then what is the output of this command:

    pip3 freeze

jgyates, 2023-09-06:
Looks like there is a pull request open for a fix for this in the issue with pyasn1 here: https://github.com/pyasn1/pyasn1/pull/39

You could uninstall pyasn1 and reinstall version .48:

     sudo pip install --force-reinstall -v "pyasn1==0.4.8"

nickabocker171, 2023-09-07:
> did you update the python libaries with this command:
> 
> ```
>  cd genmon
> ./genmonmaint.sh -r
> ```
> 
> If you did then what is the output of this command:
> 
> ```
> pip3 freeze
> ```

aiohttp==3.8.5
aiohttp-retry==2.8.3
aiosignal==1.3.1
arandr==0.1.10
astroid==2.5.1
asttokens==2.0.4
async-timeout==4.0.2
attrs==23.1.0
automationhat==0.2.0
beautifulsoup4==4.9.3
bleson==0.1.8
blinker==1.6.2
blinkt==0.1.2
buttonshim==0.0.2
Cap1xxx==0.1.3
certifi==2020.6.20
chardet==4.0.0
charset-normalizer==3.2.0
chump==1.6.0
click==8.1.7
colorama==0.4.4
colorzero==1.1
configparser==6.0.0
crcmod==1.7
cryptography==3.3.2
cupshelpers==1.0
dbus-python==1.2.16
distro==1.5.0
docutils==0.16
drumhat==0.1.0
envirophat==1.0.0
ExplorerHAT==0.4.2
flask==2.3.3
fluids==1.0.24
fourletterphat==0.1.0
frozenlist==1.4.0
geojson==2.5.0
gpiozero==1.6.2
html5lib==1.1
idna==2.10
importlib-metadata==6.8.0
isort==5.6.4
itsdangerous==2.1.2
jedi==0.18.0
Jinja2==3.1.2
lazy-object-proxy==0.0.0
ldap3==2.9.1
logilab-common==1.8.1
lxml==4.6.3
MarkupSafe==2.1.3
mccabe==0.6.1
microdotphat==0.2.1
mopeka-pro-check==0.1.0
mote==0.0.4
motephat==0.0.3
multidict==6.0.4
mypy==0.812
mypy-extensions==0.4.3
numpy==1.19.5
oauthlib==3.1.0
paho-mqtt==1.6.1
pantilthat==0.0.7
parso==0.8.1
pexpect==4.8.0
pgzero==1.2
phatbeat==0.1.1
pianohat==0.1.0
picamera==1.13
picamera2==0.3.12
pidng==4.0.9
piexif==1.1.3
piglow==1.2.5
pigpio==1.78
Pillow==8.1.2
ply==3.11
psutil==5.9.5
pyasn1==0.4.8
pycairo==1.16.2
pycryptodomex==3.18.0
pycups==2.0.1
pygame==1.9.6
Pygments==2.7.1
PyGObject==3.38.0
pyinotify==0.9.6
PyJWT==2.8.0
pylint==2.7.2
PyOpenGL==3.1.5
pyOpenSSL==20.0.1
pyotp==2.9.0
pyowm==2.10.0
PyQt5==5.15.2
PyQt5-sip==12.8.1
pyserial==3.5
pysmbc==1.0.23
pysmi==0.3.4
pysnmp==4.4.12
python-apt==2.2.1
python-prctl==1.7
pytz==2023.3
rainbowhat==0.1.0
reportlab==3.5.59
requests==2.25.1
requests-oauthlib==1.0.0
responses==0.12.1
roman==2.0.0
RPi.GPIO==0.7.0
RTIMULib==7.2.1
SciPy==1.8.1
scrollphat==0.0.7
scrollphathd==1.2.1
Send2Trash==1.6.0b1
sense-hat==2.4.0
simplejpeg==1.6.4
simplejson==3.17.2
six==1.16.0
skywriter==0.0.7
smbus==1.1.post2
sn3218==1.2.7
soupsieve==2.2.1
spidev==3.5
ssh-import-id==5.10
thonny==4.0.1
toml==0.10.1
touchphat==0.0.1
twilio==8.7.0
twython==3.8.2
typed-ast==1.4.2
typing-extensions==3.7.4.3
unicornhathd==0.0.4
urllib3==1.26.5
v4l2-python3==0.3.2
voipms==0.2.5
webencodings==0.5.1
werkzeug==2.3.7
wrapt==1.12.1
yarl==1.9.2
zipp==3.16.2

jgyates, 2023-09-08:
Here is the output from my system as a point of reference:

bleson==0.1.8
certifi==2020.6.20
cffi==1.15.0
chardet==4.0.0
chump==1.6.0
click==8.0.3
colorzero==1.1
configparser==5.2.0
crcmod==1.7
cryptography==36.0.1
distro==1.5.0
Flask==2.0.3
fluids==1.0.21
geojson==2.5.0
gpg===1.14.0-unknown
gpiozero==1.6.2
idna==2.10
itsdangerous==2.0.1
Jinja2==3.0.3
ldap3==2.9.1
MarkupSafe==2.0.1
mopeka-pro-check==0.1.0
numpy==1.23.2
paho-mqtt==1.6.1
ply==3.11
psutil==5.9.0
pyasn1==0.4.8
pycparser==2.21
pycryptodomex==3.14.1
PyJWT==2.3.0
pyOpenSSL==22.0.0
pyotp==2.6.0
pyowm==2.10.0
pyserial==3.5
pysmi==0.3.4
pysnmp==4.4.12
python-apt==2.2.1
pytz==2021.3
requests==2.25.1
RPi.GPIO==0.7.0
SciPy==1.8.1
six==1.16.0
smbus==1.1.post2
spidev==3.5
ssh-import-id==5.10
toml==0.10.1
twilio==7.6.0
urllib3==1.26.5
voipms==0.2.5
Werkzeug==2.0.3

jgyates, 2023-09-08:
Here is the output of another test system that also works, hopefully between the three you can zone in on what is different.

bleson==0.1.8
certifi==2020.6.20
cffi==1.15.1
chardet==4.0.0
chump==1.6.0
click==8.1.3
colorzero==1.1
configparser==5.2.0
crcmod==1.7
cryptography==37.0.4
distro==1.5.0
Flask==2.2.2
fluids==1.0.21
geojson==2.5.0
gpiozero==1.6.2
idna==2.10
importlib-metadata==4.12.0
itsdangerous==2.1.2
Jinja2==3.1.2
ldap3==2.9.1
MarkupSafe==2.1.1
mopeka-pro-check==0.1.0
numpy==1.23.1
paho-mqtt==1.6.1
ply==3.11
psutil==5.9.1
pyasn1==0.4.8
pycparser==2.21
pycryptodomex==3.15.0
PyJWT==2.4.0
pyOpenSSL==22.0.0
pyotp==2.6.0
pyowm==2.10.0
pyserial==3.5
pysmi==0.3.4
pysnmp==4.4.12
python-apt==2.2.1
pytz==2022.2.1
requests==2.25.1
RPi.GPIO==0.7.0
SciPy==1.8.1
six==1.16.0
smbus==1.1.post2
spidev==3.5
ssh-import-id==5.10
toml==0.10.1
twilio==7.12.1
urllib3==1.26.5
voipms==0.2.5
Werkzeug==2.2.2
zipp==3.8.1

jishnurams, 2023-09-26:
I am getting this error from the logs:-

Error in SnmpCallbackFunction: <lambda>() takes 3 positional arguments but 4 were given : gensnmp.py:577

Have tried updating libs using : -
cd genmon
./genmonmaint.sh -r

No response from SNMP agent, any suggestions?

jishnurams, 2023-09-26:
> I am getting this error from the logs:-
> 
> Error in SnmpCallbackFunction: () takes 3 positional arguments but 4 were given : gensnmp.py:577
> 
> Have tried updating libs using : - cd genmon ./genmonmaint.sh -r
> 
> No response from SNMP agent, any suggestions?

Output of pip3 freeze:-
aiohttp==3.8.5
aiohttp-retry==2.8.3
aiosignal==1.3.1
arandr==0.1.10
astroid==2.5.1
asttokens==2.0.4
async-timeout==4.0.3
attrs==23.1.0
automationhat==0.2.0
beautifulsoup4==4.9.3
bleson==0.1.8
blinker==1.6.2
blinkt==0.1.2
buttonshim==0.0.2
Cap1xxx==0.1.3
certifi==2020.6.20
chardet==4.0.0
charset-normalizer==3.2.0
chump==1.6.0
click==8.1.7
colorama==0.4.4
colorzero==1.1
configparser==6.0.0
crcmod==1.7
cryptography==3.3.2
cupshelpers==1.0
dbus-python==1.2.16
distro==1.5.0
docutils==0.16
drumhat==0.1.0
envirophat==1.0.0
ExplorerHAT==0.4.2
flask==2.3.3
fluids==1.0.25
fourletterphat==0.1.0
frozenlist==1.4.0
geojson==2.5.0
gpiozero==1.6.2
html5lib==1.1
idna==2.10
importlib-metadata==6.8.0
isort==5.6.4
itsdangerous==2.1.2
jedi==0.18.0
Jinja2==3.1.2
lazy-object-proxy==0.0.0
ldap3==2.9.1
logilab-common==1.8.1
lxml==4.6.3
MarkupSafe==2.1.3
mccabe==0.6.1
microdotphat==0.2.1
mopeka-pro-check==0.1.0
mote==0.0.4
motephat==0.0.3
multidict==6.0.4
mypy==0.812
mypy-extensions==0.4.3
numpy==1.19.5
oauthlib==3.1.0
paho-mqtt==1.6.1
pantilthat==0.0.7
parso==0.8.1
pexpect==4.8.0
pgzero==1.2
phatbeat==0.1.1
pianohat==0.1.0
picamera==1.13
picamera2==0.3.9
pidng==4.0.9
piexif==1.1.3
piglow==1.2.5
pigpio==1.78
Pillow==8.1.2
ply==3.11
psutil==5.9.5
pyasn1==0.4.8
pycairo==1.16.2
pycryptodomex==3.18.0
pycups==2.0.1
pygame==1.9.6
Pygments==2.7.1
PyGObject==3.38.0
pyinotify==0.9.6
PyJWT==2.8.0
pylint==2.7.2
PyOpenGL==3.1.5
pyOpenSSL==20.0.1
pyotp==2.9.0
pyowm==2.10.0
PyQt5==5.15.2
PyQt5-sip==12.8.1
pyserial==3.5
pysmbc==1.0.23
pysmi==0.3.4
pysnmp==4.4.12
python-apt==2.2.1
python-prctl==1.7
pytz==2023.3.post1
rainbowhat==0.1.0
reportlab==3.5.59
requests==2.25.1
requests-oauthlib==1.0.0
responses==0.12.1
roman==2.0.0
RPi.GPIO==0.7.0
RTIMULib==7.2.1
SciPy==1.8.1
scrollphat==0.0.7
scrollphathd==1.2.1
Send2Trash==1.6.0b1
sense-hat==2.4.0
simplejpeg==1.6.4
simplejson==3.17.2
six==1.16.0
skywriter==0.0.7
smbus==1.1.post2
sn3218==1.2.7
soupsieve==2.2.1
spidev==3.5
ssh-import-id==5.10
thonny==4.0.1
toml==0.10.1
touchphat==0.0.1
twilio==8.9.0
twython==3.8.2
typed-ast==1.4.2
typing-extensions==3.7.4.3
unicornhathd==0.0.4
urllib3==1.26.5
v4l2-python3==0.3.2
voipms==0.2.5
webencodings==0.5.1
werkzeug==2.3.7
wrapt==1.12.1
yarl==1.9.2
zipp==3.17.0

nickabocker171, 2023-09-26:
I was able to resolve it by 

running the following
./genmonmaint.sh -i (did not reinstall config files when asked)

then
 sudo pip install --force-reinstall -v "pyasn1==0.4.8"

after rebooting i was finally able to get snmp to work, unfourtintly the -r didnt work for me it had to be -i and i had to run it before doing the pyasn1 reinstall

jgyates, 2023-09-26:
@nickabocker171, thanks for the info.

@jishnurams, let us know if that fixes your issue.

jishnurams, 2023-09-27:
> I was able to resolve it by
>
> running the following ./genmonmaint.sh -i (did not reinstall config files when asked)
>
> then sudo pip install --force-reinstall -v "pyasn1==0.4.8"
>
> after rebooting i was [...]

@jgyates @nickabocker171 Thank you Guys, i tried with -i and then reinstalled pyasn1-0.4.08. Just like magic i am getting SNMP datas.

mstackler, 2023-10-25:
I am still having an issue with SNMP.  I have tried to reinstall several times on fresh SD cards.  Tried 32bit and 64bit as well.  I have followed the suggestions above and reinstalled genmon with -r and -i.  I installed pyasn1 4.8 but needed to use --break-system-packages to make it install.  I thought i had it working a couple times.  I would do the snmpwalk and would get genmon data on oid 58399 then 5 minutes later it would not work. Now I can't get it to work at all.  I am running the latest install of Pi OS and latest download of genmon.

mark

jgyates, 2023-10-25:
I have a pi setup with the latest version of Raspberry pi OS 64 bit with a fresh install of genmon. I am unable to duplicate your problem.  

If you have outbound email setup with genmon then you can submit your log file on the About page and I can look at them to see if there are any issues.

On the pi I run this command

     sudo apt-get install snmp
     snmpwalk -v1 -c public localhost .

The first command will install the snmp utilities. the second will show all snmp values on the local system.

When you say it does not work, what exactly are you trying and what is output? Have you validated with the the snmpget or snmpwalk utility? Here are some exampels of using these utilities:  

https://github.com/jgyates/genmon/wiki/1----Software-Overview#examples-using-snmpget-snmpwalk-and-snmptranslate

mstackler, 2023-10-25:
Here is what i get.  Just the snmp data from the Pi itself

admin@GenMonTest:~ $ sudo apt-get install snmp
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
snmp is already the newest version (5.9.3+dfsg-2).
0 upgraded, 0 newly installed, 0 to remove and 13 not upgraded.
admin@GenMonTest:~ $ snmpwalk -v1 -c public localhost .
iso.3.6.1.2.1.1.1.0 = STRING: "Linux GenMonTest 6.1.0-rpi4-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.54-1+rpt2 (2023-10-05) aarch64"
iso.3.6.1.2.1.1.2.0 = OID: iso.3.6.1.4.1.8072.3.2.10
iso.3.6.1.2.1.1.3.0 = Timeticks: (7183259) 19:57:12.59
iso.3.6.1.2.1.1.4.0 = STRING: "Me <me@example.org>"
iso.3.6.1.2.1.1.5.0 = STRING: "GenMonTest"
iso.3.6.1.2.1.1.6.0 = STRING: "Sitting on the Dock of the Bay"
iso.3.6.1.2.1.1.7.0 = INTEGER: 72
iso.3.6.1.2.1.1.8.0 = Timeticks: (5) 0:00:00.05
iso.3.6.1.2.1.1.9.1.2.1 = OID: iso.3.6.1.6.3.10.3.1.1
iso.3.6.1.2.1.1.9.1.2.2 = OID: iso.3.6.1.6.3.11.3.1.1
iso.3.6.1.2.1.1.9.1.2.3 = OID: iso.3.6.1.6.3.15.2.1.1
iso.3.6.1.2.1.1.9.1.2.4 = OID: iso.3.6.1.6.3.1
iso.3.6.1.2.1.1.9.1.2.5 = OID: iso.3.6.1.6.3.16.2.2.1
iso.3.6.1.2.1.1.9.1.2.6 = OID: iso.3.6.1.2.1.49
iso.3.6.1.2.1.1.9.1.2.7 = OID: iso.3.6.1.2.1.50
iso.3.6.1.2.1.1.9.1.2.8 = OID: iso.3.6.1.2.1.4
iso.3.6.1.2.1.1.9.1.2.9 = OID: iso.3.6.1.6.3.13.3.1.3
iso.3.6.1.2.1.1.9.1.2.10 = OID: iso.3.6.1.2.1.92
iso.3.6.1.2.1.1.9.1.3.1 = STRING: "The SNMP Management Architecture MIB."
iso.3.6.1.2.1.1.9.1.3.2 = STRING: "The MIB for Message Processing and Dispatching."
iso.3.6.1.2.1.1.9.1.3.3 = STRING: "The management information definitions for the SNMP User-based Security Model."
iso.3.6.1.2.1.1.9.1.3.4 = STRING: "The MIB module for SNMPv2 entities"
iso.3.6.1.2.1.1.9.1.3.5 = STRING: "View-based Access Control Model for SNMP."
iso.3.6.1.2.1.1.9.1.3.6 = STRING: "The MIB module for managing TCP implementations"
iso.3.6.1.2.1.1.9.1.3.7 = STRING: "The MIB module for managing UDP implementations"
iso.3.6.1.2.1.1.9.1.3.8 = STRING: "The MIB module for managing IP and ICMP implementations"
iso.3.6.1.2.1.1.9.1.3.9 = STRING: "The MIB modules for managing SNMP Notification, plus filtering."
iso.3.6.1.2.1.1.9.1.3.10 = STRING: "The MIB module for logging SNMP Notifications."
iso.3.6.1.2.1.1.9.1.4.1 = Timeticks: (4) 0:00:00.04
iso.3.6.1.2.1.1.9.1.4.2 = Timeticks: (4) 0:00:00.04
iso.3.6.1.2.1.1.9.1.4.3 = Timeticks: (4) 0:00:00.04
iso.3.6.1.2.1.1.9.1.4.4 = Timeticks: (4) 0:00:00.04
iso.3.6.1.2.1.1.9.1.4.5 = Timeticks: (4) 0:00:00.04
iso.3.6.1.2.1.1.9.1.4.6 = Timeticks: (5) 0:00:00.05
iso.3.6.1.2.1.1.9.1.4.7 = Timeticks: (5) 0:00:00.05
iso.3.6.1.2.1.1.9.1.4.8 = Timeticks: (5) 0:00:00.05
iso.3.6.1.2.1.1.9.1.4.9 = Timeticks: (5) 0:00:00.05
iso.3.6.1.2.1.1.9.1.4.10 = Timeticks: (5) 0:00:00.05
iso.3.6.1.2.1.25.1.1.0 = Timeticks: (7186020) 19:57:40.20
iso.3.6.1.2.1.25.1.2.0 = Hex-STRING: 07 E7 0A 19 0A 1B 2B 00 2D 04 00 
iso.3.6.1.2.1.25.1.3.0 = INTEGER: 393216
iso.3.6.1.2.1.25.1.4.0 = STRING: "coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_b"
iso.3.6.1.2.1.25.1.5.0 = Gauge32: 3
iso.3.6.1.2.1.25.1.6.0 = Gauge32: 188
iso.3.6.1.2.1.25.1.7.0 = INTEGER: 0
End of MIB
admin@GenMonTest:~ $

jgyates, 2023-10-25:
It looks like you have installed some other software to get SNMP data from the pi. What software was installed? The pi does not support SNMP in it's default config so it is possible that there is a conflict between the add on and the software you have installed. The add on uses the [pysnmp module](https://github.com/etingof/pysnmp) to

Here is my output:
 snmpwalk -v1 -c public localhost .
iso.3.6.1.2.1.1.1.0 = STRING: "Genmon Generator Monitor"
iso.3.6.1.2.1.1.2.0 = OID: iso.3.6.1.4.1.58399
iso.3.6.1.2.1.1.3.0 = Timeticks: (31157) 0:05:11.57
iso.3.6.1.4.1.58399.0.0.1 = STRING: "OK"
iso.3.6.1.4.1.58399.0.0.2 = STRING: "Generator Monitor running for 0:05:14."
iso.3.6.1.4.1.58399.0.0.3 = STRING: "0.00 MB of 15.00 MB"
iso.3.6.1.4.1.58399.0.0.4 = STRING: "V1.18.18"
iso.3.6.1.4.1.58399.0.1.1 = STRING: "M: 27329, S: 27322"
iso.3.6.1.4.1.58399.0.1.2 = STRING: "0 "
iso.3.6.1.4.1.58399.0.1.3 = STRING: "0.00%"
iso.3.6.1.4.1.58399.0.1.4 = STRING: "6"
iso.3.6.1.4.1.58399.0.1.5 = STRING: "0.02%"
iso.3.6.1.4.1.58399.0.1.6 = STRING: "0"
iso.3.6.1.4.1.58399.0.1.7 = STRING: "0"
iso.3.6.1.4.1.58399.0.1.8 = STRING: "0"
iso.3.6.1.4.1.58399.0.1.9 = STRING: "0"
iso.3.6.1.4.1.58399.0.1.10 = STRING: "4"
iso.3.6.1.4.1.58399.0.1.11 = STRING: "173.76"
iso.3.6.1.4.1.58399.0.1.12 = STRING: "0.0102 sec"
iso.3.6.1.4.1.58399.0.2.1 = STRING: "116.06 F"
iso.3.6.1.4.1.58399.0.2.2 = STRING: "Raspberry Pi 4 Model B Rev 1.4"
iso.3.6.1.4.1.58399.0.2.3 = STRING: "OK"
iso.3.6.1.4.1.58399.0.2.4 = STRING: "OK"
iso.3.6.1.4.1.58399.0.2.5 = STRING: "OK"
iso.3.6.1.4.1.58399.0.2.6 = STRING: "0.07%"
iso.3.6.1.4.1.58399.0.2.7 = STRING: "Debian GNU/Linux"
iso.3.6.1.4.1.58399.0.2.8 = STRING: "12 (bookworm)"
iso.3.6.1.4.1.58399.0.2.9 = STRING: "5 days, 12:15:52"
iso.3.6.1.4.1.58399.0.2.10 = STRING: "eth0"
iso.3.6.1.4.1.58399.1.0.0.1 = STRING: "Auto"
iso.3.6.1.4.1.58399.1.0.0.2 = STRING: "Off - Ready"
iso.3.6.1.4.1.58399.1.0.0.3 = STRING: "Battery Charger On"
iso.3.6.1.4.1.58399.1.0.0.4 = STRING: "Oil below 8 psi"
iso.3.6.1.4.1.58399.1.0.0.5 = STRING: "13.60 V"
iso.3.6.1.4.1.58399.1.0.0.6 = STRING: "Charging"
iso.3.6.1.4.1.58399.1.0.0.7 = STRING: "0 "
iso.3.6.1.4.1.58399.1.0.0.8 = STRING: "0.00 Hz"
iso.3.6.1.4.1.58399.1.0.0.9 = STRING: "0 V"
iso.3.6.1.4.1.58399.1.0.0.10 = STRING: "0.00 A"
iso.3.6.1.4.1.58399.1.0.0.11 = STRING: "0.00 kW"
iso.3.6.1.4.1.58399.1.0.0.12 = STRING: "0 "
iso.3.6.1.4.1.58399.1.0.0.13 = ""
iso.3.6.1.4.1.58399.1.0.1.1 = STRING: "246 V"
iso.3.6.1.4.1.58399.1.0.1.2 = STRING: "246 V"
iso.3.6.1.4.1.58399.1.0.1.3 = STRING: "246 V"
iso.3.6.1.4.1.58399.1.0.1.4 = STRING: "144 V"
iso.3.6.1.4.1.58399.1.0.1.5 = STRING: "192 V"
iso.3.6.1.4.1.58399.1.0.1.6 = STRING: "240 V"
iso.3.6.1.4.1.58399.1.0.2.1 = STRING: "07/17/23 08:53:30 Charger Warning : Alarm Code: 2770"
iso.3.6.1.4.1.58399.1.0.2.2 = STRING: "07/17/23 08:48:05 Service Schedule A "
iso.3.6.1.4.1.58399.1.0.2.3 = STRING: "07/17/23 09:22:35 Stopped - Auto "
iso.3.6.1.4.1.58399.1.0.3.1 = STRING: "Wednesday October 25, 2023 10:24:42"
iso.3.6.1.4.1.58399.1.0.3.2 = STRING: "Monday July 17, 2023 09:58"
iso.3.6.1.4.1.58399.1.1.0.1 = STRING: "Generic Air Cooled"
iso.3.6.1.4.1.58399.1.1.0.2 = STRING: "0009397581"
iso.3.6.1.4.1.58399.1.1.0.3 = STRING: "Evolution, Liquid Cooled"
iso.3.6.1.4.1.58399.1.1.0.4 = STRING: "3600"
iso.3.6.1.4.1.58399.1.1.0.5 = STRING: "20 kW"
iso.3.6.1.4.1.58399.1.1.0.6 = STRING: "60 Hz"
iso.3.6.1.4.1.58399.1.1.0.7 = STRING: "Diesel"
iso.3.6.1.4.1.58399.1.1.0.8 = STRING: "93 %"
iso.3.6.1.4.1.58399.1.1.0.9 = STRING: " "
iso.3.6.1.4.1.58399.1.1.0.10 = STRING: "3.4 L"
iso.3.6.1.4.1.58399.1.1.0.11 = STRING: " "
iso.3.6.1.4.1.58399.1.1.0.12 = STRING: "0.00 kWh"
iso.3.6.1.4.1.58399.1.1.0.13 = STRING: "0.00 gal"
iso.3.6.1.4.1.58399.1.1.0.14 = STRING: "0.00 gal"
iso.3.6.1.4.1.58399.1.1.0.15 = STRING: "0.00 h"
iso.3.6.1.4.1.58399.1.1.0.16 = STRING: " "
iso.3.6.1.4.1.58399.1.1.0.17 = STRING: " "
iso.3.6.1.4.1.58399.1.1.0.18 = STRING: " "
iso.3.6.1.4.1.58399.1.1.0.19 = STRING: " "
iso.3.6.1.4.1.58399.1.1.0.20 = STRING: "0.00 h"
iso.3.6.1.4.1.58399.1.1.1.1 = STRING: " "
iso.3.6.1.4.1.58399.1.1.1.2 = STRING: " "
iso.3.6.1.4.1.58399.1.1.1.3 = STRING: "1024 "
iso.3.6.1.4.1.58399.1.1.1.4 = STRING: "240 V"
iso.3.6.1.4.1.58399.1.1.1.5 = STRING: "48 kW"
iso.3.6.1.4.1.58399.1.1.1.6 = STRING: "4 "
iso.3.6.1.4.1.58399.1.1.1.7 = STRING: "24 "
iso.3.6.1.4.1.58399.1.1.1.8 = STRING: "1 "
iso.3.6.1.4.1.58399.1.1.1.9 = STRING: "1 h"
iso.3.6.1.4.1.58399.1.1.1.10 = STRING: "10 "
iso.3.6.1.4.1.58399.1.1.1.11 = STRING: "0 "
iso.3.6.1.4.1.58399.1.1.1.12 = STRING: "60 Hz"
iso.3.6.1.4.1.58399.1.1.1.13 = STRING: "240 V"
iso.3.6.1.4.1.58399.1.1.2.1 = STRING: "Weekly Wednesday 14:00 Quiet Mode Off"
iso.3.6.1.4.1.58399.1.1.2.2 = STRING: "12 min"
iso.3.6.1.4.1.58399.1.1.3.1 = STRING: " or "
iso.3.6.1.4.1.58399.1.1.3.2 = STRING: " or "
iso.3.6.1.4.1.58399.1.1.3.3 = STRING: " or "
iso.3.6.1.4.1.58399.1.1.3.4 = STRING: " or "
iso.3.6.1.4.1.58399.1.1.3.5 = STRING: "124 hrs or 07/16/2024 "
iso.3.6.1.4.1.58399.1.1.3.6 = STRING: "249 hrs or 07/16/2025 "
iso.3.6.1.4.1.58399.1.1.3.7 = STRING: " "
iso.3.6.1.4.1.58399.1.1.3.8 = STRING: "0.03 h"
iso.3.6.1.4.1.58399.1.1.3.9 = STRING: "V1.05"
iso.3.6.1.4.1.58399.1.1.3.10 = STRING: "V1.52"
End of MIB

Note that there is no additional info from the pi, only genmon data.

if you are using the latest genmon you should not need to force a reinstall of pyasn1 to v 0.4.8 since this is the version specified. 

Also, if you are using the latest version of Raspberry pi OS then there are significant changes may to the python subsystem so just running "sudo pip install --force-reinstall -v "pyasn1==0.4.8"" will not be sufficient. For more into, see [this thread](https://github.com/jgyates/genmon/issues/956). 

Make sure you are running the latest version of the software.

it would be usful to see the output of /var/log/gensnmp.log at a minimum. The other genmon log files may help if you can send your logs via the about page.

jgyates, 2023-10-25:
are you running snmpd?

mstackler, 2023-10-25:
when i installed snmp i used     sudo apt-get install snmpd snmp.  could that be the issue

jgyates, 2023-10-25:
unless you have some need for the pi related snmp data I would uninstall snmpd. I don't have a data point on using that with pysnmp. You can use pysnmp a few different ways and my guess is the way that I am using it (as a snmp base transport) is not compatible with snmpd as they are both trying to manage the socket used for snmp traffic.

jgyates, 2023-10-25:
do you get the same results after a restart of the pi?

it would be useful to see the output of /var/log/gensnmp.log at a minimum. The other genmon log files may help if you can send your logs via the about page. Without the logs I don't have much to go regarding why this is happening. The logs may provide more data.

Is there any other software installed on the system other than genmon?

mstackler, 2023-10-25:
it is now working and has been for 15min.  I got excited when it worked and ran an SNMP walk with my DCIM tool and think it breaks it.  I rebooted the pi an all is good.  Going to test it on site at generator tomorrow.  Thank you for all your help

mark

jgyates, 2023-10-25:
Great! I am glad it is working for you.

rsockanc, 2024-02-28:
I had the same issue; "sudo apt-get purge --auto-remove snmpd" by a reboot solved my issue after following all the other steps listed above such as  sudo pip install --force-reinstall -v "pyasn1==0.4.8"