jgyates/genmon #939: Mopeka Add on fails to connect to 2nd sensor

ojra89, 2023-08-24:
### Expected Behavior

Hi @jgyates! Thanks for Genmon! It's amazing and super useful!

I bought a prebuilt pintsize.me unit and it has been working great for everything except the Mopeka Add-on. I have a 2-100lb (vertical) tank setup and am using 2 Mopeka Pro sensors. They are both configured for 100lbs in the Mopeka app and update properly there. 

### Actual Behavior

In Genmon, however only 1 of the 2 sensors connects to the PI. 1 gauge shows the correct fuel level, the 2nd shows at 0%

I also get an email from Genmon "Unable to communicate with Mopeka Pro sensor". I have reset my bluetooth serial with the seiralconfig.py and have looked through the closed issues here but didn't find anything similar. In the genmon settings I have the 'Fuel Tank Size' set to around 44 GAL (roughly around the equivalent to the 2 100lb vertical tanks).

I have looked through all logs and see a couple of these errors:

Error in GetExternalFuelPercentage: 'Percentage' : controller.py:2555

 sudo bluetoothctl displays correct output as well.

I have tried switching the MAC addresses, added spaces, and even bought a 3rd Mopeka Sensor and the issue persists.
Example in add on (macs randomized )

`D6:41:E1:6B:58:41,D8:63:F2:12:85:19`

### Steps to Reproduce (including precondition)

Issue persists no matter the settings

### Screenshot or Pictures relating to the problem (if possible)
[image]

### Your Environment

- Generator Model: Generac 22kw Guardian Air cooled
- Genmon Logs: Submitted
- Genmon Version: Version V1.18.18

jgyates, 2023-08-24:
Hi @ojra89 

I received your logs.

one thing we can try to get more info is to enable the debug flag for the genmopeka add on. To do this log into the pi via ssh, then edit the file /etc/genmon/genmopeka.conf. Added the line:

debug=True

a the end of the file, save the file an restart genmon. Let it run for about 2 min then send your logs again. This will put additional data in file /var/log/genmopeka.log

When we are finished debugging I would recommend setting the value to debug=False in the /etc/genmon/genmopeka.conf file and restarting genmon. Enabling this for debugging is fine however you don't want to enable this long term as it put more wear on your SD card.

Let me know if you have any questions.

ojra89, 2023-08-24:
Thanks @jgyates just resubmitted logs with Debug on.

jgyates, 2023-08-24:
There are two issues. 

1) the pi is not detecting mopeka traffic from the first sensor 
2) the add on had some logic that would send confusing data to genmon if the first sensor failed, but the 2nd had a valid reading. I corrected this problem so the error in /var/log/genmon.log should go away after you update from the about page.

Regarding number 1:

1) are the two tanks about the same distance to the pi?
2) are the two sensors the same type?
3) you may want to double check the MAC address of the first sensor, that could be an easy fix, if you have not alreay.
4) try swapping the position of the MAC addresses in the add on settings, but leave the sensors where they are. Then we can get a data point regarding if the error follows the order of the MAC address or if it follows the physical sensor.

ojra89, 2023-08-24:
For point 2:

I did the upgrade from the about page

Regarding number 1:

1. are the two tanks about the same distance to the pi? Yes, they're about 15 ft away.
2. are the two sensors the same type? Yes, both are Mopeka Pro
3. you may want to double check the MAC address of the first sensor, that could be an easy fix, if you have not already. **I did do this and verified they were correct**
4. try swapping the position of the MAC addresses in the add on settings, but leave the sensors where they are. Then we can get a data point regarding if the error follows the order of the MAC address or if it follows the physical sensor. **I swapped the sensors and then both gauges were showing 0% in GenMon, I swapped back to original setting and its back at 74.13%**
5. Resending logs

ojra89, 2023-08-24:
I tried leaving a single mac address in the add-on and couldn't get any of the 2 to connect/display values. Seems like there may be an issue there.

jgyates, 2023-08-24:
your logs indicate that you can get a signal from either sensor, but not reliably. Are your sensors mounted to the bottom of the tank, and if so do you use the spacers that come with the sensor. These help the signal escape by having a gap of air below the metal on vertical tanks. The other thing to try is get a data point the range you can get with the mopeka mobile app.

This is the library that is used with this add on: https://github.com/spbrogan/mopeka_pro_check

The library controls all of the functionality of communicating with the sensors so there is not a lot I can change regarding the handling of the sensors. At the moment, with the data I have it looks like an issue with signal strenght and the pi. I will look into this a bit more tomorrow to see if there is a way to show the signal strength of each sensor as read from the pi. that is likely the next data point we need to confirm or rule out the signal strength idea.

jgyates, 2023-08-27:
I am going to close this issue however if you have any additional questions or status feel free to post to this thread.

closed 2023-08-27 by jgyates