jgyates/genmon #940: Generac wifi not working with molex bypass

notms, 2023-08-27:
Hello Genmon community. I have recently got the genmon softwarw working with RS-232 and also used a molex with a bypass for generac wifi. The weird part is I can’t get wifi for generac working when using molex, direct it works fine. I followed molex connection directions for revolution 2 with wifi. Any help would be appreciated if you have seen it before.

### Expected Behavior

Generac wifi working

### Actual Behavior

Generac wifi not working

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

[image]
[image]

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here}

skipfire, 2023-08-27:
Can you post a picture of, or link to, the bypass cable you mention?

notms, 2023-08-28:
Unfortunately I am away from the generator, will be back in a couple of days to post, I followed the molex cable making diagram from the wiki page.

liltux, 2023-08-28:
According to the display it is not connected to the WiFi network.  One thing to check is if there is a green led blinking on the back side of the Mobile Link Antenna inside the generator.  This would indicate connection status and also if the power supply to the module is good.  Also, how strong is your WiFi signal form your house to the generator?  The Mobile Link antenna requires a better signal than most other devices.

notms, 2023-09-03:
Here are the pics. I am thinking logically and would think that some level of power would
Need to be provided to the wifi on generac. The wifi has four connections I am only connected to two of them, I am assuming other is ground and power. I think I need power. When connect with the original connector to wifi I see a green light come on the wifi module, with my bypass I don’t[image]
[image]
[image]
[image]
[image]

skipfire, 2023-09-03:
You do need all 4 wires, the other two are power and ground.  The MobileLink WiFi also uses RS485, so if you followed GenMon wiring instructions you are probably connecting the wrong wires as GenMon uses the RS232 wires.  Under the RS232 wires are 12v and ground, under the RS485 wires are the 5v and ground and my MobileLink uses the 5v, ground, and the 2 RS485 pins matching the pinout on your MobileLink.  Can you get a picture of the entire bypass cable so we can see how the splices are setup?

jgyates, 2023-09-04:
I am going to close this issue for now but feel free to post to this thread if you have any follow up questions or status.

closed 2023-09-04 by jgyates