jgyates/genmon #941: Best Approach For New Installation

ssmoss, 2023-08-28:
I wanted to start by saying thank you for making this software. It really should replace the MobileLink offering as it is a thousand times better.

I am going to be getting a 22kW Air Cool Generac unit and I am trying to figure out the best way to get genmon up and running. My understanding is that my unit is going to have the wireless module installed which will block the connector port needed by genmon.

After looking over the wiki it seems that the awesome folk(s) at PintSize.Me created a solution to the problem which is this [spliter](https://pintsize.me/store/ols/products/generator-splitterinjector/v/GSI-SPL) and I really like the improvements in V1.1 though I do have a question about the disclaimer:

```
The splitter and RJ45 injector/ejector we currently are shipping (v1.1) physically divides the data pins so that one side is RS485 and the other is RS232 to ensure that only 1 RS232 device is connected; however, we do not know what the behavior of combo MobileLink interfaces that try to use both would be when connected to the RS485 only side.
```

I am assuming the V1.1 has been tested and works so is this saying something could change in the future which could pose an issue? I believe after 2018 of the Guardian Series air-cooled come with it pre-installed.

It seemed that powering the Pi from the controller connector via the GPIO connector is a very clean solution in conjunction with using one of [HATs  / pHATs](https://pintsize.me/store/ols/products/generator-hats-and-phat) from PintSize.Me until I saw this disclaimer:

```
NOTE: Some people have reported undervoltage errors on the Pi 3 (not the Pi Zero) when powering through the controller connector. While this has not been reported to prevent the application from running it may not be an ideal way to power the Pi. See [this](https://github.com/jgyates/genmon/issues/106) issue for more details.
```

With this I assume I could just a Pi Zero as I suspect that is more than sufficient until I saw this disclaimer:

```
Pi Zero, Zero W, 1, and 2 are not recommended.  Please use a Pi Zero 2, 3A+, 3B, 3B+, or 4B.
```

So at this point I was a little at a loss how to continue with the plan to use GPIO to power the Pi until I stumbled upon this on the  [Making A Cable](https://github.com/jgyates/genmon/wiki/3.1--Making-a-Cable) page. It seems that Evolution 2.0 (which I believe my generator will have) now has a 12v connection along with the 5v from Evolution 1. I downloaded the [instructions](https://github.com/skipfire/genmon-addon/raw/main/Genmon%20HAT%202.0%20Instructions.pdf) for the HATs  / pHATs I see the following:

```
Onboard power converter to drop the 12v available at the generator to the 5v needed to power the Pi and HAT. There is also a rectifier after the barrel plug to prevent reversed connections.
```

Based on this it seems like one can use the power from the generator's controller to power Pi + HAT and using Pi 3 / 4 should work without issue?

Will placing the Pi inside an enclosure inside the generator's housing cause an issue with wireless single? Is it recommended to get an external antenna or if the access point is within 20 feet it shouldn't be an issue?

Is a fully turn key solution available for purchase that just requires plugging it in?

skipfire, 2023-08-28:
I'm the guy behind PintSize.Me, the splitters and injectors are tested.  I have a Generac with MobileLink as well.  It only works with the WiFi module though, the cellular module uses RS232 and you can only have 1 RS232 device at a time which would be GenMon.  As far as I know, no accessories except the combo MobileLinks (Ether + WiFi, WiFI + Cell) use both sets of data pins, so there is no real chance of future revisioning of software from MobileLink breaking things, my WiFi MobileLink connecter only has 4 pins wired up to the 8-pin connector (from the factory that way).

Yes the Pi can be powered through the HAT/pHAT though there is a switch on it to disable that option, in which case the HAT can be powered from the Pi (if you are powering by USB or PoE).  As for lightning, it will do what it wants and the amount of power in a strike will not care about a small electronics fuse; it will even skip through a household circuit breaker if that's what it feels like doing.

The Pi Zero 2 works fine, the earlier Pi Zeros can work, however the performance is significantly lower and with some of the more recent updates with Python and such there are sometimes problems getting some of the packages to build on it correctly.  Same with the Pi 1 and Pi 2.

For powering the Pi & HAT, the PintSize.Me equipment comes with a power cable (battery to barrel) and that is recommended for all installs as there is inconsistency with how much power the Evolution 2 controller will provide before the internal sensors get thrown out of whack causing the generator to think there is a utility problem when there isn't.  In that case the generator will read something like 80 to 100 total volts instead of 240 and the generator will start.  The Pi 4 ALWAYS consumes too much power for the Evolution 2 controller, other Pi models are hit and miss, hence the HATs/pHATs coming with the battery to barrel cable to power it directly instead of through the Evolution 2 controller.

The PintSize.Me pre-loaded units are essentially turn-key.  You configure WiFi in the Raspberry Pi OS, connect the data cable, optionally a splitter/injector, and the power cable and you are good to go.

ssmoss, 2023-08-28:
Ah, I thought the OpenGenSet was simply a serial to wireless bridge / convertor and you could point genmon to the ip address and port it was transmitting on. I currently have a NUC running Ubuntu that I use for home automation so I was thinking I could use that to run genmon in addition to a few other services. I had (incorrectly it seems) assumed that Home Assistant + ESPHome was simply required to flash the device. Based on the configuration yaml file it seemed like a type of Arduino / Feather embedded controller. I also really liked the antenna that designed as well.

For the battery to barrel connection does one need to worry about condensation? I saw note about watching out for that when using the antenna for the OpenGenSet.

Since I do not have an existing Pi it sounds like getting the [Pre-loaded v2.3 Pi 3A+](https://pintsize.me/store/ols/products/generator-hats-and-phat/v/PGI-RPL) along with the [splitter](https://pintsize.me/store/ols/products/generator-splitterinjector/v/GSI-SPL) would give me all the parts I need save for an enclosure.

I have a few questions about your Pi 3A+ setup:
1. Do you have an issue with wireless reception? 
2. Do you have your splitter in the enclosure that contains your Pi?
3. Is the splitter require to be in an enclosure or is it weatherproof enough given the existing connection for the wireless unit doesn't enclose the connector?
4. What type of enclosure do you use?
5. How did you attach it to generator?

skipfire, 2023-08-28:
From a hardware standpoint the way you describe OpenGenSet is correct, however it needs the software installed and that requires ESPHome, and without HomeAssistant that appears to be command line.

I have not had any problems with condensation on any of the parts and I am in Louisiana where it is typically high humidity.  I did make sure that the cabling I have that goes from outside to inside has a good drip point so that water cannot follow cables down to circuits.

1. For wireless reception, OpenGenSet uses an ESP which is very low power which does reduce the signal.  Then the metal enclosure of the generator can reduce it even further.  I was able to have my Raspberry Pi work inside the generator, though I did move it outside but that was due to how frequently I test new products and revisions to existing products.
2. I have my splitter inside the generator enclosure.
3. I do not have an additional enclosure for the splitter.
4. I have a plastic outdoor electrical junction box and that provided a massive improvement in the signal strength.  There are several different units that people have posted pictures of on here, it all depends on what look you like, what size you want, etc.  
5. For attaching it, I just used 3M double sided sticky foam tabs to adhere it to the generator so I have no permanent modifications, not even screw holes.
6. The enclosure I used has rubber grommets on all sides, I sliced through one on the bottom side to push the Molex and power cable through, it is not water tight but is a ground facing grommet and unless I have something that would shoot water up into it I don't need to really worry about water intrusion, if you have a sprinkler system near it then this may not be suitable but would likely work otherwise.

ssmoss, 2023-08-29:
So noted on the requirement to use the command line (thanks for clarifying) though that shouldn't be an issue for me as my software skills are better than my hardware ones :)

Flashing the OpenGenSet aside in terms of reliability, performance, and security could one be considered better than the other?

My knee jerk reaction is to use the OpenGenSet over the Pi since it is lower power and in theory can be powered via the molex connector which is one less connection. That said I am concerned about the low power scenario which I assume is bad in that not only does the device not work but it causes the generator to turn on?

skipfire, 2023-08-29:
Security is probably lower on the OpenGenSet, the serial server software (written by someone else) does not have any security on it.  The OpenGenSet is a much newer product with only a few months of history on the oldest install.  The HATs on the other hand are now just over 2 years old and while I've had 6 units fail within 2 weeks, I've only heard about 1 that failed after that point so the reliability of the HATs is quite high and the data just isn't there for OpenGenSet as it's only a few months old.

The low power scenario is a small concern, I was able to replicate it on my generator using a Raspberry Pi without a battery cable and it did not cause any damage, it just temporarily makes the sensors return the wrong voltage value triggering the generator to activate and switch over.  It is not an intermittent issue though, it happens immediately upon plugin or it doesn't happen at all, I have yet to hear of an instance where it would go back and forth.  It also has only happened on 2 of the OpenGenSet units.

Overall I would recommend the Pi path.

ssmoss, 2023-08-29:
Okay I think you have convinced me to get the PreLoaded v2.3 Pi 3A+.

Since my unit has the MobileLink I am going to need the splitter as well. It looks like the Pi + Hat comes a 1 meter (three feet) data cable (8pin molex cable?) and a power cable (something like [this](https://www.amazon.com/gp/product/B01DYE54LI/ref=oh_aui_detailpage_o00_s00) or is something more simple; does have / require a fuse)?

I do I need to be concerned about EMI and the data cables?

What kind of Micro SD card is used? How big is it?

If magnetic mounts are used on the outer housing do I need to be concerned about any type of interference. I think I would prefer to do that over the tap incase I want to remove it or just move it.

@jgyates From a ware perspective how much writing it done to the SD card, is write life something that one should be concerned about or is the amount of data written over 10+ so tiny isn't a factor?

skipfire, 2023-08-29:
The power cable has loops on one end for tying to the tightening posts on the battery clamps (not the battery terminals themselves), a fuse on the positive wire close to the battery, and a 5.5mm barrel on the other end for plugging into the HAT.  The splitter comes with a 12 inch Molex cable.

I've not seen or heard of any problems with EMI on the data cables, there is a little additional interference for WiFi but I have not found it to be significant, nor have I seen reports of people losing their connectivity every time the generator runs due to EMI issues.

Current SD card is a SanDisk Ultra class 10 32GB card.

Magnets do not tend to do anything to WiFi as they are constant and not variable, EMI is typically only an issue with non-static fields (random or cyclical).  Check the surface of the generator you want to attach to and ensure that it will have a solid magnetic grab.  You will need to buy neodymium magnets if you want to do that to ensure that the strength is solid.  Some Home Depots carry MagCraft magnets, and you can order them online.  They have a lot of difference sizes, so once you have picked your enclosure, measure the spots that you could put magnets on to find the best size to use.l  

The HAT does have holes that line up with the Pi holes, but I've just left mine loose in my enclosure.  My enclosure is just big enough, so there isn't any room for it to really move around anyway.  Others have used something like a project perf board (not sure the proper name for it).  There are a lot of pictures posted by other users of how they have done theirs and even a few youtube videos you can watch to get ideas from others.

I don't recommend using a Pi 4 for this most of the time because it generates a lot more heat and pretty much requires a fan, but you don't have much in the way of ventilation options in an enclosure on the outside of the generator.  I do not use a fan and after hurricane Ida my 3A+ was inside the generator and did fine here in southeastern Louisiana where it was 106 a couple days ago, so I get some pretty good heat to heat test it.  If you happen to be someplace cold, I have sold to Alaskan and Canadian users and none of them have reported any problems from the cold.

For data writing, I can't give you exact numbers, but it is an SD card and with any of my Pis (I have about a dozen, some with no application based writes) I would not expect the SD card to last more than 2 years.  I make a backup of my GenMon configuration after I know changes are good so if I do have to back up and restore I can do so fairly easily.  I also provide a download for the image for customers should they need to redo their SD card later.

jgyates, 2023-08-29:
Hi @ssmoss 

Minimal SD writes are made to the SD card with genmon, however I recommend using an RAM disk for the /var/log directory. This wiki entry describes this in more detail: https://github.com/jgyates/genmon/wiki/Appendix-M-Tips-and-Tricks-useful-for-Genmon#extending-the-life-of-your-sd-card

ssmoss, 2023-08-29:
Okay, that makes total sense for the logs. I did something similar for a pfsense install running on a SSD.

ssmoss, 2023-08-29:
You have sold me on the Pi 3A+ + HAT with no fan.  so be on the look out for an order :-)

So my plan place the enclosure inside the generator using this [enclosure](https://www.amazon.com/gp/product/B097XQVMY6) since it has the rubber grommets all around enclosure to allow for each cable management. My plan is to attach a few of these [rectangular magnets](https://www.amazon.com/LOVIMAG-Magnets-Neodymium-Waterproof-Adhesive/dp/B0CBTSFFSB) to the back of the enclosure. The enclosure is 7.9” × 6.1” × 3.1” so my plan is to put both the Pi + HAT and Splitter in it. For the splitter I think I am going to use some double sticky tape to stick it to the enclosure as it it seems like the housing would permit this. As for the Pi + HAT I am not sure the best way to do the same. I was looking and didn't see some type of flat case that I could screw the Pi to and then use the same double sticky tape to stick it to the enclosure. I then planned to use 2x of these [data cable](https://www.amazon.com/gp/product/B000M802RG) and past them though the bottom grommets and then use some hot glue to seal it around the grommet. I would then attach the existing MoblieLink cable to the corresponding cable and then using the data cable with the kit to attach the other cable to the generator. The final part would be passing the power cable though a third grommet and sealing it with hot glue.

skipfire, 2023-08-29:
When ordering, the controller version is just data for me, it helps when someone asks for any sort of support that I can go back and look at it.  The one exception to this is the Evo1 version of OpenGenSet, it is cheaper as it doesn't have the 12v power support, it's identical otherwise.  A new air-cooled Generac should be an Evolution 2 controller (unless they suddenly sneak out an Evo3).

I use a different enclosure in the same line, I think mine was the 3.9"x3.9", but it's a little tight and if I were to buy again I would buy a larger one.

For the magnets the page says they have >30lb pull strength, so I would suggest applying them 1 at a time to see how they work.  I use Gorilla Glue to attach my magnets to things. 

I would probably keep the splitter in the generator just to avoid running the MobileLink wire through the grommets as well, the Splitter has no electronics, just plugs and wiring between the plugs so there isn't really anything to overheat and nothing you would likely care about accessing regularly.  Though there is no problem with putting it in the box as you plan, it's just more effort.

The acrylic base you list at the end is almost identical to what will be on the Pre-loaded Pi when you receive it, and it has 4 small square neodymium magnets glued to it.  You can use foam to attach the acrylic to the case and I could include an extra 4 magnets if you would want to use those in the case to mount it.  I would not recommend a tape style mounting as you may want/need to remove the Pi and having that acrylic plate taped in place would make that difficult.

ssmoss, 2023-08-29:
Take my $$$! Just placed my order.

Thanks for the Gorilla Glue suggestion, that sounds like a better option compared to the double sticky tape.

Maybe I can just put a magnet on the back of the splitter instead of putting it in the case.

Ah so attach the four magnets to the back of the case and just stick the Pi to the case that way, that sounds like a great idea.

skipfire, 2023-08-29:
Unfortunately I only have 4 of the magnets left (which I'm dropping in your order), I'm not expecting my next order of them to come in until next Monday.  That said, yes I think you could do the same with the splitter, but my splitter is just hanging off the 12" cable from the bottom of the Evolution 2.

ssmoss, 2023-08-29:
Ah, no worries and thanks for the rapid turn around. I once I get all my parts I will follow up some pictures.

closed 2023-08-29 by jgyates

ssmoss, 2023-08-29:
Sounds good. I was planning on documenting my journey. Happy to share it here and maybe if you think it’s helpful / useful it could be turned into a wiki page?

jgyates, 2023-08-29:
Absolutely. Always looking for content to make the project better. Thanks!

ssmoss, 2023-09-01:
@skipfire I got the order today. Insanely fast turn around! I wanted to say everything looks extremely polished and topnotch.

I see the unit is running Raspberry PI OS, is there any advantage to switching to Ubuntu Server (or other supported headless os) to reduce resources? Once setup I suspect I would very rarely ever need to access the GUI as I would be checking the WebUI.

ssmoss, 2023-09-01:
@jgyates I see there is included exporter for MQTT, is there one a database by any chance? I have a central PostgreSQL instance that I use to collect data from various sensor I have and use that to feed some Grafana Dashboards.

skipfire, 2023-09-01:
Raspberry Pi OS does a lot more defaulting of permissions required for the GPIO and such.  If you want to save resources the easiest would be to switch the config to default to headless, but when there is no monitor, Raspberry Pi OS doesn't load the display anyway.  Even so, if you want resource utilization it comes no where near taxing the Pi.

jgyates, 2023-09-01:
@ssmoss 

Not currently. No one has asked for any database export so far (your are the first that I can recall). There are some methods for connecting MQTT mosquitto to PostgreSQL that I have heard of but I personally have never used them. I have seen some tie MongoDB or InfluxDB to MQTT.

jgyates, 2023-09-01:
Also, if you use the "Lite" version or Rasp OS, the GUI is not even on the filesystem.

ssmoss, 2023-09-01:
That is good to know about Ubuntu. I did find this [article](https://ubuntu.com/tutorials/gpio-on-raspberry-pi#1-overview) talking about the permission issue you mentioned so I think while I am waiting for the generator I might try installing Ubuntu server on a spare SD card to see what happens.

So noted about the database export. I suppose the advantage of going though MQTT is that I get a buffer should there be an issue with the wireless link. I am capturing data from some wireless temperature sensors via rtl-433 so I wrote a little program in NodeJS to read from MQTT and write to the database. I will see if I can find the tool you mentioned or worse can I think I can just repurpose the one I wrote.

I see the MQTT configuration [file](https://github.com/jgyates/genmon/blob/master/conf/genmqtt.conf) and that you limit export of some fields. While looking though the web interface I didn't see a list of the all fields that could be exported (maybe because I do not have it connected to the generator yet). Is there a list or a piece of code I could look at to better understand the fields that be exported?

jgyates, 2023-09-01:
If the blacklist parameter is blank, then it will export all data (not gauges) on the status, maintenance, outage and monitor page. You have the ability to blacklist items by search terms (if the blacklist term is in the topic path, then it is not sent via MQTT).

https://github.com/jgyates/genmon/blob/2ce573aa27c1b0b8c615612e6e2462751b96da15/addon/genmqtt.py#L215

The blacklist allows you to not send things like the time, serial packet count or other things that you may not be interested in seeing via MQTT.

Let me know if you have any other questions.

ssmoss, 2023-09-02:
I have been trying to get a sense for the space available for an enclosure that can fit inside the generator. My plan is to use this [enclosure](https://www.amazon.com/gp/product/B097XQVMY6?th=1) (7.9”×6.1”×3.1” / 200×155×80mm) but after looking at a couple of YouTube videos I am little concerned this will be too big.

@skipfire I remember you had mentioned using an enclosure similar to the one I was looking at:
```
I use a different enclosure in the same line, I think mine was the 3.9"x3.9", but it's a little tight and if I were to buy again I would buy a larger one.
```

Do you think that one would be to large? Should I try for a slightly smaller one (5.9”×5.9”×2.8” / 150×150×70mm)?

skipfire, 2023-09-02:
If this will be on the outside of the generator enclosure I don't think the 7.9x6.1 size will look huge and having extra space gives you options in the future, you could always tape that size on the generator enclosure so you can see what the size would look like.  The cost on them is close enough that that isn't even a consideration.  Really it is going to come down to your preferences and guesswork for what the future may hold.  3.9x3.9 does fit a Pi 3A+ with the HAT and the cable going in, but has very little extra space.  I cannot add something if I wanted to do so, and an additional 2 inches is not very much if you want to put something else in the box, but is great if you just want some comfort for tidy wiring.

ssmoss, 2023-09-03:
I was thinking inside if possible. That’s why I am a bit more concerned about size. The smaller size that’s fits the Pi I would think would have no issues but I want some room for growth for all the reasons you outlined.

skipfire, 2023-09-03:
I didn't even bother with a plastic enclosure when I had my install inside the generator enclosure, I only used it when I put it outside for easier access and WiFi strength, but it does make sense to use a smaller box if inside since you can only really stick it on the inside of the front wall since that is above the battery and you need to keep easy access to the battery.

ssmoss, 2023-09-03:
Oh interesting, I was concerned about getting a literal hardware bug leaving the Pi exposed like that. Same with some concern about water on the exposed pins. Maybe I am overthinking it?

skipfire, 2023-09-03:
My enclosure was designed fairly well to keep water out if I don't open the lid, even with it open there isn't a lot of direct path for water to come in, and if it were to flood the Pi would be the least of my concerns.  As for physical bugs, circuits aren't as fragile as they used to be and the voltage is low enough that it isn't likely a bug will conduct much power.  The pins themselves are occupied by the hat so it is just the tips of the pins where soldered that are exposed rather than the tall pins.

ssmoss, 2023-09-05:
@skipfire I noticed the fuse in the power cable that came with the kit is 10A which at 12V would be 120W? Is that high for the Pi? Thanks.

skipfire, 2023-09-05:
Those are off the shelf components, so limited choices in what I get. The fuse is to protect the generator not the Pi.

ssmoss, 2023-09-05:
Does your HAT work the [Potato](https://www.amazon.com/Libre-Computer-Potato-Single-Heatsink/dp/B0BQG668P6)? I have been trying to some software updates / installations and it seems like the amount of free RAM is causing some issues. If I stop `genmon` that allows it to work but ideally I would like be to able to install / update without having to stop it. Given that Pi 3B+ are so hard to get I was looking at alternatives. My understanding is that the Potato is GPIO identical to the Pi.

skipfire, 2023-09-05:
The HAT is just hardware and while I have not tried the Potato, all the others I have tried worked fine.  They all seem to support a serial port on those pins, but it may have a different number.  Other GPIO uses also may have different numbers for the same pins, and then of course you run into the differences in OS since Raspberry Pi OS will not run on a potato to the best of my knowledge.  I have not had any trouble getting the Pi 3A+ to do any updates.  What errors were you getting?

jgyates, 2023-09-05:
Genmon does not use any libraries that are pi specific but some of the add ons do use the python library [RPi.GPIO](https://pypi.org/project/RPi.GPIO/).  If you want to use the potato with the add on programs that use GPIO then you will need to modify the add on code to use [the potato GPIO library](https://pypi.org/project/LePotatoPi.GPIO).

ssmoss, 2023-09-05:
Trying to do an ‘aptitude update && aptitude upgrade’ It would trigger the OOM killer. Occasionally it would install but get stuck on post processing actions it seemed and then get killed.

Ah so it’s not as simple as just popping the HAT off.

skipfire, 2023-09-05:
It will not be as simple as swapping the HAT and SD, pretty sure it's a ground up rebuild.  You might try using apt instead of aptitude, it might be aptitude interface that is pushing it over.  You could also split the two commands, run `sudo apt update` first then run `sudo apt upgrade` after that.

jgyates, 2023-09-05:
form the software standpoint it is about like using Ubuntu, it will work, you just have extra steps to setup and when you want to use GPIO or even change the function of GPIO, like when you enable the serial port functionality on the expansion header. The genmon project has scripts that will setup the serial port, but they assume you are running the Raspberry pi OS. The base functionality of genmon just uses the unix file system, TCP/IP sockets and unix based serial ports. Some add ons use pi specific things like gpio, other add ons do no.

This has info on apt for updating system software: https://github.com/jgyates/genmon/wiki/Appendix-M-Tips-and-Tricks-useful-for-Genmon#updating-your-pi-system-software

ssmoss, 2023-09-05:
I am guessing that increasing the swap isn’t great either due to the SD not being a great random IO source?

I will try again with apt instead. Good thought the wrapper being the cause.

ssmoss, 2023-09-05:
Since I would be using the power cable using a Pi4 wouldn’t be a problem and and since it’s a Pi moving the HAT plus SD should work?

ssmoss, 2023-09-08:
@skipfire are there any plans to integrate the splitter into the HAT. I was doing some test fitting of the parts in the enclosure and it would be really spiffy if the HAT had two of the 8 pin connectors, one for the generator and the second for the optional mobile link. I am not sure if there is enough space to put them side by side or one on top of each other.

On a sidetone, I really wish they had chosen a more compact connection like ethernet so it is easier to deal with compared to the huge 9 pin connectors.

skipfire, 2023-09-08:
The first HATs did have that, but it was not widely used and after adding the support for a fan and making the 12v power design safter there just wasn't room to keep it.

ssmoss, 2023-09-08:
Ah, I see. That's too bad. Given the choice I wouldn't mind the HAT being slightly larger than the PI if it incorporated the splitter. The 8 pin cables are a royal PITA to make tight corners with.

skipfire, 2023-09-08:
I would have to see some pictures of the problem you are having.  I have my splitter under the controller so there is no space constraint.  And yes, the 8-pin Molex is a bit of a pain, and the cables are much more expensive.

I have a couple of prototype HATs that work but never listed for sale that use an RJ45 instead, however they require an adapter that is almost identical to the splitter, but it swaps one of the Molex plugs for an RJ45 plug.  It does make cable running to the HAT have a smaller plug, but it also risks confusion with Ethernet and should that happen things would probably fry (not sure which end or if it would be both).  I have a splitter in pace for testing but I keep my MobileLink disconnected from WiFi so that Generac can't push a firmware update that may break the ability to communicate with genmon, which they tried once before (not targeting genmon specifically).  If I wasn't selling these things I would not use the splitter in my own install.

ssmoss, 2023-09-09:
Here is picture of my proposed box:
[image]

The Pi and splitter aren't secured yet but the idea is to have the Pi on the top and the splitter on the bottom. I am going to use a [cable gland](https://www.amazon.com/dp/B094ZFP22C) on the bottom of the box and route the power and two data cables out of it. I am going to glue some strong magnets to the back of it as you suggested. My thinking with this layout is that if I need / want to put it outside of the generator housing it should be pretty easy to move and just require some extension cables.

During the test fitting I had some observations / ideas / thoughts:

1.  The cable between the splitter and HAT is really stiff so working in the confined space is a little tricky. One suggestion might be to use a [fan cable](https://www.startech.com/en-us/cables/fan4ext12) or something similar as the cable should much more flexible. One downside I see to this is it makes the HAT a little less portable.
2. The layout of the ports on the splitter seem to be setup for a horizontal orientation. What I mean by this is the generator comes in on the left of the splitter. Right top goes to the Pi and right bottom goes to the ML. In a vertical orientation I think having the generator in on bottom left and ML out on bottom right would make more sense. On the top would be the link to the Pi. This would remove the the need for the cable to make 180 as shown in my picture.
3. When you mentioned RJ45 style plug it got me thinking, would it be possible to make 8 pin to RJ45 adapter part / board / piece? One could be attached to the generator port and another to the ML cable. The splitter would then have 3 RJ45 ports doing what it does currently now. Finally the HAT would have RJ45 instead of the 8 pin. RJ45 cable is pretty flexible, cheap, available in many pre-cut lengths, easy to crimp if you want to make your own, and come in an outdoor variant. It doesn't solve the issue you mentioned above that although these use the RJ45 connector they are not "ethernet" spec and so plugging it into a router, PoE injector, or other things of that nature would be bad.

Do you have more information on the Firmware incident? Did it break `genmon`? Does not having ML enabled cause any issues with extended warranty or the installer if you used one not being able to monitor the generator?

jgyates, 2023-09-09:
See number 11 and 15 of the known issues for a summary of the firmware incident.

https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues

I know of no one who has had warrant or installer issues due to genmon.

ssmoss, 2023-10-05:
@jgyates I just got eyes on my generator today and I noticed two interesting things. First there is a USB port next to the fuse on the top of the control panel. Second there is a sticker with information about the wireless module such as the MAC address and it also listed a private ip address (192.168.x.x). Are either of these new? Are you able to tap information via USB or is that to do a firmware restore for the controller. Does the private ip address for the wireless module give any information?

jgyates, 2023-10-05:
@ssmoss 

Neither of these are new. 

the USB port is for firmware upgrades. It does not support any other function. This is standard on Evolution 1.0 and 2.0 controllers.

The wireless module was introduced with Evolution 2.0. Genmon does not sue this connection a the firmware limits what is transmitted over the wireless module based on a subscription.

the short answer is, no. These two things do not provide any means of communicating to genmon.

ssmoss, 2023-10-11:
@jgyates @skipfire Have either you ran into this using a USB adapter? I saw [this](https://github.com/jgyates/genmon/wiki/Appendix-M-Tips-and-Tricks-useful-for-Genmon#configuring-usb-wifi-on-a-raspberry-pi) section but the module wasn't stock so I got [this](https://www.amazon.com/gp/product/B01CCMUN8C) which required compiling some drivers.

I initially had it going though a unpowered four port USB hub that worked fine in testing inside my house but seemed to get very unhappy once the generator turned on. I then plugged the adapter directed in it also seemed unhappy. Could this have been a vibrational issue? Power? (I am powering from the battery not via the Molex)

```
[ 1230.468383] usb usb1-port1: disabled by hub (EMI?), re-enabling...
[ 1230.468420] usb 1-1: USB disconnect, device number 2
[ 1230.533001] rtl8821cu 1-1:1.0: Runtime PM usage count underflow!
[ 1230.680364] Indeed it is in host mode hprt0 = 00001501
[ 1230.872320] usb 1-1: new high-speed USB device number 3 using dwc_otg
[ 1230.872506] Indeed it is in host mode hprt0 = 00001101
[ 1231.060408] NYET/NAK/ACK/other in non-error case, 0x00000002
[ 1231.060457] NYET/NAK/ACK/other in non-error case, 0x00000002
[ 1231.060481] NYET/NAK/ACK/other in non-error case, 0x00000002
[ 1231.268353] NYET/NAK/ACK/other in non-error case, 0x00000002
[ 1231.268404] NYET/NAK/ACK/other in non-error case, 0x00000002
[ 1231.268425] NYET/NAK/ACK/other in non-error case, 0x00000002
[ 1231.480322] usb 1-1: device not accepting address 3, error -71
[ 1231.480498] Indeed it is in host mode hprt0 = 00001501
[ 1231.672287] usb 1-1: new high-speed USB device number 4 using dwc_otg
[ 1231.672435] Indeed it is in host mode hprt0 = 00001101
[ 1231.880932] usb 1-1: New USB device found, idVendor=0bda, idProduct=c811, bcdDevice= 2.00
[ 1231.880962] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1231.880973] usb 1-1: Product: 802.11ac NIC
[ 1231.880982] usb 1-1: Manufacturer: Realtek
[ 1231.880991] usb 1-1: SerialNumber: 123456
[ 1234.436631] usb usb1-port1: disabled by hub (EMI?), re-enabling...
[ 1234.436690] usb 1-1: USB disconnect, device number 4
[ 1234.580565] Indeed it is in host mode hprt0 = 00001501
[ 1234.768491] usb 1-1: new high-speed USB device number 5 using dwc_otg
[ 1234.768730] Indeed it is in host mode hprt0 = 00001101
[ 1234.977327] usb 1-1: New USB device found, idVendor=0bda, idProduct=c811, bcdDevice= 2.00
[ 1234.977368] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1234.977387] usb 1-1: Product: 802.11ac NIC
[ 1234.977403] usb 1-1: Manufacturer: Realtek
[ 1234.977419] usb 1-1: SerialNumber: 123456
[ 1236.377397] usb usb1-port1: disabled by hub (EMI?), re-enabling...
[ 1236.377444] usb 1-1: USB disconnect, device number 5
[ 1236.524493] Indeed it is in host mode hprt0 = 00001501
[ 1236.712415] usb 1-1: new high-speed USB device number 6 using dwc_otg
[ 1236.712561] Indeed it is in host mode hprt0 = 00001101
[ 1236.928675] usb 1-1: device descriptor read/all, error -71
[ 1236.928813] Indeed it is in host mode hprt0 = 00001101
[ 1237.116441] usb 1-1: new high-speed USB device number 7 using dwc_otg
[ 1237.116601] Indeed it is in host mode hprt0 = 00001101
[ 1237.325041] usb 1-1: New USB device found, idVendor=0bda, idProduct=c811, bcdDevice= 2.00
[ 1237.325069] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1237.325080] usb 1-1: Product: 802.11ac NIC
[ 1237.325089] usb 1-1: Manufacturer: Realtek
[ 1237.325098] usb 1-1: SerialNumber: 123456
[ 1238.602122] usb usb1-port1: disabled by hub (EMI?), re-enabling...
[ 1238.602155] usb 1-1: USB disconnect, device number 7
[ 1238.744530] Indeed it is in host mode hprt0 = 00001501
[ 1238.932484] usb 1-1: new high-speed USB device number 8 using dwc_otg
[ 1238.932642] Indeed it is in host mode hprt0 = 00001101
[ 1239.141066] usb 1-1: New USB device found, idVendor=0bda, idProduct=c811, bcdDevice= 2.00
[ 1239.141095] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1239.141106] usb 1-1: Product: 802.11ac NIC
[ 1239.141115] usb 1-1: Manufacturer: Realtek
[ 1239.141124] usb 1-1: SerialNumber: 123456
[ 1240.350437] usb usb1-port1: disabled by hub (EMI?), re-enabling...
[ 1240.350472] usb 1-1: USB disconnect, device number 8
[ 1240.492646] Indeed it is in host mode hprt0 = 00001501
[ 1240.624834] Indeed it is in host mode hprt0 = 00001501
[ 1240.952523] usb 1-1: new high-speed USB device number 9 using dwc_otg
[ 1240.952849] Indeed it is in host mode hprt0 = 00001101
[ 1241.140511] usb 1-1: device descriptor read/64, error -71
[ 1241.248654] Indeed it is in host mode hprt0 = 00001101
[ 1241.456550] NYET/NAK/ACK/other in non-error case, 0x00000002
[ 1241.456587] NYET/NAK/ACK/other in non-error case, 0x00000002
[... 43 lines cut ...]
[ 1247.644674] usb 1-1: new high-speed USB device number 13 using dwc_otg
[ 1247.644828] Indeed it is in host mode hprt0 = 00001101
[ 1247.853341] usb 1-1: New USB device found, idVendor=0bda, idProduct=c811, bcdDevice= 2.00
[ 1247.853381] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1247.853392] usb 1-1: Product: 802.11ac NIC
[ 1247.853401] usb 1-1: Manufacturer: Realtek
[ 1247.853410] usb 1-1: SerialNumber: 123456
[ 1249.023536] usb usb1-port1: disabled by hub (EMI?), re-enabling...
[ 1249.023570] usb 1-1: USB disconnect, device number 13
[ 1249.164778] Indeed it is in host mode hprt0 = 00001501
[ 1249.352775] usb 1-1: new high-speed USB device number 14 using dwc_otg
[... cut ...]
[ 1249.936771] NYET/NAK/ACK/other in non-error case, 0x00000002
[ 1249.936809] NYET/NAK/ACK/other in non-error case, 0x00000002
[ 1249.936827] NYET/NAK/ACK/other in non-error case, 0x00000002
[ 1250.144836] NYET/NAK/ACK/other in non-error case, 0x00000002
[ 1250.144886] NYET/NAK/ACK/other in non-error case, 0x00000002
[ 1250.144914] NYET/NAK/ACK/other in non-error case, 0x00000002
```

I was able to find [one](https://www.amazon.com/dp/B0BHW6T96R) based on the same chipset from eBay which I will try next.

skipfire, 2023-10-11:
Sorry @ssmoss, I've never used a USB network adapter (WiFi or Ethernet) with a Pi.

jgyates, 2023-10-11:
I have not had this exact issue, but powering devices over USB can be an issue on the pi. There is a limit on what the pi can source in terms of current to the USB device and the pi's power supply must be up to the task as well. Unfortunately I don't know the exact limits and I assume the change with each pi version so google will likely provide better answers that I can. Keep in mind that adding a hub will add to the total USB power requirements. You may be better off using a USB-A extension instead of a hub as the hub takes power from the pi.

ssmoss, 2023-10-11:
Makes sense I am going to ditch the hub going forward. I am thinking the one I got is a newer chipset (faster / more bands) so more power hungry. I will report back when the ones with the older chipset come in.

@skipfire I am not sure how hard it would be, but it could be cool for those of us that need an antenna to have a version of the Pi HAT with one of these older wireless chips on them.

skipfire, 2023-10-11:
@ssmoss unfortunately all networking I have seen that goes through GPIO uses the same UART pins that GenMon uses.  I was hoping that the Pi 5 would have introduced an option for an external antenna but no luck.

ssmoss, 2023-10-11:
That external antenna is just too useful for them to add. Also kinda stinks if they did add it because the new Pi is sooo power hungry.

Would it be possible to use a "U-Shaped" USB connector and route that to the wireless chip on the HAT? I think I saw that done for some Pi 4 hats for using an external HDD. I guess you need a "hub" chip as well since it only has a single USB port?

skipfire, 2023-10-12:
That kind of connector could certainly work to get USB up to the HAT and if it is only a the WiFi then a hub chip shouldn't be needed since there would only be one USB device on the HAT.  Though the board is already a bit crowded so I'd probably end up just making a new board and if making a board that's just a USB device I'd probably just make it a dongle.

Do you need any GPIO other than UART?  If so, a NanoPi may be something to look at as the WiFi enabled ones have whip antennas.  I made a very small batch of test boards that fit them, but there is more lift to get the NanoPi going since they are not running Raspberry Pi OS.  The Neo Plus2 was easier than the Air since it had ethernet that I could connect to configure the WiFi.
https://www.amazon.com/gp/product/B0B971WS85/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
https://www.amazon.com/gp/product/B0B2K8QB3F/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

ssmoss, 2023-10-12:
For me at least the purpose of the device is to connect to the generator only and relay that to genmon. I am not tied to having genmon on that device either as my home network is battery backed as well as my home automation machine (which could host genmon).

Maybe the OpenGenSet board you have is a better way to do this as you have the external antenna already. I assume the only encryption on that is based on the security of the wireless network (WPA2)?

skipfire, 2023-10-12:
You are correct that it does not have any encryption on its own, just whatever WiFi encryption you have.  It does power over the 12v or 5v line from the Molex cable.  I have GenMon running on an Ubuntu VM that hosts a few other services when I'm not running a Pi directly connected.  It does of course run into the issue of Serial over TCP not being the greatest protocol ever built and lost packets sometimes result in some confusion in the system (always resolves itself) and potentially false outage notifications, but I adjusted a couple of the advanced delays and I get almost none now.  Moving GenMon to the VM instead of running on the Pi with a serial bridge setup reduced a lot of my errors, I think the lower performance of the Pi added just enough lag that it contributed to the problems.

ssmoss, 2023-10-12:
I saw that there is a SSH library for the ESP 32, is there any way that could be integrated into OpenGenSet? Load a key onto it and then use `autossh` from the VM to create a tunnel that the Serial over TCP passes though. With that the OpenGenSet would never expose anything save to the authorized hosts?

skipfire, 2023-10-12:
I really have no clue, I didn't create any software for OpenGenSet, just consolidated hardware that worked with the existing instructions for setting up the bridge.  I also have no clue if or how that may impact configuration.