jgyates/genmon #942: RS232/TTL failed

wrvolz, 2023-08-29:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Yesterday, Genmon went off line. It's did this about 1-2 months ago. Then, rebooting the RPI didn't help. But I found that a power reset did fix it. It went off line during a thunderstorm but lightning was several miles away. Power cycling didn't help this time.  I replaced the RS232/TTL interface and it's working again. This is the second time the RS232/TTL interface failed.

I suspect a static spike from somewhere. The RPI is powered by POE with a POE splitter to USB-c cable. The RPI is grounded to the generator battery. The POE injector is plugged into a UPS inside the house. I don't think the spike came from some induced voltage on the ethernet cable. I think it came from the utility power. The house does have a whole house surge suppressor but perhaps the spike wasn't enough to activate that. Rather than use POE, I could use a motorcycle USB-c converter to power the RPI from the generator battery and then use wifi so the ethernet cable would be eliminated.  Would this help?

Have you seen this before and is there anything I can do about it? Thanks

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here}

skipfire, 2023-08-29:
I would doubt that PoE is contributing to that, the Pi would probably get fried before the TTL board would if a surge was coming through PoE; and if it is coming through PoE then even having the ethernet line still connected may allow it to still happen.  Eliminating the ethernet cable entirely and using a motorcycle style adapter would be the best way to try to isolate it.

Quality of many TTL chips available on Amazon is questionable, if you are getting them from someplace like Adafruit may give slightly higher confidence of board quality.

wrvolz, 2023-08-29:
I don't think it came in from the POE, but possibly induced spike on the ethernet cable. But as you said, it would likely fry the RPI first and I'd also see other devices get hit. I think it was something from the utility The TTL interface is the first thing from the generator.

The interface I'm using is this one https://www.amazon.com/HiLetgo-MAX3232-Converter-Module-Serial/dp/B00LPK0Z9A/ref=asc_df_B00LPK0Z9A/?hvadid=390329205806&hvpos=&hvnetw=g&hvrand=12057979747322657643&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9010585&hvtargid=pla-906418052077&psc=1&tag=&ref=&adgrpid=79862087277&hvpone=&hvptwo=&hvadid=390329205806&hvpos=&hvnetw=g&hvrand=12057979747322657643&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9010585&hvtargid=pla-906418052077 [https://www.amazon.com/HiLetgo-MAX3232-Converter-Module-Serial/dp/B00LPK0Z9A/ref=asc_df_B00LPK0Z9A/?hvadid=390329205806&amp;hvpos=&amp;hvnetw=g&amp;hvrand=12057979747322657643&amp;hvpone=&amp;hvptwo=&amp;hvqmt=&amp;hvdev=c&amp;hvdvcmdl=&amp;hvlocint=&amp;hvlocphy=9010585&amp;hvtargid=pla-906418052077&amp;psc=1&amp;tag=&amp;ref=&amp;adgrpid=79862087277&amp;hvpone=&amp;hvptwo=&amp;hvadid=390329205806&amp;hvpos=&amp;hvnetw=g&amp;hvrand=12057979747322657643&amp;hvqmt=&amp;hvdev=c&amp;hvdvcmdl=&amp;hvlocint=&amp;hvlocphy=9010585&amp;hvtargid=pla-906418052077]

Which is one you recommend on your hardware page. I guess just bad luck - maybe a bad batch. Thanks, You can close this.

Bill Volz
832-724-5183

closed 2023-09-01 by jgyates