jgyates/genmon #943: Unknow error

BillyEll, 2023-09-01:
I just got this error UNKNOWN ALARM: 00020072 when my generator was trying to exercise. It said it was cranking in alarm, so I don't know if this was alarm already there or one that it generated when starting?
 
### Your Environment

- Generator Model: Generac 22kw air cooled, Natural gas.
- Genmon Logs: Done
- Genmon Version:  V1.18.18

jgyates, 2023-09-01:
I received your logs and registers. I have seen this alarm code before in logs that were automatically submitted, but you are the first to create an issue, which is good as I don't have enough info to resolve this without some additional input. 

When I look at your registers you have an entry in your alarm (actually a few)

     08/04/23 12:51:45 No Rotation Warning : Alarm Code: 0000

This time and date is the generator time and date

You also have an entry in /var/log/genmon.log that shows this:

     2023-08-04 12:51:50,667 : Alarm : Reg 0001 = 00020072

This time an date is the raspberry pi time and date

The way the Evolution controller works is that the value of Reg 0001 (value 00020072) the 0002 part means cranking, the 00072 part is the alarm or warning.  Looking at your alarm log I see a several "No Rotation Warning" messages, however in your genmon.log is exactly 2 of the 0072 unknown alarms. No Rotation is a warning and it looks like this is new for Evolution 2.0 and I am guessing that it is only active momentarily. It is possible that genmon could miss a warning if it is reset quick enough by the controller, which would explain why we do not see the exact number of these instances in genmon.log as we do in your alarm log.

The logs I have seen before did not have the data that allows me to correlate this 72 alarm code with an entry in the alarm log, but it looks like with yours I can safely say that code 72 = "No Rotation" warning. I have update genmon and checked it into github so you should be able to update your software form the about page and this will resolve the Unknown alarm message. You will still get a notification however it will be labeled "No Rotation Warning" instead of "UNKNOWN". The version number has not changed so the version will still be 1.18.18.

Thanks for submitting your logs. I am going to close this issue, but let me know if you have any questions or issues.

closed 2023-09-01 by jgyates

BillyEll, 2023-09-01:
Thanks!Billy ElliottSent from my iPhone