jgyates/genmon #944: Genmon Listen On 127.0.0.1 Instead Of 0.0.0.0

ssmoss, 2023-09-03:
Is it possible to configure `genmon` to run on `127.0.0.1` instead of `0.0.0.0` or on an explicitly configured address? I see in the configuration a way to set a custom port for  HTTP / HTTPS:

```
# (Optional) This parameter will allow the HTTP port to be set by the web
# interface. The default is 8000, but this setting will override that
# value. This option is only applicable to the web app.
http_port = 8000

# (optional) This parameter will override the default port for HTTPS, which is
# 443. Uncomment and change this value to use a non-standard port for HTTPS
https_port = 443
```

Looking at the server file on line [4394](https://github.com/jgyates/genmon/blob/master/genserv.py#L4394) and [4412](https://github.com/jgyates/genmon/blob/master/genserv.py#L4412) it seems like `0.0.0.0` is hardcoded.

My hope is to run `genmon` on `127.0.0.1` and using both a SSH and `ProxyPass` front `genmon` behind an Apache server. The   Apache server is setup with SSO and SSL allowing for single place to manage that.

I think I could make a PR for this if it is something you are amiable to unless there wider reaching implications to make this change.

skipfire, 2023-09-03:
@ssmoss I'm pretty sure that there is no config for it.  GenMon uses flask for the server, and looking at the code it is hardcoded to 0.0.0.0 (lines 4393 & 4412).

jgyates, 2023-09-03:
I am fine with the listen IP address being a configuration option for the loopback IP. This will help those who want to use apache instead of flask for the web server. This is the IP passed to flask for the web server and is in genserv.py. 

genmon also listens on all IP addresses (0.0.0.0) on port 9082 ( this is configurable in genmon.conf). It does this for the socket interface used for communication between genmon and genserv.

https://github.com/jgyates/genmon/blob/12590f5968c65cee60da2977f122466b990aff2f/genmon.py#L1539

this allows genmon to run on one system and genserv to be on another. All add on programs use this socket interface, including ClientInterface.py

jgyates, 2023-09-03:
I added the option for two new entries in /etc/genmon/genmon.conf

genmon_server_address  - this entry, if present, will accept an IP address in dotted decimal notation that will replaced the "listen all" address for the socket server in genmon

flask_listen_ip_address - this entry, if present, will accept an IP address in dotted decimal notation that will replaced the "listen all" address for the flask web server.

i have not updated the version number yet. Let me know if you have any questions,

closed 2023-09-03 by jgyates

ssmoss, 2023-09-03:
@jgyates holy crap man. Thanks so much!