jgyates/genmon #945: Can I run on a Synology Virtual Machine

tcflint, 2023-09-11:
I have a functioning setup on a pi but was thinking of transferring over to gregmac's setup with the ESP32 Serial Bridge.
I am running HA and would like to pull this into it.

I tried to setup a VM on the Synology NAS running Raspberry pi Desktop and installing the genmon software but I'm not able to log into it with the ip that Raspberry pi desktop is reporting.

Am I going about this the wrong way or will it not even work the way I'm thinking,
Thanks in advance

skipfire, 2023-09-11:
I have an Ubuntu VM for my host when I'm using an ESP32 bridge, I didn't see any benefit to trying to use Pi OS without a Pi, and now I host other stuff on that Ubuntu VM as well.

skipfire, 2023-09-11:
The same, I just say no for configuring serialport/bluetooth as that doesn't apply.  Raspberry Pi OS and Ubuntu are quite similar so instructions work just fine for the VM.  If you install Ubuntu on the Pi you have to do some extra work around permissions to access GPIO and such, but that doesn't apply when it's VM running Genmon to connect to an ESP32 bridge.

tcflint, 2023-09-11:
> The same, I just say no for configuring serialport/bluetooth as that doesn't apply. Raspberry Pi OS and Ubuntu are quite similar so instructions work just fine for the VM. If you install Ubuntu on [...]

So it should be the similar using the Synology NAS creating  a Linux setup?

skipfire, 2023-09-11:
I've never done a Synology, but if that leaves you with a working Linux instance that is Debian or Ubuntu (or anything that uses apt commands) then the genmon instructions should work.

tcflint, 2023-09-11:
> I've never done a Synology, but if that leaves you with a working Linux instance that is Debian or Ubuntu (or anything that uses apt commands) then the genmon instructions should work.

I should have some time this evening. I'll give it a go.
Thank you for the guidance

jgyates, 2023-09-11:
you can also search past issues on the "docker or vm" label for others with similar experience. If you are using a VM then using serial over IP is your best bet. When using serial over IP genmon only needs an internet connection running on a linux based system. The setup script does use apt for installing packages.

I am going to close this issue for now but feel free to post to this thread with questions or status.

closed 2023-09-11 by jgyates