jgyates/genmon #946: Question: Remotely Clear Schedule Warning

rsockanc, 2023-09-13:
Is there a way to remotely clean the A or B maintenance schedule warning?  I have a site that is only visited once per year so maintenance is performed slightly off and before the internal schedule.  After I left a few months ago, the warning for B came on and I won't be back out there for another year so it's just an annoyance.

- Genmon Version: 1.18.18

jgyates, 2023-09-13:
The short answer is possibly. What controller do you have?

I assume you have remote access to the pi?

jgyates, 2023-09-13:
Let's try this:

1) Update to the very latest software on the about page. The latest version is 1.18.18, which you have, but there are a few updates that have been done that have not constituted a version number change so be be sure we are on the same page, update to the very latest.

2) Try entering this command in the web browser:

     http://192.168.1.10:8000/cmd/setremote?setremote=resetmaintalarm

where http://192.168.1.10:8000 is the address you would normally use to access genmon. If you are using a username/login you must log in first then issue the above command.

3) Let me know if this works for you and what your controller model is. If this works I can enable this as a button on the maintenance page. If it does not work the the firmware in your controller is not supporting this command.

rsockanc, 2023-09-13:
Thanks for the quick reply!

Re-update complete:  HEAD is now at 2fde3cc V1.18.18
Reset complete: "Remote command sent successfully"
Pi rebooted for good measure.

The status page still says:

Status :

    Engine :
        Switch State : System in Alarm
        Engine State : Off - Ready
        System In Alarm : Service B Due
        Battery Voltage : 13.00 V
        RPM : 0
        Frequency : 0.00 Hz
        Output Voltage : 0 V
        Output Current : 0.00 A
        Output Power (Single Phase) : 0.00 kW

jgyates, 2023-09-13:
OK. thanks for the feedback. Which controller do you have? It tells you the controller on the Mainteance page. This let's me know what it does not work with.

closed 2023-09-13 by jgyates

jgyates, 2023-09-13:
That is an Evolution 1.0. Is your model a 50Hz, if so could you send your logs on the About page. This allows me to see the controller registers contents for testing. I don't have a lot of data on 50Hz models. 

The controller serial number will read a unique value instead of a formatted serial number if it has been replaced (FF FF FF FF FF FF). Looking at the registers I can validate this is the case.

rsockanc, 2023-09-13:
Sent

rsockanc, 2023-09-13:
The Genmon part is working great thanks to you; you may get a kick out of how I had to connect this one. The Generac add-on Wifi Mobile LInk died after 1 year and support was no help.  Luckily I had a 50ft cat5 and a spare Cisco serial to USB and a Pi was already onsite for another task.  I stole the wiring harness from the Mobilink and it worked. 

[image]

[image]

[image]

[image]

jgyates, 2023-09-13:
That is one better than replacing mobile link, reusing the cable is a great idea.

I did not recieve you logs, you have to have outbound email setup for this to work. alternatively you can do this to get me your logs if it is not too much touble.

1) Run this command from your browser:  

      http://192.168.1.10:8000/cmd/get_logs

this will download your log files in a file named genmon_logs.tar.gz in your browser's download folder.

2) Run this command from your browser:

     http://192.168.1.10:8000/cmd/support_data_json

Copy and past the output of support_data_json into an email and attach the file genmon_logs.tar.gz and send the email to generatormonitor.software@gmail.com

Thanks
Jason

jgyates, 2023-09-13:
to them. Thanks!