jgyates/genmon #949: Hardware Clarifications: Starting from scratch

thomasshawn, 2023-09-20:
Found this yesterday and have been deep in thre rabbit hole now for a few days....
I have multiple ESP32 units and an  RPI 3 i was previously running home assistant on.

My plan is to use the RPI and a ram disk (to save the SD card).
I am torn between using a piHat vs the USR-W630- which essentially same as the W610 many have used here (curveball I'm on a briggs generator so RS485-benefit is no special cable needed)

Watching the "genmon Briggs" YouTube videos, He mentions the W630 is generally easier to setup but one issue is that the Rpi will poll the device every second which could tax the home network. Seems like it could be a big deal and he mentions excluding it on some way.

in my mind that's 1 of 2 ways: 
1. direct connection between pi and W630-hardwired in some way- ethernet or other cable
2. seperate network for the generator data wireless or wired from the home network. 

 So I started  reading the not so great documentation on the W630, it isn't clear but can the W630/610 be directly wired to the RPI? seems like it is meant to transmit the data wirelssly. But since Pi and the W630 both have ethernet ports seems it might work. how would I exclude it from the netowrk. 

I assume those people using thre W610 also have the same issues. so how do you handle that issue, or is it really not that big of a deal? 

if not, is it better/easier to just get an RS485 hat for my Pi and cut out that interface? is there a significant advantage to the W630/610 over a PiHat? the PiHat are much cheaper.

say screw it and go with ESP32 which avoids SD card issues but still has choice between W630 and some add-on card 

ultimately my plan is to have the data sent to my home assiatnt, likely via MQTT.

jgyates, 2023-09-21:
I would not worry about taxing your home network. If you use a serial over TCP device like the ESP32 or the WR-630 the network traffic will be minimal due to the serial link communicating at 9600 bits per second and genmon will ask for one modbus register, then wait until one is received. This results in extremely modest network requirements as most networks are a minimum of 10 million bits per second, most are 100 million bits per second. have ran serial over IP with both wifi and Ethernet. 

The only real issue to be aware of regarding using the WR-630 or other wifi based serial over IP devices is that a weak wifi signal will result in data errors. See issue number 16 on [this page](https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues). Also review the [page on serial over IP in the wiki](https://github.com/jgyates/genmon/wiki/Appendix-F----Serial-over-IP) if you have not already.

Also, I have tried running RS-422/485 as my generator is at a distance greater than RS-232 will support. That works well for me also. [This page](https://github.com/jgyates/genmon/wiki/Appendix-O-RS-232-Greater-Than-100-Feet) has some info on that.

Let me know if you have any other questions.

thomasshawn, 2023-09-21:
My final paragraph about simply running a Cat6 as my transmission line from the generator to the Raspberry Pi RS 485 Hat and plugging it directly into the network would be the most direct and Least Complicated setup though correct? Max distance would be about 75 ft but probably much less.

I'm honestly just looking for the most reliable first and then the Least Complicated second and I'm wondering if there is an easy choice

jgyates, 2023-09-21:
I think that the RS485 would be the most direct and it is quite reliable. It is what I am currently using in my setup. I used [these connectors](https://www.amazon.com/SchmalzTech-Screw-Terminal-Breakout-ST-RJ45-R-MIN/dp/B09GYNJBKP?ref_=ast_sto_dp) and a Cat6 cable. You can also use the connector to easily make a loopback.

If you have the option, I would stay away from wireless as it is inherently error prone and modbus is not great at handling timeout errors as the protocol does not define a timeout or the canceling of a packet request. When you are wired there at zero errors and it just works.

thomasshawn, 2023-09-28:
OK, i managed to get everything installed. I run genmon and I have no gauges and I have a red exclamation point. I'm obviously getting some data, see screenshots but it's not all coming over and I can't tell why. I'm using this 485 piHat
[https://wisdpi.com/products/pi_485](url)
instructions are here
[https://wiki.wisdpi.com/common/how-to-use-raspberry-pi-uart](url).

I've checked serialtest and serialconfig and both pass. using the commands that are shown on the link above I get the correct 
**lrwxrwxrwx  1 root   root           7 Sep 27 18:44 serial0 -> ttyAMA0
lrwxrwxrwx  1 root   root           5 Sep 27 18:44 serial1 -> ttyS0**

but it tells me i need to be getting the opposite (see above) i followed the plan to have it as above and moved on.
I've tried serial and tcp over IP and still no difference.

I have checked continuity at the generator to the raspberry pi and  that is good on all 3 leads. wires are connected A-A B-B and Grd-Grd.

[https://imgur.com/Ryrc7d6](url)
[https://imgur.com/0rgVV3N](url)
[https://imgur.com/YkC0zAS](url)
[https://imgur.com/OX7ZPXw](url) -obviously sees the generator and identifies it
[https://imgur.com/caHIF4G](url)
[https://imgur.com/74VSv5R](url)
[https://imgur.com/mw3IMCT](url)
[https://imgur.com/Eh3RchD](url)

What else am i missing? only thing I wonder is if the toggles on the pihat are causing an issue?

jgyates, 2023-09-28:
@thomasshawn 
After looking at the screenshots it looks like your are not communicating with the generator. The Monitor page shows communication stats and you have not received any packets back from the controller so this validates the communication error. 

The web interface is showing the controller correctly because you have selected the Briggs custom controller in the advanced settings, not because gemon is communicating with the controller. 

Based on your comments above you are using 2 wire RS485. What are you using on the other end (the controller end)? 

Two wire RS485 requires that both ends (genmon and the Briggs controller) participate in throttling the direction on communication on the two wires since two wire RS485 is half duplex. Does the other end support automatic tranmit enable / disable? The RS485 adapter you are using has an option in the DIP switch for "Automatic DE/RE control". This should be enabled as genmon does not toggle support separate transmitter / receiver enable signal. Without the "Automatic DE/RE control" enabled the softeware would need to toggle GPIO 18 to determine if it is transmitting or receiving an gnemon does not do this.

Another thing to do would be to validate your communications with a loopback test to validate the operation of the pi, RS485 adaption and cable, but you can do this with the "Transmitter connected to Receiver" option enabled, but this needs to be disabled in normal operation as this would confuse gnemon to get back everything that it transmits. See [this page](https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting) for details about running a loopback test. If you run the loopback test make sure you stop genmon as only one program can access the serial port at a time.

I use RS422/RS485 trancievers in my setup but I use them in RS422 4 wire mode. It is possible to make 2 wire RS485 work, but both ends need to support the transceiver direction (like the "Automatic DE/RE control" option).

Let me know if this makes sense or if you have any other questions.

thomasshawn, 2023-09-28:
I started trying things and as i clicked over the dipswitch on #3 it came alive on the monitor page but still no updates on any of the gauges

[https://imgur.com/NMO0Z0s](https://imgur.com/NMO0Z0s)
[https://imgur.com/QoVuxkg](https://imgur.com/QoVuxkg)

so now it looks like i am communicating but not seeing anything on gauges

jgyates, 2023-09-28:
did you try refresh the page in your browser? When the page loads it asks genmon for how the gauges are configured so you may need to refresh the page. 

Also, you have the same number of validation errors as the number of received packets so this tells me that you still have an issue. 
 How is switch 2 set? this should be off (i.e. the opposite direction of switch 3).

If you outbound email is working your can submit your logs on the About page and I can look for errors or you can look at these files for errors:

/var/log/genmon.log   -- validation error are reported here, this should give some more details on the type of validation failure.
/var/log/mymodbus.log -- report modbus exceptions packets, and CRC errors and timeouts.
/var/log/myserial.log. - reports serial hardware errors

I don't know about the settings on the Briggs controller, have you validate that the serial rate is 9600 and the modbus ID is 2 on the controller side?

thomasshawn, 2023-09-28:
yes refreshed browser

Using python serial library V3.5b0
Serial port opened
write data: sent test string
waiting to received data....
PASSED! Loopback successful

I'll send the logs but i made an error pin 3 is up (default) pin 2 which is tie the transmitter to the receiver is the one that i turned on and it started working.

jgyates, 2023-09-28:
Just to clarify, switch 3 is always on, switch 2 is only on for a loopback. I have not received any logs yet. You can test your outbound email settings on the Settings page. Also when you restart genmon you will get two emails when genmon starts if you outbound email is working. genmon sends the logs via email. you can also issue the command in the browser:

http://192.168.11.14:8000/cmd/get_logs

this will download a compressed archive (named genmon_logs.tar.gz) of the logs which you can email to generatormonitor.software@gmail.com if your outbound email is not working yet.

jgyates, 2023-09-28:
also, keep in mind that if you had switch 2 on while genmon is running, this could be the cause of you getting packets, but with validation errors so there is still a likely possibility that the controller is not getting the data.

If switch 2 was on then genmon would receive every packet it sent, but the packets would fail a validation test as they are not in the format expected to be returned from the controller.

jgyates, 2023-09-28:
I got your logs

thomasshawn, 2023-09-28:
Also I initially purchased the 2 wire because it was the recommended one from genmon briggs if you weren't going to use the wifi bridge adapter.

jgyates, 2023-09-28:
is your controller set to Even parity? I would try to set this to No parity and then make genmon the same setting and see if that makes a difference. You are getting errors like this:

2023-09-27 21:04:38,762 : Validation Error, length: Master12 Slave: 3
2023-09-27 21:04:38,762 : Master = [0x01,0xe6,0x00,0x00,0x00,0x06,0x02,0x04,0x00,0x0b,0x00,0x01]
2023-09-27 21:04:38,763 : Slave = [0x02,0x04,0x00]

the received packets are to properly formatted modbus packets so it looks like something is wrong on the serial side of things.

also, I checked in a change that corrects an issue that may prevent the serial port from being configured to even parity so you can up the software from the About page and see if that helps. The version number did not change. After you update, send your logs so I can validate if that made a difference.

jgyates, 2023-09-28:
with your latest logs it looks like you are getting timeout errors for all packets sent, which indicates that either the controller is not receiving the data that genmon sends, or genmon is not received the data that the controller sends (i.e. genmon sends a request but no response is received).

Photographs of your wiring may be needed. Also I never got a response on the connections on the Briggs side, is it native RS485 or do you have a converter of some type?

thomasshawn, 2023-09-28:
2nd set of logs is with the #2 dip switch turned off. and I'm back to getting zero information

Per the "genmon briggs" youtube videos that you link in your wiki, he says it is a native rs485 and says we have to use "Use Modbus FC4 instead of FC3" flag.

i am able to change many of the settings in the controller on the briggs side but I cannot change the baud rate or the parity.

so we are stuck with even parity but that's how he set it up on the youtube vids you sent as well.

I used Cat5 cable (~20ft) I test continuity before I used the wire after everything was hooked up.
I wired A-A, B-B and Ground-Ground. 

I wonder if it is an issue with the piHat?

[https://www.youtube.com/watch?v=Kcm8S2HIKFk](https://www.youtube.com/watch?v=Kcm8S2HIKFk) starts at 10secs talking about RS485 then at 4:50 point on video he discusses the 2 wire hat

thomasshawn, 2023-09-28:
it's only a 2 wire setup on the briggs side
[https://www.youtube.com/watch?v=OtRALU_7f2w](https://www.youtube.com/watch?v=OtRALU_7f2w) jump to 0:50 on video

jgyates, 2023-09-28:
one of the drawback with 2 wire is that is hard to do a loopback through the cable since Tx and Rx are on the same pair of wires. With switch 2 on we are looping back on the RS485 adapter on the pi but the data is not going thru the cable but we have yet to validate the data through the cable and with two wire we really can't do that independently of the controller. With a 4 wire interface you can cable it in two wire or four wire mode but a 2 wire interface can only operate in 2 wire mode.

Try this first: Just as a data point you may try swapping A and B on the controller connection and see if that makes a difference. This will just invert the data so it will not harm anything to try this.

thomasshawn, 2023-09-28:
Will try and let you know. Also ordered a 4 wire. Looking online I jumper the + together (tx/RX) and the - (tx/RX) together and then run those to the A/B on the controller (briggs ) side? I guess at that point a/b are irrelevant ?

thomasshawn, 2023-09-28:
swapped wires. no change. i have a couple of new serial pihats on the way. Should be here tomorrow. will try to get them swapped out and let you know.

is it as simple as powering down the Rpi and swapping the adapters and retesting serialtest (although mine have always passed) or will i need to rerun the genmon setup ?

jgyates, 2023-09-28:
you may have some settings with switches like before. Another thing that I don't see is terminator resistor settings. There should be a terminator resistor on both ends of the cable. Typically this is down with a setting or switch on the RS485/RS422 adapter. It looks like your adapter has a jumper for termination. Is the termination resistor enabled? You want it to be enabled. If you had more than 2 adapters on the two wire network you would only enabled it for the two on each end of the cable and disable it for the others.

thomasshawn, 2023-09-28:
I tried that as well. Currently the resistor is on the jumper is in position 2-3 from the spec sheet I posted above.

thomasshawn, 2023-09-28:
I'd like to keep this as a hardwired system. Ultimately I had planned on running this over to my router and plugging my RPI into my network switch. So the whole setup avoided Wi-Fi.

But I'm beginning to wonder if the wireless bridge option isn't a viable consideration (only because in the setup videos for the Briggs linked on this page he describes the Wi-Fi Bridge as the easiest to set up and describes the Hat as more fiddly) . Or, They actually have one that is an RS 485 to ethernet and I could plug that into the RPI to receive the signals. And then only use the Wi-Fi to access the genmon software front end when needed. But I'm not familiar enough with the pie to know if it can handle two data streams across two different ports like that, essentially functioning as a router.

I guess I will play with the other hat options and if those don't work I will move on to the Wi-Fi Bridge. I would just love to keep that data on a separate Network but maybe that's just me being overly concerned and I shouldn't worry about the bandwidth/security.

jgyates, 2023-09-28:
if you are going to try the serial over IP route, you could just put both the pi and the serial converter on Ethernet. This would avoid wifi. You will have to address routing issues on the pi if you try to use Wifi and Ethernet at the same time with the pi. It can easily handle the bandwidth, the issue is telling the pi what resides on each network and when to use each network. It is doable, but requires more advanced configuration of the pi networking. As a data point, my pi is running serial an the CPU utilization is a 5%. If you use a pi zero it can be higher but generally the bandwidth is not an issue as 9600 is baud is quite slow and the pi has no issues with that rate.

You may want to post some pics of your setup (the cable and the configuration on the Briggs). To date we have no indication that the Briggs controller has sent anything back. 

You can attach pictures to this thread if you use the github web interface or use imgur link like you did before.

jgyates, 2023-09-28:
also, on the last couple of log submissions it looks like switch 2 is enabled or loopback is still active as there are validation errors with garbage data.

thomasshawn, 2023-09-28:
Yes I followed those instructions. I'll post pics this afternoon. I didn't take any when I installed but mine looks like his. Other end plugs into the a, b, G  on the hat.

jgyates, 2023-09-28:
I just go your logs, still seeing the validation errors. If switch 2 is off then you may have a short in your cable. Is switch 3 still enabled? Include a photo of your pi when you are able to take pictures. 

Another thing to try is to set switch one to off (on is Receiver always on, which we do not want). In two wire mode we want to transmit with our receiver off and when we are not transmitting have the receiver on. The wiki on the board is a bit lacking and the only data sheet I can find on the IC they use is in Chinese but I would try changing switch 1.

thomasshawn, 2023-09-28:
I have remote access to genmon so I can sendogs and make tweaks but can't change the pi at this point.

Default on switch 1 is off. And that's where it is now.
So currently we are OFF OFF ON OFF.

I tried it with switch one ON this morning and it didn't seem to make a difference.

On the cable, I checked continuity Before and after I wired it up. I'm showing continuity from the connecting screw on the terminal block to the terminal block on the piHat on all 3 wires. But I can definitely swap it out.

thomasshawn, 2023-09-28:
Here's the wiring. I rewired with short jumpers to remove the shorted wire possibility 
Sent new logs just now 

[image]
[image]
[image]
[image]

thomasshawn, 2023-09-28:
In the video specifially on the wifi bridge he addresses these settings
baudrate 9600, parity Even, Stop 1 and adaptive baudrate

are these settings I need to try and recreate on the RPi?
[image]

jgyates, 2023-09-28:
Those appear to be the settings you need for the Briggs controller. The pi only needs an IP address and a port number.

Here are some screen shots of when I used the 610. The settngs are not going to be the exact same, but they will be close (excpet for serial parity is even for Briggs an your IP and port would be unique to your network)
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

thomasshawn, 2023-09-28:
Perfect. I'm hoping not to switch to the 630. I'm just wondering if I need to try and set any of those other settings for the current serial connection? I'm just looking for a reason why it's not working. I went so far as to pull another RPI 4B out of another application swap cards throw that hat on it hoping that maybe it was a bad RPI however it did exactly the same thing. I then went and got a traditional serial cable and plugged it in to the DB9 port and ran jumpers off of the appropriate pins check for continuity to make sure everything was connected and same results.

I've checked loopback I've checked serial. I guess at this point I'm in a holding pattern until the four-wire shows up.

 two wire 485 setup is actually three wires because one goes to ground. However the four wire setup does not allow for a ground wire.

My question,  is the ground wire not required on the four wire setup?

jgyates, 2023-09-28:
you don't technically need a ground for either of them since it is a differential interface, however a ground certainly will not hurt. A 1 or a 0 is determined by the difference in voltage between A and B so you don't need to be on the same ground reference (e.g. the difference between A and Ground and B and Ground).

Your last few logs did not show the validation errors, only timeout errors. Did something change? That may be a clue.

Before you change away from serial another data point would be: do you see any difference in the serial communication stats failure if the cable is connected vs disconnected from the controller?

thomasshawn, 2023-09-28:
I'm not sure I understand your last comment. Are you asking if it's any different with the DB9 cable jumpered versus hardwired from terminal block to terminal block?

Also when I have the generator opened up to access the wiring I can see the motherboard and there is an LED for CAN, AVR, and 485. At rest in standby mode the Can flashes slowly, and the AVR stays lit. But the 485 has never come on.

I don't know if that's just a status light that will only come on when there's communication across that 485 terminal. But it hasn't once blinked
[image]

jgyates, 2023-09-28:
The last sentence is a test that would possibly shed some light on why you get validation errors sometimes and you do not at  other times. If you get validation errors, do you also get them when you disconnect the cable from the Briggs controller?

That is not a great sign about the 485 LED. I guess you don't have any other device that you could use to validate that the RS485 port actually works on the controller?

thomasshawn, 2023-09-28:
I think i was getting timeout errors because I wasn't connecting.
So I went back and ran serial tests again.
Disconnected Pi from the generator,

jumper on pins 8/10 passes.
jumper across the DB9 plug (piHat) or the terminal block (piHat) fails serial test.

the only way it passes a test is if it has switch 2 turned to ON. then it passes whether I'm jumpered or not.

so I think the PiHat is not doing it's job.

jgyates, 2023-09-28:
I am not sure what you mean by "i was getting timeout errors because I wasn't connecting."

I was not talking about a loopback test. I was referring to running genmon with serial, not TCP. When you ran genmon you said that switch 2 was off yet you still got validation  (not timeout) errors? If that is true then this could be a clue as to why this is not working.

Regarding a loopback test, what you describe above with the loopback is normal for two wire 485 with the receiver not always enabled (switch one off). With two wire the transmit wire and receive wires are connected so anything that is transmitted is received instantly, but if the received is disabled when transmitting the loopback will fail. This is great for a loopback, but is not desirable if the application (like genmon) was written for anything else besides 2 wire RS485 as it is not expecting to receive what is just sent out, but rather is is expecting to receive from another device (the controller) only. 

For loopback you want the receiver on (switch one one) while you transmit.
For genmon operation you do not want the receiver on always (switch one off). 

If you set switch 1 on and turn switch 2 off, and switch 3 on, can you jumper across the db9 and run the loopback test? If you can then remove the DB9 jumper and disconnect the cable from the controller, then put the jumper on A-B at the far end of the cable, then run the loopback test again. This should allow you to do a loopback test through your cable.

thomasshawn, 2023-09-28:
i don't think it was connecting it was showing not connected with the exclamation point.

**_If you set switch 1 on and turn switch 2 off, and switch 3 on, can you jumper across the db9 and run the loopback test?_**
_Serial port opened
write data: sent test string
waiting to received data....
PASSED! Loopback successful_
**_then remove the DB9 jumper and disconnect the cable from the controller, then put the jumper on A-B at the far end of the cable,_** 
_Using python serial library V3.5b0
Serial port opened
write data: sent test string
waiting to received data....
PASSED! Loopback successful_

that is working but those settings aren't conducive to genmon i guess. so we need to figure something else out. I have a 4 wire and a different 2 wire arriving tomorrow we can try again then.

Will I need to rerun setup or do anything different besides swap the piHats?

jgyates, 2023-09-28:
if you hover your mouse over the exclamation point it will give you more detail. If it says "Disconnected form server" that means that genmon is not running and the web interface can not communication with genmon.py. This occurs when you stop genmon but the web page is still in a browser window. That message has nothing to do with serial communications. If you have serial communication issues and you get and exclamation point the message will be something like "Not receiving data".

Regarding the setup with a different Hat, probably not, but it really depends the Hat. Genmon does not know anything about RS232 or RS485 so there is nothing to do regarding genmon. The Hat may requires something, but if it has an "auto enable transmitter" switch, just set that.

The two test you ran have validated that data is being passed form the pi through the Hat and looped back through the cable. The loopback tests use the same code for serial communication that genmon does. Now, if you set switch one to off and try to communicate with the controller and it does not work, to me this is a strong indication that the controller could be the problem. That with the fact that the 485 LED never comes on is not a good picture for your controller working with RS485. We can still test with the new Hat and the WR630 if you like.

thomasshawn, 2023-09-29:
[image]
That was the must unintuitive device ever.

I don't know how I finally got it working but I did.
This is using the w630.

I tried 3 different hat devices. This worked once  I got the server situated. Wow. What a pain.

It claims to have wifi, and 2 ethernet ports. So I'm wondering if I can do what I asked before and have it be on a sepetate network.

Thinking about that.... The pi needs to be on 2 networks I guess not the 630. W630 hardwired into the pi via ethernet. The pi talks to my network via wifi-or via some USB to ethernet to keep everything hardwired?

thomasshawn, 2023-09-29:
I'm just trying to keep the extra traffic off the wifi and also I just prefer things to go wired when possible. 

[image]
I also picked up this one as a backup but price iIsn't dramatically different and I already have the 630 working, so if I can make it work over ethernet that might be better than battling another device.

jgyates, 2023-09-30:
I am glad it is working. 

did you try more than one hat? and you never got the hat working correct? 

I am going to close this issue but feel free to post to this thread with follow up status or questions.

closed 2023-09-30 by jgyates

thomasshawn, 2023-09-30:
Multiple Pihats were tried and configured no success.

Also for anyone reading this later the status LED on the generator for the RS485 protocol doesn't light up even once connected and receiving information at least on my unit.