jgyates/genmon #950: Current sensor HAT by PintSize Me

philruel, 2023-09-23:
I have a Current Sensor HAT by PintSize Me for more than a year.  Yesterday, I was doing an Enhanced Exercise with Transfer and took some screen shots of the discrepancy between Amps and kW that I have been noticing since I installed it.

### Expected Behavior
[image]
I had all these devices powered on.  Those are the approximate max I was pulling. Some devices were cycling.

### Actual Behavior
[image]
The External Curent of 88.33 amp is probably right.  But the External Power of 10.6 kW is most probably wrong.  It's only using one leg  to calculate. 

Is there a way to correct the 10.6 kW.  Should it be 21.2 kW?

Thanks you,
Phil

- Generator Model: 22 kW  0065520 with no current transformers in machine
- Genmon Logs: Submited
- Genmon Version: V1.18.18

skipfire, 2023-09-23:
For the math you should not do totals but add the legs.  Rather than 88.33 x 240 you should do  43.43 x 120 + 44.90 x 120 which is 10,599.6W and spot on with your screenshot.

jgyates, 2023-09-23:
@philruel 

Thanks for the question. Genmon takes the readings of the two CTs and the generator output voltage. It divides the output voltage by 2 (since the controller does not give the per leg voltage) and it calculates the power as @skipfire describes. Let us know if you have any other questions. I am going to close this issue but feel free to post here with any follow up questions.

closed 2023-09-23 by jgyates

philruel, 2023-09-23:
Thanks for your replies.
The numbers are right.
The needle of dial gauge for External Current is in the red zone and shouldn't be.  I'm pulling 88.33 total amps and in the red zone at 10.6 kW (in the green zone).  If I continue drawing up to my maximum rated load of about 90 amp, that number per leg would be off the chart.

Thank you,
Phil

jgyates, 2023-09-23:
@philruel 

Maybe I am calcuating it incorrectly. The maximum rated current output of your 22kw generator is 91 for LP and 81 for NG according to the spec sheet, which you can downlaod on [this page](https://www.generac.com/service-support/product-support-lookup/product-support-details?productid=c4b6c88b-d4b6-4419-a670-ae40b760cd61&serialNumber=G007042-0&registered=false).

The red zone on the External Current gauge is calculated based on Rated watts / nominal line volts (240V) which is in range with the spec sheet.

If I calculated the power the way you originally suggested, then 45.5 A on each leg (with a maximum of 91 total) would result in 21.8kw which would make the max amps and max watts more inline.

If I change this calculation of the power calculation to this:

   (CT1Amps + CT2Amps) * OutputVoltage = Total Power

Is this in line with your thinking? 

@skipfire, any thoughts?

skipfire, 2023-09-23:
I think the problem is just the External Current graph, not any math.  An ideal situation would show a needle for each leg, but since I don't think it supports that, you could go with either the higher of the 2 legs or an average of both legs.  Either way would get it correctly representing about 50% on the External Current graph.  Each leg can do 91A for 11kW (if on propane, 15% less on natural gas).

jgyates, 2023-09-23:
Would making the External Current gauge start in the red at 182A instead of 91A be a better representation?

skipfire, 2023-09-23:
That would be fine as long as the legs are not significantly out of balance; but the generator would not like that situation anyway.

jgyates, 2023-09-23:
@philruel 

I checked in a change that will show 2 gauges (Current L1 and Current L2) instead of one (External Current). Let me know how this looks to you.

philruel, 2023-09-23:
That's awesome.  Can't wait to try it next week.
Very impressed of the fast and detailed replies.

philruel, 2023-09-24:
On that same Current Sensor HAT, it has an interface for a temperature sensor.  The Wiki says:  "Support for the temperature sensor is provided in the [gentemp add on](https://github.com/jgyates/genmon/wiki/1----Software-Overview#gentemppy-optional)."
I've read through gentemp.py section but I don't understand what I have to do to enable the temperature sensor.
I've turned on the Add On External Temperature Sensors but don't know what values to enter.

[image]

Thanks again!

jgyates, 2023-09-24:
To use the temp sensors, the one wire interface must be enabled on the raspberry pi and you must have a supported temperature sensor connected to the 1 wire interface on the CT Hat. You can enable the 1 wire interface with the raspi-config program. Run this with root privileges like this:

     sudo raspi-config

Once you run the program select "Interface Options" then "1 Wire", then enable the interface. Then exit from the raspi-config program and reboot your pi. Then enable the temp sensor add on. Once genmon restarts you should be able to see the temp sensors on the Status page. The temp sensors will only show up in the text on the status page unless add values for the "Sensor Name" "Sensor Nominal Values" and "Sensor Maximum Values". If these values are present then a gauge will be created for each temp sensor.

Let me know if you have any other questions.

philruel, 2023-09-25:
I enabled "1 Wire" in config and saw the temperature in the Status page.  I then started entering values for the "Sensor Nominal Values" and "Sensor Maximum Values" and the gauge displayed.  Not knowing what to enter as values, I entered random numbers and tried to figure out a pattern.  After 10-15 tries, Genmon tried to load for many hours and never getting through.
[image]

I disabled the "1 Wire" in config, rebooted and Genmon loaded correctly. I then disabled the Add On "External Temperature Sensors" and reenabled "1 Wire" in config and Genmon loaded fine.

Now, as soon a I enter values in "Sensor Nominal Values" and "Sensor Maximum Values", Genmon doesn't load.
I submitted logs.

Thanks!

jgyates, 2023-09-25:
Can you tell me what values you entered for "Sensor Nominal Values" and "Sensor Maximum Values" and how many sensors are attached to the pint size me hat?

Sounds like you have entered data in a format that the software is not expecting. I can add more bounds checking if you tell me what you entered.

You should have the same number of entries that you have sensors, for example if you have one sensor then you should use something like this:

Sensor Name = Sensor One
Sensor Nominal Value = 170
Sensor Maximum Value = 250

If a sensor was present, this would add a gauge named "Sensor One" with a maximum range of 250 and the normal range would be 170. These values are in Fahrenheit unless you have Use Metric enabled.

If you have two sensors then  

Sensor Name = Sensor One, Sensor Two
Sensor Nominal Value = 170,175
Sensor Maximum Value = 250,300

I have attached examples of a running system.
[image]
[image]

jgyates, 2023-09-25:
Hopefully the above will help, but in my testing I found a bug that will likley not effect you, but I also added some additional bounds checking on gentemp.py. If the number of entries in the name, nominal and max do not match then it will not attempt to show a gauge. Let me know if you have any issues.

philruel, 2023-09-25:
I don't remember what values I put at the end.
I have only 1 sensor.
I copied your values and works now.

Awesome work!