jgyates/genmon #951: Genmon Login Locked Messages without https enabled.

liltux, 2023-09-28:
Just updated Genmon and turned off use HTTPS and login.  Now I keep getting alerts every so many minutes that the login is locked due to too many logins.

[image]
 
Submitted Logs and Registers

jgyates, 2023-09-28:
The file /var/log/genserv.log shows this:

2023-09-28 10:32:43,387 : Invalid login: admin
2023-09-28 10:32:43,396 : Invalid login: admin
2023-09-28 10:32:43,399 : Invalid login: admin
2023-09-28 10:32:43,422 : Invalid login: admin
2023-09-28 10:33:13,357 : Invalid login: admin
2023-09-28 10:38:13,773 : Invalid login: admin
2023-09-28 10:38:13,829 : Invalid login: admin
2023-09-28 10:38:13,831 : Invalid login: admin
2023-09-28 10:38:13,864 : Invalid login: admin
2023-09-28 10:38:43,424 : Invalid login: admin
2023-09-28 10:43:56,440 : Invalid login: admin
2023-09-28 10:43:56,573 : Invalid login: admin
2023-09-28 10:43:56,697 : Invalid login: admin
2023-09-28 10:43:56,702 : Invalid login: admin
2023-09-28 10:44:26,433 : Invalid login: admin
2023-09-28 10:49:26,487 : Invalid login: admin
2023-09-28 10:49:26,580 : Invalid login: admin
2023-09-28 10:49:26,663 : Invalid login: admin
2023-09-28 10:49:26,682 : Invalid login: admin
2023-09-28 10:49:56,523 : Invalid login: admin
2023-09-28 10:54:56,541 : Invalid login: admin
2023-09-28 10:54:56,580 : Invalid login: admin
2023-09-28 10:54:56,644 : Invalid login: admin
2023-09-28 10:54:56,702 : Invalid login: admin
2023-09-28 10:55:26,418 : Invalid login: admin
2023-09-28 11:00:26,473 : Invalid login: admin
2023-09-28 11:00:26,611 : Invalid login: admin
2023-09-28 11:00:26,615 : Invalid login: admin
2023-09-28 11:00:26,667 : Invalid login: admin
2023-09-28 11:00:56,471 : Invalid login: admin
2023-09-28 11:05:56,699 : Invalid login: admin
2023-09-28 11:05:56,774 : Invalid login: admin
2023-09-28 11:05:56,920 : Invalid login: admin
2023-09-28 11:05:56,953 : Invalid login: admin
2023-09-28 11:06:26,615 : Invalid login: admin
2023-09-28 11:11:26,678 : Invalid login: admin
2023-09-28 11:11:26,727 : Invalid login: admin
2023-09-28 11:11:26,748 : Invalid login: admin
2023-09-28 11:11:56,495 : Invalid login: admin
2023-09-28 11:11:56,532 : Invalid login: admin
2023-09-28 11:16:04,776 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
2023-09-28 11:16:26,739 : Invalid login: admin
2023-09-28 11:16:26,741 : Invalid login: admin
2023-09-28 11:16:26,764 : Invalid login: admin
2023-09-28 11:16:26,832 : Invalid login: admin
2023-09-28 11:16:56,657 : Invalid login: admin

4-5 failed login attempts every second.

Is this exposed on the internet?

liltux, 2023-09-28:
Negative and I have Https turned off and there is not login.

closed 2023-09-28 by liltux