jgyates/genmon #952: Specific Wireless Interface On Monitor Page

ssmoss, 2023-10-08:
Is there a way to specific which wireless interface is monitored under the `Platform Stats` page? I currently have two, the onboard one `wlan0` and a usb one `wlan1` which I am treating as the primary. The `Platform Stats` page seems ti just `wlan0` though.

```
Platform Stats :
        CPU Temperature : 134.42 F
        Pi Model : Raspberry Pi 3 Model A Plus Rev 1.0
        Pi CPU Frequency Throttling : OK
        Pi ARM Frequency Cap : OK
        Pi Undervoltage : OK
        CPU Utilization : 7.21%
        OS Name : Debian GNU/Linux
        OS Version : 11 (bullseye)
        System Uptime : 5 days, 15:41:06
        Network Interface Used : wlan0
        WLAN Signal Level : -64 dBm
        WLAN Signal Quality : 46/70
        WLAN Signal Noise : -256 dBm
        WLAN ESSID : xxx
        System Time : Sunday October 8, 2023 13:02:57
```

jgyates, 2023-10-08:
Not presently. If you are not using wlan0 I would just disable that adapter. Even if you don't disable it, but just don't allow it to connect to an access point it should not be selected. 

or just "sudo ifdown wlan0"

ssmoss, 2023-10-08:
My plan is to run both and use the internal one as a fallback. I have adjusted the metric for the usb one to be lower compared to the internal one to it is favored.

closed 2023-10-08 by jgyates