jgyates/genmon #953: MQTT Output As JSON Blob

ssmoss, 2023-10-08:
For the MQTT output it seems like it writes each attribute out to its own topic as shown below:
```
generator/Monitor/Communication_Stats/CRC_Percent_Errors: '0.00%'
generator/Monitor/Communication_Stats/Timeout_Errors: '512'
generator/Monitor/Communication_Stats/Timeout_Percent_Errors: '99.81%'
```
This seems very useful for powering a stateless gauge but it makes it a bit harder for wanting to keep a historical snapshot of the data.

Is there a way or it is possible to add the ability to have it output a JSON encoded message of all the data per poll interval? It could be as simple as taking each subordinate topic to `generator` and rolling it up into JSON. An example of this would be:

```json
{
    "Time": "CLOCK TIMESTAMP / CURRENT SYSTEM TIME"
    "Status" :
    {
        ...
    },
    "Maintenance" :
    {
        ...
    },
    "Outage" :
    {
        ...
    },
    "Monitor" : 
    {
        "Communication_Stats" :
        {
            "CRC_Percent_Errors" : '0.00%'
            "Timeout_Errors" : '512'
            "Timeout_Percent_Errors" : '99.81%'
       }
    }
}
```
This could be published to `generator/Rollup` if enabled. This structure would support the three existing configuration options of `JSON for Numerics`, `JSON for String Lists`, and `Remove Spaces in Topic Path`.

ssmoss, 2023-10-08:
Looking at the existing code https://github.com/jgyates/genmon/blob/master/addon/genmqtt.py#L236 it seems like a simple if statement could be added after this line:

```python
if (self.UseRollup) {
    self.Callback("/generator/Rollup", json.dumps(GenmonDict))
}
```

Looking at the existing code https://github.com/jgyates/genmon/blob/master/addon/genmqtt.py#L404 it seems like the configuration option could added around this line:
```python
            self.UseRollup = config.ReadValue(
                "use_rollup", return_type=bool, default=False
            )
````

jgyates, 2023-10-09:
This is possible, however unless I am missing something, this appears to be redundant and a recipe for degraded system performance on the network and / or your MQTT broker. One of the main advantages of MQTT is that it only sends data when it changes. If your MQTT broker stores the data, you have a historical record of the data that operates much like a differential backup since you only record the data that changes. The addition of this feature would erase that advantage. Adding this would guarantee that **all** of the data is sent each polling period (the default is 2 seconds). Since this data includes the system time and the number of packets sent the JSON data would be different every 2 seconds. If your MQTT broker is setup to store the data, that would increase the data storage exponentially.

Also, if you do want all the data resent you can set the flush interval to a non zero value on the MQTT add on. This will resend all topics, even if they have not changed.

Alternatively, ff you want the raw JSON data, you can get it from the genmon web API like this:

     http://192.168.1.10:8000/cmd/status_json
     http://192.168.1.10:8000/cmd/maint_json
     http://192.168.1.10:8000/cmd/outage_json
     http://192.168.1.10:8000/cmd/logs_json   (MQTT does not send this info)
     http://192.168.1.10:8000/cmd/monitor_json

Or does the MQTT transport facilitate this in a better way? Just trying to understand the use case better.... Adding the code is easy, the support of redundant code is the thing I try to avoid if possible.

ssmoss, 2023-10-09:
Yea, I do see the point about filling up the MQTT broker if one isn't careful and generating more network traffic due to the always sending. One advantage at least in my mind about using the MQTT is that it would act like a buffer should the consumer be offline. In my case I am running a local broker on the Pi and would be exporting the data to a database running on a home automation computer. MQTT would act as a buffer should the link to the home automation computer be down but this is rather specific to me. I took a stab the a PR https://github.com/ssmoss/genmon/tree/mqtt-improvements .

That said using the HTTP API might just be a simpler more straightforward way to do it. No sense reinventing the wheel.

Let me try that out and see.

ssmoss, 2023-10-09:
So I took a quick look at the output and think I can make that work. I assume the options to remove spaces and break numeric values up isn't supported?

```json
:8000/cmd/status_json | jq
{
  "Status": [
    {
      "Engine": [
...
        {
          "Battery Voltage": "0.00 V"  # {"BatteryVoltage": {"unit": "V", "type": "float", "value": 0.0}}
        },
...
```

jgyates, 2023-10-09:
you can use this command:

    http://192.168.1.10:8000/cmd/status_json
 
or this one:

     http://192.168.1.10:8000/cmd/status_num_json

The first will give the data with units. This is how the info on the web page is displayed. The scond one will give it like you have it in the comment.

ssmoss, 2023-10-09:
Ah cool, thanks! Based on this it would seem my ask is redundant.

closed 2023-10-09 by ssmoss