jgyates/genmon #955: [Question] - Command Clarification: setremote

ssmoss, 2023-10-12:
From [genmonpy-required](https://github.com/jgyates/genmon/wiki/1----Software-Overview#genmonpy-required)
```
The setremote command issues commands directly to the controller. Start and stop commands
should be back to back, i.e. is is not recommended to issue start followed by starttransfer, you
should have a stop after each start command. Also, when the starttransfer command is used,
once a stop command is issued the controller will set the transfer switch back to "utility" but the
generator will continue to run in "cool down" mode for few minutes. This is normal and the
behavior is controlled by the controller.

Remote commands that start and stop the generator engine will not work if the following
conditions are true:

Mains power is disconnected, in any mode
The generator is in 'off'
The generator is running in 'Manual'
```
A few points of clarification on these commands:

1. What happens if you issue a `start` or `starttransfer` command and the power goes out or the breaker in the transfer switch is thrown?
2. Are the `start` and `stop` command there for testing purposes only or are there some intended use cases for them? 
3. If I know the power is going to be going out due to scheduled work, is there any issue with sending the `starttransfer` command proactively so that power is never lost? (ties into the first question a bit)

jgyates, 2023-10-12:
Hi @ssmoss 

In regards to your questions:

1) I don't know. That is up to the controller firmware and I have not tested these commands to that extent.
2) Since these are not documented commands by generac, I assume these commands are for testing purposes.
3) I don't know of any issues with doing that.

skipfire, 2023-10-12:
1. On my Evo2 controller I did tests with start+transfer and power outage and the generator ends up in an ON+ON state, the generator stays on until both the stop command is issued and the power comes back, sequence did not matter.
3. You can, but the generator will not shut off when the power comes back on if you haven't sent the stop command.

jgyates, 2023-10-12:
I am going to close this issue but feel free to post to this thread with any follow up.

closed 2023-10-12 by jgyates

ssmoss, 2023-10-12:
@skipfire I tried issuing the `starttransfer` and I believe that the generator only started but it never transferred. I was having connection issues to `genmon` so I want to test again. Out of curiosity are you running the same `hardware` and `firmware`  as me:
```
Hardware Version : V1.03
Firmware Version : V1.17
```

skipfire, 2023-10-12:
I am on older versions.  Does it still transfer with an outage?  I found an electromagnet wire popped off when I had a transfer not happen with a scheduled test, though it had worked a few days before during a real outage.  And just to confirm, you are on an Evo2 controller, correct?

Hardware Version : V1.01
Firmware Version : V1.11

ssmoss, 2023-10-12:
Yes I am using an EVO2 on a 22kW air-cooled. It doesn't transfer if It throw the main in the ATS. Although I have the wireless connected I have not configured it so unless the installers did something this is the version the unit came with.

skipfire, 2023-10-12:
It should transfer when you throw the breaker at the top of the ATS, if it isn't something isn't setup right.

ssmoss, 2023-10-12:
Ugh, sorry, it does transfer when I throw the breaker at the top of the ATS / MSP.

ssmoss, 2023-10-12:
I plan on testing again in a couple of days. I am wondering the fault usb wireless adaptor was causing issues. I really hope they didn't do something in the firmware to prevent this.

@jgyates is there debugging I can turn on to capture useless data during my test?

jgyates, 2023-10-12:
you could view the register page by clicking the gear in the upper right once. This allows you to view the registers as they are read. Other than that there is not a lot going on except the software writing the command.

ssmoss, 2023-10-30:
I have done two tests so far and it is working like it should. I think the initial trial failed due to some now resolved issue. Thanks.