jgyates/genmon #957: venv activate error on bookworm - fresh install

skipfire, 2023-10-14:
### Expected Behavior

Genmon starts without errors on a fresh bookworm install

### Actual Behavior

genmonpi@genmon:~ $ ./genmon/startgenmon.sh start
using binary: ./genenv/bin/python
Starting genmon python scripts
./genmon/startgenmon.sh: line 20: genenv/bin/activate: No such file or directory
sudo: ./genenv/bin/python: command not found
./genmon/startgenmon.sh: line 26: deactivate: command not found

### Steps to Reproduce (including precondition)

All steps performed morning of October 14, 2023
Installed a fresh copy of bookworm using Raspberry Pi Imager v1.7.5
Installed Log2Ram
sudo apt update -y && sudo apt upgrade -y
sudo apt install -y locales-all git i2c-tools cron git nano
git clone https://github.com/jgyates/genmon.git
cd genmon
bash ./genmonmaint.sh -i
Answered yes to all 3 prompts
No errors displayed during install
Rebooted, genmon did not start
Called startgenmon.sh start manually, received error above.

jgyates, 2023-10-14:
I don;t suppose you have the output of the install? You should have seen a message like this:

     "Managed system packages found, installing python virtual environment"

then it would install this:

     sudo apt-get -yqq install python3-venv

which should install the activate link. 

Let me try to reproduce later today.

skipfire, 2023-10-14:
I'll reimage and try to grab the output.  I did notice the output was much longer this time.

skipfire, 2023-10-14:
I found the problem, startgenmon.sh can only be called while in the directory.  You cannot call it from the root with `genmonpi@genmon:~ $ ./genmon/startgenmon.sh start`.

skipfire, 2023-10-14:
So it starts when I manually start it, but it isn't launching on boot, nothing in any of the genmon logs after the reboot; which appears to be an issue with crontab.  I found the following at the end of the install log.

```
sudo: ./genenv/bin/python: command not found
Finished setting up the serial port.
~/genmon
Done.
Start genmon on boot? (y/n)?y
Updating crontab....
no crontab for root
Updating crontab...
adding < @reboot sleep 30 && /bin/bash /home/genmonpi/genmon/startgenmon.sh start > to crontab
```

closed 2023-10-14 by skipfire

jgyates, 2023-10-14:
I think I know the issue here. The path to the activate / deactivate is relative. Let me fix this and test. I will let you know once it is checked in.

jgyates, 2023-10-14:
I checked in an update to genmonmain.sh and startgenmon.sh that should fix that issue. It also had the side effect of failing to run serialconfig.py during the install, but that is fixed too.Let me know how your testing goes.

skipfire, 2023-10-14:
1. Some updates need to made to the serial testing documentation about how to activate the environment to run the test.
2. For some reason my Pi 4 has a serial1 but no serial0, once I figured out activating the serialtest passed with serial1.  Once I updated the genmon.conf to serial1 it all starts fine, but no clue why I'm getting a serial1.

skipfire, 2023-10-14:
Also, doesn't the update process reset the entire genmon directory?  Will that not wipe out the genenv folder?

jgyates, 2023-10-14:
I will look at the serialtest app. Good point.

Regarding serial1 vs serial0, [this page has more info](https://www.raspberrypi.com/documentation/computers/configuration.html#configuring-uarts)

The short story it depends on the pi version and if bluetooth is enabled. With a pi4 if BT is enabled it will be serial0 if BT disabled it is serial1.

The update will not change any files that are not part of the repository so the genenv folder is left unchanged.

skipfire, 2023-10-14:
I'm surprised I haven't run into that before since I've only used Zero 2, Pi 3, and Pi 4 and always picked yes for the disable BT.  Any chance the install can make the decision between serial0 and serial1 since it is already asking about bluetooth?

Thanks for all the bookworm work.

jgyates, 2023-10-14:
The software can, I just don't know all the combinations. The info about a pi4 serial1 vs serial0 is empirical. I will look at the link and see if I can figure out the logic. serialconfig.py was designed to run without any of the needed python libraries since it is run during setup and the serial port is off by default os it has to know what the OS will do based only on the platform and if BT is enabled.

jgyates, 2023-10-14:
I added this wiki entry:

https://github.com/jgyates/genmon/wiki/Appendix-S---Working-in-a-Managed-Python-Environment

Both serialtest.py and modbusdump.py will show this link if you are in a managed environment and the libraries are not installed.

skipfire, 2023-10-14:
Would it be possible to have the code `ls /dev/serial*` and pick the first entry?  That would at least get it to a point where genmon loads instead of dying with an invalid serial port.  Another nifty feature would be if the config page serial port selection was a dropdown from that list or displayed a list of the serial ports next to it.

jgyates, 2023-10-14:
that is possible, but would need to add code to make sure that only ran on a raspberry pi (/dev/serialx is not used on all linux platforms), and also had a way to override it if people were using a USB serial port. 

First I am going to attempt to make serialconfig.py more predictable and print a suggestion on what it should be, do some testing than then possibly make it modify /etc/genmon/genmon.conf with the serial port name. 

Just playing around with a pi4 I can make it be either serial0 or serial one dependon on these settings:

# enable the onboard uart
enable_uart=1
# disable BT
dtoverlay=disable-bt
#make bt use mini uart instead of UART0
dtoverlay=miniuart-bt
# needed if you use the miniuart-bt setting
force_turbo=1

jgyates, 2023-10-15:
I checked in a change that I have tested on a pi3 and pi4 with bookworm that appears to consistently set /dev/ttyAMA0 to /dev/serial0