jgyates/genmon #958: Enhanced Exercise Suggestion

T-C123, 2023-10-14:
My Generac 18 kW generator is set to exercise for five minutes bi-weekly and does not have the option to change the time period. I think this is too short to get everything up to temperature and burn off any condensation in the oil. I have used the Genmon Enhanced Exercise add-on to extend the exercise period to about 15 minutes by restarting after the Generac exercise period has run. This has worked well and there is a noticeable difference in the clarity of the oil telling me that moisture is being removed.

My only problem with this system is coordinating the bi-weekly run times for the generator and the enhanced exercise periods. If I have to reboot the Pi, the two systems often get out of sync. This results in a normal exercise one week and the enhanced on the next, defeating my desire to run for an extended time.

The ability to set the Enhanced Exercise to run beginning "this week" or "next week" would help with this. I would have to figure out what schedule the generator was on and adjust from there.

A better solution would be the ability to select that the Enhanced Exercise add-on run immediately after a generator exercise run. Genmon would have to sense that an exercise period has ended (not an emergency or manual run) and start the Enhanced Exercise run.

I've seen a number of online posts complaining of Generac's short exercise period and think this would be a good solution.

Thanks very much for your consideration and continuing support!

jgyates, 2023-10-14:
I can add this to the feature list an update this thread when it is implemented.  I would be nice if there was a way to read the controller and determine how long until the next bi weekly exercise, but I don't know of a way to do that at the moment.

The "this week" or "next week" option would work, but seams a bit cumbersome as you would have to reset the value every time you restarted genmon. Possibly another option would be to have an option to run one minute after an exercise cycle is detected. This way you would not have to keep changing the setting. Let me know what you think about this option.

skipfire, 2023-10-14:
I have a 1 hour transfer test on the first of each month for burning off moisture from the oil and to ensure that the transfer is still working; I actually caught a transfer switch failure thanks to this rather than catching it in an outage.  My generator tech suggested running it under load once a month for 30 minutes to an hour for burning water off the oil, all this Louisiana humidity really encourages water in the oil.

T-C123, 2023-10-14:
The "one minute after" option would be perfect. That is basically what I do now except that I have to guess what week of the cycle I'm in. If Genmon could sense an exercise (not emergency or manual run) and then start the enhanced cycle, problem solved.

T-C123, 2023-10-14:
Good idea about the transfer test. I have a manual disconnect that makes it easy to simulate a power outage and test transfer and run. I do it occasionally on my own schedule so I can run it without upsetting the rest of the house.

jgyates, 2023-10-17:
I am going to close this thread for the time being. I will update in this thread when complete or if I have any questions.

closed 2023-10-17 by jgyates

jgyates, 2023-11-06:
The choices for Exercise Frequency are not Daily, Weekly, BiWeekly, Monthly and Post Controller. 

https://github.com/jgyates/genmon/blob/master/changelog.md

Let me know if you have any questions or issues.

eBoon123, 2023-11-21:
So does the Post Controller setting sense that a "normal" exercise is running and just take over and use the Enhanced Exercise arguments?

eBoon123, 2023-11-21:
OK - got it. Generator will finish the normal 5 minute exercise (and shut down) and then immediately(?) start the enhanced exercise. Wonderful! Thank you!