jgyates/genmon #959: Working towards using one of the DIY tank monitor solutions

Falcon-118, 2023-10-15:
I've been running GenMon for a couple of years now and am pushing towards running one of the DIY Tank Monitoring solutions for use with gentankdiy.py.    At this point I'm primarily looking to implement the solution posted by Mike/@zekyl314 running the ADS1115 there are some options I'd like to clarify.   I'm assuming the component labeled AD22100 is something like the Rochester angular position Sensor mentioned in the text??   If so, is there a preference on where the ADS1115 should be located?   Near the sensor or near the GenMon Raspberry Pi?  I ask because my cable run from the tank used for the Generator to the Raspberry Pi will be about 40-45ft.   So it's either a case of running a long 3 conductor cable to an enclosure (type-c conduit body?) with the ADS1115 then a short 6 conductor to the Pi or using a short 3 conductor to an enclosure with the ADS1115 and a longer 6 conductor to the Pi?   Not sure what the signal/strength looks like from the gauge sensor.  

I'd also like to clarify the drawing in regards to the line from SDA going to Pin3/GPIO2(sda) and there is no tie to 5v power on Pin4.

Any help appreciated!

Mike, the only contact I have for you was the twitter/X address and it appears you don't allow DMs?   If you'd like to discuss this in more detail offline, I'd be glad to route my contact info to you via Jason if that is appropriate?

zekyl314, 2023-10-16:
Hi @Falcon-118,

First, that is great you have been running this for year.  I am also very happy I found this project and was able to contribute to it.  

Yes, the component labeled AD22100 is the sensor that you connect / attach to the Rochester gauge.  I've found them on eBay a few times, I wish I could find a supplier and I'd look into selling a kit, but I haven't really spent time on that.  I placed my ADS1115 inside my generator, and then ran 1/2" PEX conduit from the generator to the tank.  I think my run is around 45' as well.  I used a CAT5 cable from the ADS1115 to the sensor wires.

You are correct for the pinout, PIN3 goes direct to the SDA, it is a bad route on the drawing.  If I still have the drawing, I will update for clarity.

Happy to continue to talk here, if you want a DM, I'll double check Twitter/X, I don't recall stopping DMs, but I really don't check there often.

-Mike

Falcon-118, 2023-10-16:
Thanks Mike for your response.   I too have had real problems getting any information about the Rochester sensor from anyone but the receptionist.  I can send you their list of distributors if you like, there are only 3.   (btw, if you do want to try to talk to someone about the sensors at Rochester, the guy's name is Romeo)  Fortunately, I have a family owned propane company whose service is A+.  I talked to them about a contact for the sensors and he has a couple of tank monitors dropped off by salesmen that were gathering dust on a shelf that have those sensors on them and I'm picking those up today.   

Thanks for clearing up the drawings and cable run for the sensor... I'll extend the sensor wire to the ADS1115. 

I held off on having a twitter/x account until just this summer... and the only reason I have one is to watch the SpaceX launches that Elon pulled from YouTube.   Communicating here is fine.

Thanks, again.
Rick

zekyl314, 2023-10-16:
Thanks for the info, as of now, I'll just leave it be, as I'm not sure how much interest there is in a package for the DIY sensor.

I also updated the wiring diagram and sent it over to @jgyates and he updated the wiki page already.  Thanks for pointing that out.

Good luck!
-Mike

jgyates, 2023-10-17:
I am going to close this issue but feel free to post to this thread with any follow up discussion on this topic.

closed 2023-10-17 by jgyates

Falcon-118, 2023-11-12:
Mike, Jason,
I had to spend a little more time with this than anticipated as I realized this really needed an Intrinsic Safety Barrier on the power circuit.   I thought about buying one from Stahl, but in the end, I just just built one after talking to the SME at Rochester Sensors.  New Rochester sensors with a 6ft lead are available from Squibb Taylor for $70.   

First run with the GenMon hat I had didn't work as there wasn't a good 5 vdc source on that board.  I fell back to a older unit and got a good signal.  A bit of tweaking on the config file and it works.   Thanks guys!!

[image]
[image]

jgyates, 2023-11-12:
Nice! Thanks for the update.