jgyates/genmon #960: Mopeka Sensor assitance

msteil4vse, 2023-10-18:
Hello,

Seeking, input from someone out there that is successfully using the Mopeka Pro+ sensors with Genmon, to compare notes with.

This issue is most likely outside the general scope of Genmon, I am intentionally keeping this short, so ask questions as needed.

General info is I have 2 independent systems so far on and attempting to get them to display data from Mopeka Sensors, without much success.  Each Pi/Genmon patform has been upgraded to the most recent code available, and the documentation has been followed to the letter.  The Genmon/Pi Platforms seem to be having trouble "Syncing" with the sensors or at least that is where things go off the rails with the documentation.   Ultimately the genmopeka.log log reports as follows:

2023-10-18 09:14:15,454 : starting scan
2023-10-18 09:14:30,456 : stopped scan
2023-10-18 09:14:30,459 : No sensor comms detected for BDAddress('F2:34:95:E4:B7:95')
2023-10-18 09:14:30,460 : Tank Data = {"Percentage": 0, "Tank Name": "Mopeka Sensor Tank"}
2023-10-18 09:14:30,463 : OK

Controller B8:27:EB:EE:51:B6 (public)
        Name: HeadendGenMon.vsenterprisesltd.com
        Alias: HeadendGenMon.vsenterprisesltd.com
        Class: 0x002c0000
        Powered: yes
        Discoverable: yes
        DiscoverableTimeout: 0x000000b4
        Pairable: yes
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v1D6Bp0246d0537
        Discovering: yes
        Roles: central
        Roles: peripheral
Advertising Features:
        ActiveInstances: 0x00 (0)
        SupportedInstances: 0x05 (5)
        SupportedIncludes: tx-power
        SupportedIncludes: appearance
        SupportedIncludes: local-name
[bluetooth]# 

sudo python3 mopeka_utility.py
sudo: unable to resolve host HeadendGenMon.vsenterprisesltd.com: Name or service not known

NOTE: This program will look for Mopeka Pro Sensors. The SYNC button must be pressed and held for the discovery process to work.

Starting Discovery....
Discovery Stats ReadStats ( Ignored Ad Count: 1, Processed Ad Count: 0, Error Ad Count: 0), Zero Data Ad Count: 0)

Finished Discovery.  Found 0 new sensor(s):

Now, I have a BLE scanner, and it sees the broadcast from the sensor, (see below) and the proximity of the devices is such there should be no RF issues.

1 address-type=random address=F2:34:95:E4:B7:95 name="F2:34:95:E4:B7:95" 
   persist=no mtik-key="" rssi=-41 
   last-data="0DFF5900086F420040E4B795F8DD0302E5FE" 
   last-seen=2023-10-18 09:19:17 beacon-types="" 

Lastly, the sensors are reporting that are using the 0.0.98 code.

Thanks to those who read this, hopefully someone has had this difficulty before and has the piece of the puzzle I seem to be missing.

contact me directly if you wish
msteil@trvnet.net

jgyates, 2023-10-18:
Hi @msteil4vse 

I don't use the mopeka sensor regulariy as my generator does not use propane, but based on the data you provide here are my thoughts:

One thing that is odd is that you get a name resolution error when running mopeka_utility.py.

     sudo python3 mopeka_utility.py
     sudo: unable to resolve host [HeadendGenMon.vsenterprisesltd.com](http://headendgenmon.vsenterprisesltd.com/): Name or service not known

I am not sure where this error is coming from or why it is there. It looks like it is coming from the sudo command.

Also, the genmon add on uses [this python library](https://github.com/spbrogan/mopeka_pro_check/), which has been abandoned by the developer. The mopeka_pro_check library only supports specific sensors. Looking at the code in the project here:

https://github.com/spbrogan/mopeka_pro_check/blob/62d9af0f38162370e05ba461e934123064a3fada/mopeka_pro_check/advertisement.py#L132

the project is looking for one of three specific IDs in the returned data and based off the data packet you posted it does not look like the sensor you are using is supported. Specifically, which mopeka sensor are you using?

msteil4vse, 2023-10-18:
Hi, and thanks for the response.

 

You are correct the host resolve issue is an issue in sudo, and my laziness.  I need to update some DNS records on the local network and the error will go away..  it’s on my to-do list.

 

The hardware I have is both a “Pro” and a “Pro+” the Pro is supported, and the only difference between the Pro and the Pro+ is an external antenna port on the + units. 

 

I am really suspicious that Mopeka has updated the firmware and the library needs to be tweaked, I have attached some documentation that Mopeka provided, and I was in hopes that someone would chime in as to the code revision, working sensors were in service.

 

Mark 

 

https://github.com/spbrogan/mopeka_pro_check/blob/62d9af0f38162370e05ba461e934123064a3fada/mopeka_pro_check/advertisement.py#L132

jgyates, 2023-10-18:
If you reply in email, github will strip off the attachments. you can attach them via the github web interface or email them to generatormonitor.software@gmail.com

jgyates, 2023-10-23:
I am going to close this issue. I have not received the docs you mentioned since github will not send email attachments. Feel free to send them when you can.

closed 2023-10-23 by jgyates

Raptor607, 2023-12-02:
Experiencing same thing here with a new Mopeka Pro Check Universal. Android BLE scanner picks it up, but none of the apps that have been previously reported to have supported that sensor are working (GenMon Mopeka Add-On, Home Assistant Mopeka, Home Assistant ESPHome integrations).  Should you receive the documentation mentioned by @msteil4vse and require any help testing, I'm happy to help!

jgyates, 2023-12-04:
Hi @msteil4vse 

I have two of the older sensors now. I have a propane tank, just not for my generator (I use diesel). This document looks like exactly what I need. I will let you know in this thread when I have something to test.

Thanks,
Jason

jgyates, 2023-12-05:
I checked in an update that allows the mopeka devices described in the doc to be used with the genmon add on. To use it update to the latest from the About page and follow the directions to [setup here](https://github.com/jgyates/genmon/wiki/1----Software-Overview#genmopekapy-optional). Let me know how it goes or if you have any issues.

jgyates, 2023-12-05:
Before the add on was using [this library](https://github.com/spbrogan/mopeka_pro_check/) to handle the discovery and collection from the sensor. The library only used device IDs 3,4 and 5. I remove the mopeka_pro_check library and created the same functionality but added more device IDs for all of the sensors listed in the above PDF (3,4,5,8,9,10,10,11,12).

Based on the capture above it looks like your device ID is 8 so if I am interpreting the capture correctly this will solve your issue. Let me know how you testing goes.

Raptor607, 2023-12-05:
Well done - I'm now receiving a reading from the sensor!  The value isn't completely accurate, but that's likely due to having moved it to the side of the tank for ease of reach while troubleshooting.  Thanks to you both!

msteil4vse, 2023-12-05:
> Can you tell me what’s specifically is different?  This seems to be the exact same procedure I have been following, many times over.  The train goes off the rails when you get to “sudo bluetoothctl” and the sensor MAC is never detected, and I can confirm with certainty from monitoring the Bluetooth returns from my KNOT that the data is being send and the sensor is in pairing mode.
>
>
>
> But I will try it again.

Qualified success…

 

Sorry for the delay, it snowed last night here, and my new knees don’t take kindly to getting down on the cold hard ground to push the pairing button on the sensor under the tank.  So had to solicit some youthful assistance.  

 

Before I started, on the web interface the fuel gauge indicated empty.  Did the procedure as outlined and now the gauge indicates full.   The tank is at 75% so basically full.  So my question is it fixed, or is the gauge now just stuck on full instead of empty?   If I recall isn’t there some sort of logging that can be turned on or viewed to see the data being  captured?  

 

Mark

msteil4vse, 2023-12-05:
Not relevant to the Mopeka sensors –  but did just notice that SNMP does not seem to be working on my units, have you had any inquiry’s since the release of V19, as it appears that it has been some time since there has been a response. 

 

***@***.***:~$ ping -c 10 192.168.30.31

PING 192.168.30.31 (192.168.30.31) 56(84) bytes of data.

64 bytes from 192.168.30.31: icmp_seq=1 ttl=64 time=1.08 ms

64 bytes from 192.168.30.31: icmp_seq=2 ttl=64 time=1.52 ms

64 bytes from 192.168.30.31: icmp_seq=3 ttl=64 time=3.54 ms

64 bytes from 192.168.30.31: icmp_seq=4 ttl=64 time=1.37 ms

64 bytes from 192.168.30.31: icmp_seq=5 ttl=64 time=1.15 ms

64 bytes from 192.168.30.31: icmp_seq=6 ttl=64 time=1.40 ms

64 bytes from 192.168.30.31: icmp_seq=7 ttl=64 time=1.48 ms

64 bytes from 192.168.30.31: icmp_seq=8 ttl=64 time=2.93 ms

64 bytes from 192.168.30.31: icmp_seq=9 ttl=64 time=1.15 ms

64 bytes from 192.168.30.31: icmp_seq=10 ttl=64 time=1.19 ms

 

--- 192.168.30.31 ping statistics ---

10 packets transmitted, 10 received, 0% packet loss, time 9014ms

rtt min/avg/max/mdev = 1.081/1.680/3.544/0.802 ms

***@***.***:~$ snmpwalk -v1 -c monitor 192.168.30.31 1.3.6.1.4.1.58399

Timeout: No Response from 192.168.30.31

***@***.***:~$

jgyates, 2023-12-05:
You can add the line 'debug=True' to the file /etc/genmon/genmopeka.conf to get extra output in the file /var/log/genmopeka.log. With this setting it will add lines for each BT packet received. Be sure to remove the line after you are done checking as this writes a lot of extra data. You have to restart genmon after each change to the conf file.

No reports of snmp issues. It is working for me. You can send your logs if you are having issues.

msteil4vse, 2023-12-05:
Hmm…  my bet is Gauge is stuck on Full, what do you think?

 

tail -f /var/log/genmopeka.log

2023-12-05 13:36:14,638 : Tank Type: 500_GAL

2023-12-05 13:36:14,639 : Error in GenMopekaData init: name 'MopekaBTSensor' is not defined : genmopeka.py:362

2023-12-05 13:40:47,877 : Tank Address = d1:49:fb:da:17:f9

2023-12-05 13:40:47,895 : min: 38.1 , max: None

2023-12-05 13:40:47,896 : Tank Type: 500_GAL

2023-12-05 13:40:47,896 : Error in GenMopekaData init: name 'MopekaBTSensor' is not defined : genmopeka.py:362

2023-12-05 14:26:00,977 : Tank Address = d1:49:fb:da:17:f9

2023-12-05 14:26:00,982 : min: 38.1 , max: None

2023-12-05 14:26:00,983 : Tank Type: 500_GAL

2023-12-05 14:26:00,983 : Error in GenMopekaData init: name 'MopekaBTSensor' is not defined : genmopeka.py:362

jgyates, 2023-12-05:
That is definitely a bug I introduced with the new code. The good news is that is and easy fix. I should be back home later this evening
 I will let you know when a fix is in.

jgyates, 2023-12-06:
I checked in a fix for this. This should do it for you. Let me know how your testing goes.

msteil4vse, 2023-12-06:
Hey, success on the Mopeka front!  Next I’m going to tempt fate, I have a site with 3 tanks, and tomorrow will order the sensors to outfit that site, so we will see if the multi-tank setup will work.By any chance do you know if there are any SNMP issues with v19?  I can start a ticket once I’ve gathered the info.  Just wondering.Sent from my iPad

jgyates, 2023-12-06:
you can look at the file /var/log/gensnmp.log for errors with the snmp add on.

msteil4vse, 2023-12-06:
Everything is looking really good, Below is a snip from the genmopeka.log.  I am wondering do you do anything with the sensor battery data?  If you don’t could I put a request in for enhancement were the battery info is displayed on the GUI and possibly some sort of alert when the battery get below say 20% or some reasonable number?

 

2023-12-06 06:19:50,754 : starting scan

2023-12-06 06:19:50,755 : Start Scan Filtered

2023-12-06 06:19:52,424 : Unsupported GAP report type 0x2 on sensor BDAddress('D1:49:FB:DA:17:F9')

2023-12-06 06:19:52,425 : data: ['0x1', '0x0', '0x1', '0xf9', '0x17', '0xda', '0xfb', '0x49', '0xd1', '0x12', '0xd', '0xff', '0x59', '0x0', '0x8', '0x6b', '0x26', '0x93', '0xc4', '0xda', '0x17', '0xf9', '0xcd', '0xf9', '0x3', '0x2', '0xe5', '0xfe', '0xc0']

2023-12-06 06:19:52,426 : Sensor Type: PRO+ Bottom-up Boosted BLE sensor, all commodities

2023-12-06 06:19:52,427 : Battery Volts: 3.34375 V

2023-12-06 06:19:52,428 : Battery Percent: 100.0%

2023-12-06 06:19:52,428 : Temp: 28.4 F

2023-12-06 06:19:52,429 : Reading MM: 536 mm

2023-12-06 06:19:52,430 : Reading inches: 21.1 inches

2023-12-06 06:19:59,429 : Unsupported GAP report type 0x2 on sensor BDAddress('D1:49:FB:DA:17:F9')

2023-12-06 06:19:59,430 : data: ['0x1', '0x0', '0x1', '0xf9', '0x17', '0xda', '0xfb', '0x49', '0xd1', '0x12', '0xd', '0xff', '0x59', '0x0', '0x8', '0x6b', '0x26', '0x93', '0xc4', '0xda', '0x17', '0xf9', '0xd2', '0x0', '0x3', '0x2', '0xe5', '0xfe', '0xb4']

2023-12-06 06:19:59,431 : Sensor Type: PRO+ Bottom-up Boosted BLE sensor, all commodities

2023-12-06 06:19:59,431 : Battery Volts: 3.34375 V

2023-12-06 06:19:59,432 : Battery Percent: 100.0%

2023-12-06 06:19:59,433 : Temp: 28.4 F

2023-12-06 06:19:59,434 : Reading MM: 536 mm

2023-12-06 06:19:59,435 : Reading inches: 21.1 inches

2023-12-06 06:20:02,938 : Unsupported GAP report type 0x2 on sensor BDAddress('D1:49:FB:DA:17:F9')

2023-12-06 06:20:02,939 : data: ['0x1', '0x0', '0x1', '0xf9', '0x17', '0xda', '0xfb', '0x49', '0xd1', '0x12', '0xd', '0xff', '0x59', '0x0', '0x8', '0x6b', '0x26', '0x93', '0xc4', '0xda', '0x17', '0xf9', '0xd0', '0x6', '0x3', '0x2', '0xe5', '0xfe', '0xbf']

2023-12-06 06:20:02,940 : Sensor Type: PRO+ Bottom-up Boosted BLE sensor, all commodities

2023-12-06 06:20:02,941 : Battery Volts: 3.34375 V

2023-12-06 06:20:02,941 : Battery Percent: 100.0%

2023-12-06 06:20:02,943 : Temp: 28.4 F

2023-12-06 06:20:02,944 : Reading MM: 536 mm

2023-12-06 06:20:02,944 : Reading inches: 21.1 inches

2023-12-06 06:20:05,756 : Stop Scan Filtered

2023-12-06 06:20:05,756 : stopped scan

2023-12-06 06:20:05,757 : Tank Level in mm: 536

2023-12-06 06:20:05,758 : Tank Level in inches: 21.1

2023-12-06 06:20:05,759 : Gallons Left: 297.05

2023-12-06 06:20:05,760 : Tank Percentage: 59.27

2023-12-06 06:20:05,760 : Tank1 = 59.27

2023-12-06 06:20:05,761 : Tank Data = {"Percentage": 59.27, "Tank Name": "Mopeka Sensor Tank"}

2023-12-06 06:20:05,763 : OK

msteil4vse, 2023-12-06:
Here is what I am getting back, in the log and I can confirm there is no SNMP response from a “get” request.

 

2023-12-06 06:27:58,825 : Error in SnmpCallbackFunction: <lambda>() takes 3 positional arguments but 4 were given : gensnmp.py:585

2023-12-06 06:27:58,826 : Error in SnmpCallbackFunction: <lambda>() takes 3 positional arguments but 4 were given : gensnmp.py:585

2023-12-06 06:27:58,826 : Error in SnmpCallbackFunction: <lambda>() takes 3 positional arguments but 4 were given : gensnmp.py:585

jgyates, 2023-12-06:
Your snmp issue is on that I have seen before in [this thread](https://github.com/jgyates/genmon/issues/938). It appears that this is an issue with the SNMP library and it's dependancies. The folks in this thread resolved their issues and the instructions they took are described in that thread. If you have any issues trying these things you can respond in the SNMP thread.

msteil4vse, 2023-12-06:
Thank you again, doing the “genmonmaint.sh -r” corrected the issue.

jgyates, 2023-12-06:
I forgot to mention, the notification of a low battery occurs if you enable the "Send Notices" setting on the Add On page. This will send an email if the battery percentage is below 15%. You outbound email must be setup on the main Settings page (email server settings) and Notifications page (for the destination email address).

msteil4vse, 2023-12-06:
Excellent, and thank you for efforts.

 

Do you have any documentation or other users that use more than one tank sensor?  I placed and order for more sensors today, and when they come I plan to use 3 sensors on one Genmon system and am hoping to get data about each of the 3 tanks, and ultimately monitor the tank levels be SNMP query.   All the single tank systems are working very well now.

jgyates, 2023-12-06:
Yes, you can search the issues on Mopeka and find other using this, some with multiple sensors. 

https://github.com/jgyates/genmon/issues?q=is%3Aissue+mopeka

I initially only supported one but due to user requests I added up to four sensors. Also, only one was supported via SNMP, but I updated the SNMP config today to allow it to report up to four tanls via SNMP. Let me know if you have any questions,

jgyates, 2023-12-06:
Update your software before trying to read 4 sensors via SNMP as this was just updated today.