jgyates/genmon #961: My Genmon Setup with Oil Pressure & Temperature Monitoring

T-C123, 2023-10-20:
My experience with Genmon began after purchasing a Generac 18kW, air-cooled, natural gas fueled, standby generator. I wanted to be able to monitor the operation of the generator and be alerted to potential problems before they became critical. The Mobile Link app that Generac offers is terribly inadequate and even the paid plans do not give the information that I wanted. Fortunately, I stumbled across Genmon (https://github.com/jgyates/genmon/wiki). In addition to what is offered in Genmon, I wanted to be able to monitor oil temperature and oil pressure. Oil temperature is important because it gives an accurate reading of the operating temperature of the air-cooled engine. Generac only includes a high temperature cut-off that shuts the generator down at 310° F. I want to know about a potential overheating condition before it gets to that critical point. The oil pressure is also important to know. High oil pressure could indicate clogged oil flow and low oil pressure could indicate low oil volume. Again, Generac only provides a low oil pressure engine cut-off which shuts down the generator at the critical point, no warning that problems are developing. I thought I’d describe my solutions to overcome these shortcomings and share my experience so others may benefit.

I am not electronics engineer or software writer. My coding days were 30 years ago and although I am an avid do-it-yourselfer, I am not a circuit designer. As someone else stated, I am a cook, not a chef.  All of the circuits that I used were adapted from what others have done and are just one way of accomplishing what I wanted. They may not be the best solutions and I give no guarantees that they will perform over the long run. Suggestions for improvements are always welcome. A big shout out goes to @jgyates for the countless hours he has put into developing and supporting Genmon. Also, thanks to @lmamakos for his work on a thermocouple interface board and @skipfire for the advice and handholding getting my system working. And of course, none of this would have happened without my friend Joe who introduced me to the Raspberry Pi, developed the oil pressure sensor circuit and wrote the code to get and display accurate readings.

My system is built around a Raspberry Pi 4 Model B running Debian v. 11, Bullseye software. The Pi is mounted in a modified aluminum case mainly to help support the interface connectors. The case is left open in order to allow better cooling. The assembly was then mounted on an aluminum plate for installation in an enclosed, steel box. The aluminum plate has a magnet attached to mount the Pi assembly in the steel box which allows easy removal for maintenance. This assembly is located inside a generator cover that requires tools to open so I included a remote, guarded power switch for the Pi so I can easily cut the power in case I need to reboot. The switch is easily accessible when the top of the generator is opened.

[image]

[image]

Communication with the generator is through a generator HAT from PintSize.Me using their data and battery power cables. I shortened the data cable to about 16” and installed a new Molex connector. The HAT came with extended GPIO pins so I could add another HAT on top with my home-brew circuits. It also has a built-in fan to provide CPU cooling. I didn’t think the fan was required but, since I planned to install everything in a small, enclosed box, I wanted to provide extra cooling if needed. The fan came set up to run continuously and I felt this was unnecessary so I added a circuit to control the fan with the Pi’s preferences setup. When the CPU gets above 75° C, the Pi turns on GPIO pin 18 triggering a transistor which controls the fan. 

I needed to use an external antenna outside of the generator’s metal enclosure for Wi-Fi connection so I disabled the on-board Wi-Fi and added a USB Wi-Fi adapter. Although I had a good Wi-Fi signal, occasionally I would lose my connection and it wouldn’t automatically reconnect. I tried a number of software suggestions that were supposed to force a reconnect but was not successful. I purchased a Raspberry Pi 4B with an external antenna connector from an ebay seller. No more Wi-Fi problems.

[image]

[image]

I also included a USB, Micro SD card reader so I could easily back up the Raspberry Pi’s software and operating system. An old laptop was converted to a dual boot Windows XP/Linux machine so I could communicate with the Pi using SSH or RealVNC remote access computing software across my home Wi-Fi network. The laptop also allows me to become more familiar with the new-to-me, Linux operating system leaving the Pi dedicated to the generator. Now I could run Genmon on the Pi inside the generator enclosure and control everything from indoors.

Next step was setting up temperature monitoring. @lmamakos has a great add-in board plan that permits multiple thermocouples (https://github.com/lmamakos/RPi-pHat-Thermocouple). Unfortunately, I did not feel comfortable assembling the tiny, surface mount components. I noticed that his circuit was based on MAX31850 thermocouple amplifiers so I purchased a preassembled amplifier that uses the MAX31850 chip from Adafruit along with a 3’ Type K thermocouple. Adafruit provides great documentation and links to all libraries needed to get things running. The thermocouple amplifier board was mounted on a piece of printed circuit prototype board along with a DS18B20 digital thermometer, analog to digital converters, and the circuit for controlling the fan on the PintSize.Me HAT. This board plugs into the extended GPIO connector on the HAT – kind of a double HAT. 

[image]

I covered the thermocouple cable with high temperature, heat shrink sleeving to add extra protection and added a thin coat of high temperature epoxy to the junction to prevent short circuits. I wanted to measure the oil temperature but not risk an oil leak so my first attempt was to thread the thermocouple down the oil drain tube. This worked great except that I found the oil sump provides cooling for the oil and does not indicate engine oil temperature accurately. Using an infrared thermometer, I found that the aluminum housing that provides a mount for the oil filter and oil pressure sensors transmits heat directly related to the engine oil temperature. In fact, this is where Generac connects their high oil temperature cut-out switch. Still leery of the possibility of creating an oil leak, I didn’t want to tap into the oil passages so I epoxied the thermocouple tip to a piece of glass which was then clamped to the aluminum housing. The glass (with the addition of some heat sink paste) provides good thermal transmission while insulating the thermocouple circuit from the engine ground. My oil temperature sensor now provides real time readings and could be adapted to measure other temperatures if desired.

[image]

[image]

The DS18B20 digital thermometer was added just because I could. While the temperature of the Pi enclosure is not very important, this device can be used to monitor any temperature from -67° F to 257° F. It is interesting to see on a freezing day, the Pi case maintains 70° F just from the heat of the CPU. The thermocouple and digital thermometer are enabled and configured via the Add-Ons page in the Genmon web interface.

Reading the oil pressure was the next project. I could not find information online by anyone who has done this but my research indicated that the Raspberry Pi along with an analog to digital converter was easily able to take an analog voltage and convert it to a computer friendly, digital signal. My friend Joe was sure he could write a Python program to take that information and output a pressure reading. Once again, I started at Adafruit and purchased 10-bit and 12-bit analog to digital converters. The 10-bit converter is just a chip which runs on the SPI bus. The 12-bit ADC circuit board uses the I2C bus and has amplifying capabilities. My initial experiments had problems with the I2C communications so I included both converters on my DIY board so I could have the option to switch between them. After encountering data interference problems on the I2C bus, my final version uses the SPI bus circuit and ignores the I2C data.

I felt it would be best to have the oil pressure sending unit isolated from ground so I chose a VDO #360-410, 0 – 80 PSI sending unit. This sending unit is teed off of the engine’s oil system where the factory, low oil pressure swi [... cut ...]
RP-SMA MALE RIGHT ANGLE to RP-SMA Female 36” RF Extension Cable 
Molex 8-pin connector with male & female housings w/pins - Mini-Fit Jr ™
Double Sided Prototype Printed Circuit Board

Grainger:

INSULTAB Heat Shrink Tubing, 0.13 in I.D. Item #4WND1
8” X 6” X 4” metal box Item #32FG50
(In order to fit in my generator, I had to trim 1” off the side of this box and weld it back together making an 8” X 5” X 4” box.)
https://www.grainger.com

Real VNC:

Remote access computing
https://www.realvnc.com

jgyates, 2023-10-20:
This is great info. I update the Adafruit part numbers so github would not insert links that were not relevant. I will look into the oil pressure info. That sounds very interesting.

lmamakos, 2023-10-23:
Wow, very nice work!  Glad you were able to repurpose some of my project like this.   I wonder if there's a thermocouple available that could be include into the same sort of arrangement that you used to tee off the oil pressure sensor; I suppose that oil isn't circulating and probably not representative.   In my case, I lacked the imagination that you used and I epoxied a thermocouple to a fin on the oil cooler heat assembly.  That's not nearly as accurate as what you're measuring, though it does serve the purpose of noticing when something goes horribly wrong and the temperature spikes up.

I do like the idea of that accessible power switch to reboot the Raspberry Pi; have to think about adding something like that the next time I have to get in there.

Thanks for sharing, @T-C123!

T-C123, 2023-10-24:
@lmamakos Glad you like my project! Some sort of sensor teed in where I installed the oil pressure sensor would work since oil circulates through the oil filter there. There are automotive oil temperature sensors that would work but, then I'd be re-inventing the wheel with the electronics. After working out the issues with the oil pressure system, I'm sure a similar circuit and software could be used to adapt an auto temp sensor to work. I started with your circuitry and inserting the thermocouple through the unpressurized oil drain line was a simple solution but didn't give responsive readings. I couldn't think of a way to adapt the braided thermocouple line to the pressurized oil passage without risking a leak so I opted for the external connection.

Jkh62, 2024-01-20:
Just ordered a pi and hat to build a system for my 22kw generac.
Did my 1st oil change this summer and lots of condensation / milky oil.
Knowing the oil temp is a great bit of info, especially to dial in the exercise time.
Its something I know I want to add if practical. Excellent work gentleman!

skipfire, 2024-01-20:
My generator tech recommended running it for 30 to 60 minutes once a month under load (the add-on schedules it) to burn off condensation in the oil, running the 5 minute weekly exercise isn't enough for that.

T-C123, 2024-01-20:
> My generator tech recommended running it for 30 to 60 minutes once a month under load (the add-on schedules it) to burn off condensation in the oil, running the 5 minute weekly exercise isn't enough for that.

That's good advice. It also forces you to test the transfer switch and identify any potential problems there.

skipfire, 2024-01-20:
I actually found that the coil wire had popped off and it didn't transfer because of that.

Jkh62, 2024-01-20:
WAY more diagnostic and feedback options than the generac app.
Running under load will give feedback on the entire system and take care of condensation in the oil. Water is the oil is not good and generac really is creating a problem by the short exercise time. My factory monitoring expires in April and I will have my hardware ready to plug it then. Fingers crossed they don't monkey with the firmware and I can pull the plug on their wifi.

Jkh62, 2024-01-21:
Downloaded some Generac pdf files on my 22KW model. Looking at the wiring schematic it lists two oil switches in circuit.
Low oil pressure switch (LOP) J1-19 and a high oil temp switch (HOT ) on J1-20 that runs to the controller. Both on the same ground.

The high oil temp switch caught my eye and was wondering if there is any actual feedback from the switch or is it simply open / closed from its set parameters ?

T-C123, 2024-01-21:
The switches on my 18 kW machine are simple open/closed, designed to shut down the generator when the parameters are exceeded. That is why I added temperature and oil pressure sensing to my machine.

Jkh62, 2024-01-22:
Adding a PI manual power switch to reboot the Pi is another good idea on the enclosure.
I picked up one of the $12 Harbor Freight cases to hold my Pi.