jgyates/genmon #963: WiFi off on new implementation of Genmon, Pi 4 B

daved44, 2023-10-22:
I have a new Pi 4 B that I am setting up for Genmon, thus far without a connection to the generator. I configured it headlessly using Windows Subsystem for Linux and SSH on a Windows 10 PC, exactly following the directions in  the 3.3 Setup genmon software section, while connected to the Pi ethernet port. It all worked flawlessly (considering my total lack of skills with Linux) and boots into the genmon software without problems. 

I can continue testing with the ethernet connection, but I need WiFi connectivity and the Pi WiFi does not show up in my local WiFi scan list (using the Advanced IP Scanner Windows application), and I have not been able to configure it. I set up WiFi initially in the Pi Imager, checking the "Configure wireless LAN" option, adding an SSID and a PW. Several online sites have been of some assistance, mostly with using tools like sudo nmtui, which is unable to activate any WiFi SSID I generate, and  sudo raspi-config, which does not display any SSIDs I set up and returns errors. Additionally, one site suggested editing or adding a boot level config file (don't remember the file name) with the SSID and PW of a local WiFi connection, which I did offline, i.e., with the SD card in a reader attached to the PC. The config file did not initially exist in the boot folder, and my additions appeared to disappear when the Pi rebooted. 

I am probably missing something essential, but perhaps there is a hardware problem with the Pi 4. Thanks for any assistance. 

### Expected Behavior
WiFi connectivity with a new unaltered Genmon configuration

### Actual Behavior
WiFi is dead on the Pi 4.

- Generator Model: 20KW Evolution 1, but N/A at this time
- Genmon Logs: N/A
- Genmon Version: latest as of 10/21/23

skipfire, 2023-10-22:
For the boot file, if it was the wpa_supplicant.conf it is normal that it is consumed upon boot, it is a one-time config file that is deleted after importing upon boot.

If you are connected by ethernet, try the command `ip address`, this will give you the Pi's view about it's network adapters.  You should see lo (local loopback), eth0, and wlan0.  The output for that may help determine what is going on and both adapters can be in use at the same time.

If you installed Raspberry Pi OS with GUI rather than Lite, you may want to try logging into the OS with a keyboard, mouse, and monitor attached and try configuring through the UI, there is a wireless icon in the top right corner.

If you installed Raspberry Pi OS Lite then you can try `sudo raspi-config`, then choose `System Options`, then `Wireless LAN`, you may be able to get it configured through that path.

jgyates, 2023-10-22:
also, when using raspi-config, in addition to enabling the wifi and setting the SSID and password as @skipfire mentioned, be sure to set the proper country code for the wifi, otherwise your access point may not be able to detect the pi's wifi.

daved44, 2023-10-22:
> If you are connected by ethernet, try the command `ip address`, this will give you the Pi's view about it's network adapters. You should see lo (local loopback), eth0, and wlan0. The output for that may help determine what is going on and both adapters can be in use at the same time.
> 
ip address returns the following:

1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether d8:3a:dd:3e:8f:dd brd ff:ff:ff:ff:ff:ff
    inet 10.0.0.213/24 brd 10.0.0.255 scope global dynamic noprefixroute eth0
       valid_lft 159365sec preferred_lft 159365sec
    inet6 2601:5ca:8100:2580::8318/128 scope global dynamic noprefixroute
       valid_lft 281472sec preferred_lft 281472sec
    inet6 2601:5ca:8100:2580:cef7:f2da:8723:86e2/64 scope global dynamic noprefixroute
       valid_lft 300sec preferred_lft 300sec
    inet6 fe80::9a67:d2b:b478:771a/64 scope link noprefixroute
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether d8:3a:dd:3e:8f:df brd ff:ff:ff:ff:ff:ff
    inet6 fe80::af93:bfb1:9f4a:36e2/64 scope link noprefixroute
       valid_lft forever preferred_lft forever

> If you installed Raspberry Pi OS with GUI rather than Lite, you may want to try logging into the OS with a keyboard, mouse, and monitor attached and try configuring through the UI, there is a wireless icon in the top right corner. I see you already tried through sudo raspi-config, so not sure what else to check on the Lite side of things.

I installed the full 64-bit version, released on 2023-10-10. But I don't have the equipment to run it with keyboard, mouse, and display; I cannot run it other than headless. 

I may image another SD card with the 64-bit full image, as I did with the other but without installing Genmon, leaving it as is and booting it up. This would eliminate anything I did during the Genmon install that might have fouled things up. I may not be able to get to the OS with the Windows Linux subsystem (or could I?), but if WiFi works in that mode I would be able to see it when I scan IPs.

Would configuring a vanilla Pi OS for this purpose be a good idea?

Also, is it necessary to "Configure wireless LAN" on the Settings Advanced options when imaging the Pi OS? Seems like specifying an SSID may get in the way of things. This is, after all, just another device that needs to connect to one of my already-existing WiFi SSIDs...

Many thanks for your help so far.

daved44, 2023-10-22:
> also, when using raspi-config, in addition to enabling the wifi and setting the SSID and password as @skipfire mentioned, be sure to set the proper country code for the wifi, otherwise your access point may not be able to detect the pi's wifi.

I specified US as the country in both the settings for the Pi OS imager and on the raspi-config screens.

skipfire, 2023-10-22:
Installing and configuring genmon should not impact wifi in any way.  Since it says the wifi is up, the most likely causes are the country selection needing to be set in raspi-config or SSID and/or key having a typo.  The already stored wpa config is in `/etc/wpa_supplicant/wpa_supplicant.conf`.  Here is what mine looks like for reference and my WiFi uses WPA2-Personal security, if you use WEP (you should upgrade security) or something else that could be part of the problem too as the config does differ.
```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
network={
    ssid="skipfirewifi"
    psk="somechars"
    key_mgmt=WPA-PSK
}
```

jgyates, 2023-10-24:
I am going to close this issue but feel free to post any follow up questions or status to this thread.

closed 2023-10-24 by jgyates

daved44, 2023-12-31:
Jon,
Not that it matters this time of the year, but I discovered yesterday that the PintSizeFanManager.service was not running because I had not changed the username from genmonpi to my username in the service (I must have reloaded the package sometime in the last few months and forgot to make that change).
After making the change yesterday, I decided to place the line “print(current_time + " " + str(temp) + " " + str(fanOn))” back into PintSizeFanManager.py, restart the service and look at the logfile, which eventually gave me the print output and assured me that it was running. It was, and I commented that line back out and restarted. 
But then I got to wondering why the other “print(“Fan on”)” and “print(“Fan off”)”, which remain in PintSizeFanManager.py do not also log. This may be a basic Python question, but I could not find the answer. Can you help?
Incidentally, as I learn my way through Raspberry Pi OS terminal, I have come up with other ways to get to some of this. First, I learned that I have to wait around 1.5 hours after restarting the service before journalctl will display logging from this latest restart, e.g., the output from that line I just put back in (due to delayed flushing?). Second, it is becoming evident that journalctl is an extremely useful program as, for example, I can use the --since operand to get to only the latest and all of the latest entries. Plus “ps -aux | grep python”, “ps -aux | grep FanManager” , etc. will return good information. The skills continue to build!
Finally, when I looked at your github add-on, it appears that you have changed the install location to /home/username/genmon-addon/FanManager and the PintSizeFanManager.py remains in that location for execution with the updated service moved as usual to /etc/systemd/system/. Is that so, or am I missing something? In any case I still run my .py directly from the username folder. Do I need to change?
Thanks again for your help.
Dave Dalton

skipfire, 2023-12-31:
I struggle my way through the python code, it's not one of my normal languages (I normally live in the C chain; over 20 years in C, C++, C#, Java, JS), and I have no clue why it logs some and not others, though if you don't use Log2Ram I suggest you not have a ton of log writing in place.

I don't know why journalctl would have a 1.5 hour delay, the only thing I can think of is if you have Log2Ram on there and maybe journalctl is only reading once it flushes out of Log2Ram.

The location did move, the GitHub repository was getting a bit messy and I had to organize it.  I actually hope to eliminate the service soon (mostly a documentation thing) in favor of raspi-config's fan settings.  I didn't know that option existed, but after testing the controllable fan circuit on the HAT is compatible.  It's in raspi-config under performance and uses GPIO 9.  Though the PintSizeFanManager would need to be disabled so they don't fight over the setting.

daved44, 2023-12-31:
Jon, thanks for the quick reply. I still do not know, however , why the "print" functions otherwise in the PintSizeFanManager.py do not go into the log. I suppose you didn't answer this because you don't know...

I got so enthusiastic about this whole Pi business that I gave my son a Pi 4 a few weeks ago and he has since become very interested in learning Linux. My reasons were somewhat selfish in that I was hoping that some of his new-found Linux skills, especially with terminal commands and Python, would rub off on me.   He is nearly 60 years old with many years of programming experience and has been messing with Android for as long as it has been around, writing apps and games for fun. But his primary skills, like yours, are in C, C++, Java, and beyond, and he told me yesterday  that he will probably do most of his Linux coding in C and C++, since that is where his skills lie, even though that code has to be compiled before use. Too bad for me...

I also bought a second 8GB Pi 4, and recently a third one, an 8GB Pi5 (it's a screamer, running browsers as fast as a Windows desktop!). So I am familiar with raspi-config on the Pi 4 (most of those configuration options are also in the GUI Raspberry Pi Configuration version). Those fan settings disappear on the Pi 5, since it has a dedicated fan connector and is set up for non-adjustable (as far as I can tell) multi-speed fan operation. But I suppose your hat fan on the Pi 5 could also be controlled from this Pi 5 connector.

Good to see also that you are shipping out your pre-configured units with the GUI OS. It's so much easier to do real work on the genmon Pi by getting to it through RealVNC (as it is on the other two, but my focus is still terminal mode for learning).

skipfire, 2023-12-31:
Yes, as in my reply "I have no clue why it logs some and not others".  It doesn't make sense that the same command would work sometimes and not others, but I have observed it as well.

As for the Pi 5, I would expect no one would use a fan on the HAT if they are using a Pi 5, they would just use the Pi 5 active cooler instead, and the next revision of board will see the fan opening moved to line up with the Pi 5's fan position (which will still work for the Pi 4) at which point I also hope to have optional riser kits available to allow airflow and prevent shorts due to how tall the heatsink on the Pi 5 is.

jgyates, 2023-12-31:
Not sure of the solution to the print as I have never run this code but this may be helpful:  

https://fhackts.wordpress.com/2014/11/27/systemd-journal-not-showing-python-3-print/

systemd should redirect your stdout and stderr but python may be getting in the way. Python may be trying to do some error handling since stdout and stderr are not owned by the console anymore (just a guess).

I had similar issues with genmon (genmon just runs as a background process, not a systemd service) at one time an the situation became problematic as some of the libraries being used (flask,etc) were using print and ended up breaking thread based functionality (e.g. the web server stopped working). I opted to redirect all output to a logging stream instead of to print and reduce the use of prints in genmon code to only before the logging functionality was setup. Now genloader.py loads background processes and redirects the stdout and stderr when each module is loaded.

https://github.com/jgyates/genmon/blob/17671b567764105ca3fcac1caa6a6c8c6d4138f3/genloader.py#L1108

daved44, 2024-01-01:
I did some testing this morning on the Pi 4 at my desktop, genmon and PintSizeFanManager installed (no HAT, of course).  After adding skipfire's print line  "print(current_time + " " + str(temp) + " " + str(fanOn))" to the fan manager py and restarting the service, it ran for 102 minutes until journalctl returned log activity from the time of starting. During that time, the following log entries occurred:

Jan 01 08:21:35 PlayPi systemd[1]: Stopped PintSizeFanManager.service - PintSize.Me Fan Service.
Jan 01 08:21:35 PlayPi systemd[1]: Started PintSizeFanManager.service - PintSize.Me Fan Service.
Jan 01 10:03:37 PlayPi python[4338]: 08:21:35 58.4 False
Jan 01 10:03:37 PlayPi python[4338]: 08:21:50 59.9 False
. . . more lines . . .
Jan 01 10:03:37 PlayPi python[4338]: 08:25:05 58.4 False
Jan 01 10:03:37 PlayPi python[4338]: 08:25:20 57.9 False
Jan 01 10:03:37 PlayPi python[4338]: 08:25:35 57.9 False
Jan 01 10:03:37 PlayPi python[4338]: 08:25:50 58.9 False
Jan 01 10:03:37 PlayPi python[4338]: 08:26:05 59.4 False
Jan 01 10:03:37 PlayPi python[4338]: Fan on
Jan 01 10:03:37 PlayPi python[4338]: 08:26:20 61.3 True
Jan 01 10:03:37 PlayPi python[4338]: Fan off
Jan 01 10:03:37 PlayPi python[4338]: 08:26:35 58.9 False
Jan 01 10:03:37 PlayPi python[4338]: 08:26:50 58.9 False
Jan 01 10:03:37 PlayPi python[4338]: 08:27:05 57.9 False
Jan 01 10:03:37 PlayPi python[4338]: 08:27:20 58.9 False
Jan 01 10:03:37 PlayPi python[4338]: 08:27:35 57.9 False
Jan 01 10:03:37 PlayPi python[4338]: Fan on
Jan 01 10:03:37 PlayPi python[4338]: 08:27:50 60.8 True
Jan 01 10:03:37 PlayPi python[4338]: Fan off
Jan 01 10:03:37 PlayPi python[4338]: 08:28:05 57.9 False
Jan 01 10:03:37 PlayPi python[4338]: 08:28:20 58.4 False

All the print functions, including the two imbedded in the if/elif stings, printed into the log. So I'm not sure the whole question of not all of the print() functions working is a problem, or perhaps it seems to work correctly intermittently. 

I also tried the suggestion from gyates's link above to use sys.stdout.flush() just after the print() functions and they failed with Python errors. I tried other suggestions from additional research online, including attempting to print to stderr, and all those failed as well.

The only question that remain is how to flush (write) print output to the SD card as needed or otherwise to get to recent Python script print() output before the system writes it out to the SD card, which I believe to be the problem. I know PintSizeFanManager is going away, but this questions is at a somewhat generic level.  Perhaps I will start a question on either the Raspberry Pi forum or on the Raspberry Pi section of stack exchange.

jgyates, 2024-01-01:
If you want to add this line from the above link:

    sys.stdout.flush()

you must also add this near the top with the other import statements:

     import sys

this will fix the python errors.

daved44, 2024-01-02:
Oops! Adding "import sys" fixed it. Well, it fixed it in that the last 10 fan manager py print() function outputs, with a flush statement below each, are included in the systemctl PintSizeFanManager.service status display.  But journalctl display is still empty. 

Yesterday I tried moving the PintSizeFanManager.py boot invocation to /etc/rc.local and removing the service extension from PintSizeFanManager.service. I then piped the PintSizeFanManager.py output to a local log file, updating with each 15-second run. That works well, as I can see easily see update activity in a variety of ways, in terminal or through the GUI. 

Before it warms up later this year I will be moving the pintsize HAT fan control to the Pi 4 control via raspi-config or the GUI configuration per skipfire's new fan implementation. But I don't plan for the PintSizeFanManager.py Python script to go away entirely, as I will probably alter it to return dynamic information on CPU temperature and fan status, plus placing log files on USB-attached storage. Remember, this is all for fun and learning! 

So unless you have any ideas about how to get journalctl or some other terminal command to return a full display of print() output starting right after execution, I will stop bothering you.  Thanks to both J&Js for the ongoing assistance!