jgyates/genmon #964: How to retrieve add-on parameters

eBoon123, 2023-10-23:
Just started using the Enhanced Exercise add-on and would like to get the setup parameters into Home Assistant somehow. Can you point me to where I can find those parameters? Is there an easy way to get those parameters into Home Assistant?

Thank you for GenMon - love it and it works flawlessly!!

eBoon123, 2023-10-23:
...and just found it - genexercise.conf. Any thoughts on how to monitor that file and push changes (or all params) on updates?

jgyates, 2023-10-23:
in python you can monitor os.stat(filename).st_mtime

in a shell script look at the output of "sudo tail -f /etc/genmon/genexercise.conf"

tail  will output the entire file if it is changed as the entire conf file is re-written. If you are looking for a diff output, that is doable, but more complex, you would probably need to make a copy of the file and diff them to get any differences.

closed 2023-10-23 by jgyates