jgyates/genmon #965: I have access to several generators.  Any interest in creating files for GenMon monitoring?

mstackler, 2023-10-24:
I stumbled upon GenMon while looking for better generator monitoring for our office locations.  Currently we are using Omnimetrix which is a cellular based monitor that we need to log in and get the data.  We are looking for something that feeds the data directly into a monitoring system and Genmon looks like it will work well with SNMP.  We have a few generac generators and i have built a raspberry pi and enclosure as detailed in the posts on this site.  I will be installing it tomorrow at the location for testing.  I would love to be able to convert some other generators that we have to use genmon as well.  I can provide all the data and testing if someone will help write the files.  Being new to modbus that is currently over my skill level.

I have access to the following generators

EDI Energy dynamics Inc 2MW using Buslar 2020 controller
Kohler using APM603 controller
Caterpillar unsure what controller

mark #

jgyates, 2023-10-24:
I can create the files, however I would need the modbus register definitions to be able to create the files. 

I found the Kholer AMP603 modus definitions online. 

Do you have any technical data on the modbus register definitions for any of these controllers? If you do you can email them to me at generatormonitor.software@gmail.com

The steps to make this work would be:

1) get the info on modbus registers on each controller
2) Setup a pi that can capture modbus data from each controller and then use this project's modbusdump.py program to get a capture of the modbus registers.
3) I will create the needed files based on the 1 and 2
4) You will test and I can make any needed changes based on the testing

jgyates, 2023-10-27:
@mstackler 

I am going to close this thread as we will communicate off line to get these working. If anyone else has interest in these models and willing to test please let us know in this thread.

closed 2023-10-27 by jgyates

mstackler, 2023-10-27:
Working on the basler units tomorrow. Will email you what I can gatherMarkSent from my Mark's  iPhone