jgyates/genmon #966: modbus register definitions for Evolution 2.0

pendlet, 2023-10-27:
Trying to see if I can get the mod bus registers for the evolution controller. I have a Generac generator in a remote location i am trying to bring back status via an RTU I have at the site. It will do modbus but i need the know all the registers to be able to do it. I tried searching the internet and through some of this code, but I wasn't able to I find a good list. Any help would be appreciated.

Thanks
Tim

liltux, 2023-10-27:
@pendlet it is in the generac_evolution.py file.

jgyates, 2023-10-27:
@pendlet 

This project was created without any help from generac documents so I can't really help in providing you any documentation other that the reverse engineered info in the source code:

@liltux  is correct in that the generac_evolution.py file has what you are looking for. Starting at this line of code: https://github.com/jgyates/genmon/blob/f75fe1cb8b5770839b01fa35554a68d6add8a43b/genmonlib/generac_evolution.py#L147

The base_registers and the prime_registers dictionaries are the registers that genmon uses to read the evolution and nexus controllers. If you have an Evolution 2.0 the link is encrypted so additional steps would need to be taken to communicate with modbus for this controller.

I am going to close this thread but feel free to ask specific questions in this thread if you have them.

closed 2023-10-27 by jgyates