jgyates/genmon #967: Pintsize generator HAT fan, size and getting it to work

daved44, 2023-10-30:
### Expected Behavior
PintSize.ME generator hat with cooling fan connected to HAT fan connector pins. 

### Actual Behavior
I purchased 25x25x10mm fan; wrong thickness, poor match of pins to HAT fan connector, and I could not get it to run. 

### Steps to Reproduce (including precondition)
I placed this issue here because skipfire's list of issues is empty.

Latest 64-bit Debian. Everything runs great, except the fan. Kudos to jgyates and skipfire! I'm so excited about the setup that I will probably move my weather station URL from an old Windows laptop to a Pi. Plus buy a Pi 5 just for fun and learning Linux generally. 

RPi is inside my 20K Evolution 1 generator, on an acrylic board affixed above the battery to the "firewall" near the outside with rubber coated magnets, enclosure not needed. Ethernet attached.  Testing showed that during a run with the ambient temp was in the 80s, the Pi CPU temperature remained in the 90s, but after shutdown when the engine heat entered the battery enclosure the temp rose to the 1-twenties. This isn't a problem now, but will be in the summer.  So a fan is necessary, although I have time to work through this.

I followed https://github.com/skipfire/genmon-addon instructions. It didn't work.  Then I brought the unit inside and worked through the GUI. I'm not sure where the PintSizeFanManager.py should be located, but if I remember right it was alone somewhere a folder or two off of the home folder, not in a folder with other .py files. (The unit is back in the generator, so finding it through the file manager is not possible and I'm not yet familiar enough with the Linux command structure to get to it otherwise.) The service is in /etc/systemd/system/ where it should be, although I believe it may have been preloaded there already? I'm also not sure it is started--how do I tell?

I will purchase a 7mm fan, hopefully one with a connector that fits without having to bend the HAT pins upward (any suggestions?), put it in, and with your help make sure that it runs when necessary by lowering the temperature in the PintSizeFanManager.py for testing. 

I don't believe that turning on External Temperature Sensors in the Add-ons is necessary?

The questions:

25x7mm fan necessary with your attaching screws?
Suggestions for a fan that best fits the connector?
Enabling External Temperature Sensors in the Add-ons is necessary?
Location of PintSizeFanManager.py
Displaying status of services...
Maybe skipfire can document the fan thickness needed, if preferably it is 7mm?

### Your Environment

- Generator Model: 20kW Evolution 1 Generac
- Genmon Logs: Submit via the About page
- Genmon Version: latest as of 2023 10 30

skipfire, 2023-10-30:
My first guess is it is because you are using Debian instead of RPi OS.  It is possible that the expected libraries for temp or IO control are not available, missing user permissions (I am not sure if Debian defaults users to have IO permissions), or possibly other causes.  The provided files were built for RPi OS and not tested elsewhere.

I will say that the fan will not help with the shutdown temperature spike.  My generator ran for 6 days after Hurricane Ida and temps on my Pi were fine until shutdown, at which point there was no air intake and the generator was radiating massive amounts of heat with no airflow to help cool things down.  I opened my lid for an hour so the engine could cool.  After my last outage when the power came back on I got my lid opened before the generator shut off and the Pi's temp never went up. 

The 7mm thickness is what the included screws will fit.  If you have taller screws a thicker fan can be used without a problem.  I will update the product information on my site to specify the height to work with the provided screws.

I never found a fan I was really happy with, they all seemed to die.  The initial fan used header pins and the pins did not need to be bent at all, the more readily available fans with the locking connector do require them to be bent up to about 30 degrees from the board, that will not hurt the pins or the board, just be gentle when lifting it so you don't damage any of the traces, there is a trace under the white ground line.

Everything after this assumes you have a v2.3 or higher board, prior to that the fan pins are always on.

Enabling the add-on in GenMon is optional and unrealted to the fan's function.  The fan's function in the provided code files is based on CPU temperature.

PintSizeFanManager.py ends up in the user directory of the file system on the pre-loaded installs, and for self-installs it will go to wherever you were in the file system when you executed the wget commands (by default you start in the user directory).  I did just find a gap in my instructions (which I will update), the service file has a hardcoded username that matches what is used on the pre-loaded units, but would need to be updated for manual installs on RPi OS and definately on Debian.  Line 7 would need to have `genmonpi` replaced with whatever the username is you have for logging in.  Below is what that line should be if you made your username `dave`.
`ExecStart=/usr/bin/python /home/dave/PintSizeFanManager.py`

Service status can be seen with `sudo systemctl status PintSizeFanManager.service` and with all the information I've put above, I'm guessing it is not running due to the username issue I mentioned, then possibly permissions and library issues.

daved44, 2023-10-31:
I updated the PintSizeFanManager.service file to the username of my user directory (which is where the PintSizeFanManager.py file had landed when I fetched it). I then rebooted the Pi and checked that the service is active. I also ordered a couple of 7mm thickness fans and changed the temperature in the .py file to 35 C, which is 95 F, which should give me the opportunity to observe whether it starts in the off state when I activate the Pi  and turns on at higher temps. 

I will update this issue when I get the fan installed. That should be soon as long as Amazon doesn't lose the fans, which they are apt to do with small items. Thanks, as usual, for your assistance. 

Dave

jgyates, 2023-10-31:
I am going to close this issue for now, but feel free to use this thread for updates and follow up questions.

closed 2023-10-31 by jgyates

daved44, 2023-11-01:
The fans arrived; I took the unit inside and installed a fan, then hooked up keyboard, mouse, and display, and booted it. It did not work. 

Some observations: 1) One time when booting the fan spun up to a healthy rpm, but stopped at what appeared to be when the boot finished. So I know the fan functions. 2) While messing with the unit trying to get it to work across several reboots, the temperature climbed up to the mid 140s; it usually runs in the mid 1-teens or low 120s.  3) I'm running with a 5V power supply through the USB connector on the Pi. The unit runs in this configuration with the Buck switch in either position. (The GENPWR switch is always set to OFF, inside or outside in the generator, as I have an Evolution 1.)

Questions: 
I presume that it is not necessary to have the unit connected via the Molex plug to the generator while I am trying to get this fan to run?
Will the position of the various red switches have any effect on the operation of the fan?

I'm attaching some screen shots that may help.
[image]
[image]
[image]

skipfire, 2023-11-01:
That temp is really high even without a fan, do you have heatsinks installed?

None of the 5 switches impact the fan operation.

The brief fan run was probably the pin temporarily having a state set during boot and then it went to floating later in the boot process, that does at least indicate the hardware should all be good.

Next 2 things to try would be calling `vcgencmd measure_temp` to ensure that returns a valid value, second would be to manually run the script and see if any error messages come out `python3 PintSizeFanManager.py`.  You can also check for installed gpio libraries `find /usr | grep -i gpio`.

daved44, 2023-11-02:
I ran the find command this morning and there are several hundred entries with gpio in them. I then started the PintSizeFanManager.py script per your instructions and it ran and continued to run, cycling on and off around the 35 C temperature I had the script set to. It displayed many "Fan On", "Fan Off" messages during the half hour or so that I let it run.  It continued to cycle after terminating the script with a ^C command. Is there a clue here??

I then rebooted the Pi and the fan ran for perhaps 30-40 seconds, starting well after boot appeared to finish (the GUI had appeared on the display). Then it stopped, the temp rose to about 43 C, and nothing happened. 

I then again started the script manually per your instructions and it cycled normally again, but the fan stopped after terminating with a ^C command. After rerunning the script again and again terminating it, the fan continues to cycle, as it did earlier this morning. 

Something wrong with the service? I manually run the script with python3; the service appears to use python (FWIK about this, which is not much!).  Here's the latest activity, after which the fan continues to cycle.

david@raspberrypi:~ $ sudo systemctl status PintSizeFanManager.service
● PintSizeFanManager.service - PintSize.Me Fan Service
     Loaded: loaded (/etc/systemd/system/PintSizeFanManager.service; enabled; preset: enabled)
     Active: active (running) since Thu 2023-11-02 08:12:03 EDT; 2min 44s ago
   Main PID: 1783 (python)
      Tasks: 2 (limit: 1578)
        CPU: 178ms
     CGroup: /system.slice/PintSizeFanManager.service
             └─1783 /usr/bin/python /home/david/PintSizeFanManager.py

Nov 02 08:12:03 raspberrypi systemd[1]: Started PintSizeFanManager.service - PintSize.Me Fan Service.
david@raspberrypi:~ $ python3 PintSizeFanManager.py
Fan on
Fan off
Fan on
Fan off
Fan on
Fan off
Fan on
^CTraceback (most recent call last):
  File "/home/david/PintSizeFanManager.py", line 26, in <module>
    time.sleep(15);
    ^^^^^^^^^^^^^^
KeyboardInterrupt

FWIW, I do have heat sinks installed. The tiny fan and heatsinks appear to bring the temperature down several degrees very quickly.

skipfire, 2023-11-02:
What about the output of vcgencmd measure_temp?

daved44, 2023-11-02:
Jon, that output was normal, matching the GUI and Status monitor
temperatures. I'll be away today, so probably won't be getting back to you
until this evening.

skipfire, 2023-11-02:
To your earlier question of there being something wrong with the service I've not had any problems myself nor have any other users reported problems, though I suspect everyone has used it on Raspberry Pi OS and not Debian.  Perhaps move the temp down to 10 and see if it turns on and stays on, if the Pi is holding around 35 it's possible that the fan running for a few seconds brings enough cool air in to get below the temp and then a short period later it heats up just enough to be over the temp.  Changes to the code do require restarting the service or stopping the service and running manually.

daved44, 2023-11-02:
That the fan is starting and stopping on a regular basis to keep the temperature around 35 is not the problem. In that regard it is working exactly like it should. The problem is that the fan doesn't get started at all with boot, or perhaps it does gets started, as I have reported that it appears to from time to time (and this would likely be because the initial CPU temperature at boot is above 35) but it stays off once it goes off. 

The only way I get it to run and cycle on and off as it should is to start the fan script manually as you suggested by running the following: python3 PintSizeFanManager.py.  When I run that command, the fan begins cycling on and off, keeping the temp right around 35, and continues doing so until the next boot. 

So it really does look like the fan service is not starting the script, or that the script doesn't continue running after it is started.

skipfire, 2023-11-02:
Your screenshot indicates it does start and isn't failed out.  It is probably something with Debian vs Raspberry Pi OS, GPIO permissions or user spaces would be my two guesses, but I'm not sure how to confirm/reject either of those as causes.

daved44, 2023-11-03:
It appears that we have run into a wall with this problem. It needs to work before I put the unit back into my generator. To potentially clear things up, I am documenting the problem with a step-by-step description. 

1. Reboot from a working system where the fan control was cycling as it should (from previous actions on my part to make it work, much as I will be describing below) and the temperature was around 35 (C). The temperature of the CPU, now ambient, went up a bit after the Pi was disconnected, so at restart will be slightly above the 35 level.
2. 32 seconds to appearance of the GUI, at which time the fan starts running. It stops about 10 seconds later while the temp is still above 35, maybe 38-40. It does not start again. 
3. Run  python3 PintSizeFanManager.py, which starts the script in interactive mode in my terminal window. The fan turns on, brings the temp down to below 35 in 2-3 minutes, turns off, turns on again as the temperature climbs above 35, etc. "Fan on" and "Fan off" messages are being printed to the terminal console. The fan is running exactly as it should, keeping the temp around 35. 
4. ^C to kill the interactive script. This may be important: It appears that if I do the ^C while the fan is off, it never turns on again. If I do the ^C while the fan is on, it continues to cycle as it should. Too much testing time involved to determine for sure whether this is so; perhaps the problem is that killing the script and having the fan continue to cycle, or not, may be an intermittent problem with another cause. 
5.  Start the script again, and the fan again cycles properly.
6.  Display the status of the service at any time, and it looks normal:
david@raspberrypi:~ $ sudo systemctl status PintSizeFanManager.service
● PintSizeFanManager.service - PintSize.Me Fan Service
     Loaded: loaded (/etc/systemd/system/PintSizeFanManager.service; enabled; preset: enabled)
     Active: active (running) since Thu 2023-11-02 08:12:03 EDT; 2min 44s ago
   Main PID: 1783 (python)
      Tasks: 2 (limit: 1578)
        CPU: 178ms
     CGroup: /system.slice/PintSizeFanManager.service
             └─1783 /usr/bin/python /home/david/PintSizeFanManager.py

You often mention the possibility that the latest Raspberry Pi OS I am running may be the problem. I could relatively quickly install the version you use for your prebuilds and test with it. While this will take a while, it certainly will not take as much time as we have spent trying to figure this out and, if it works, we could move forward from that point; if it doesn't, more effort required. 

But I'm not sure which OS release would be the one you are using, since all of the Raspbian versions have been renamed.  I install from the  Windows Raspberry Pi Imager v1.7.5, and there are many images I can use. Looking at the Raspberry Pi OS - Wikipedia description, I may be able to make an educated guess. But I would rather get this from you. But first try to make further headway with this problem from my detailed description above. If no success, if you would, run Raspberry Pi Imager v1.7.5 and/or look at the Wiki and let me know the exact image you use.  I have spare microSD cards available.

skipfire, 2023-11-03:
Your initial post said you were running the latest Debian 64-bit, did you install Debian or Raspberry Pi OS?  There are a number of differences between the two.  There are other differences between Raspberry Pi OS Bullseye (pre-October 2023) and Bookworm (October 2023 and on).  If you are using Bookworm then I will do specific testing on that over the weekend.

daved44, 2023-11-03:
Jon, sorry for any misinformation. I installed the recommended image, Raspberry Pi OS (64-bit) (described as a port of Debian Bookworm) from the Windows Raspberry Pi Imager.  This is the only image I've installed on the Pi. I see now why you may have been reluctant to look further. Thanks for continuing your effort to make this work.

daved44, 2023-11-05:
Update.  Yesterday I unplugged The Pi and left it until this morning, at which time I started it up again. The fan ran properly! I left it for an hour or so and the fan cycled regularly, with no problems. 

I then shutdown (unplugged the power cable) and started it up again.  The fan problem was back.  So this is an intermittent problem and because I can manually start the fan script and it works properly every time, it is not a hardware problem. So I executed the following:  sudo systemctl restart PintSizeFanManager.service. It worked! I then executed the stop function, then the start function. It continues to work correctly. I did this several times.

So what is this? A timing problem? Can the initial fan service start function executed at boot time be delayed for a minute or so? Might that fix the problem?

skipfire, 2023-11-05:
What you describe is a possibility, if the fan service starts up and turns on the fan, then something else turns the pin off, the service would not turn it back on as it thinks it already is as the script only sets the pin when it should change.  There are 2 straightforward solutions.  One is to add a sleep command near the top of the script to make it wait for 60 seconds, but I think the better solution is re-setting the pin every time it checks in case some other service tries to clear it beyond boot time.  I hadn't been able to replicate the behavior you are getting, but I'll make an update for the service.  It will be available later today.

skipfire, 2023-11-05:
@daved44 below is an updated script that sets the fan state every time it comes through, not just when it notices it should change.  That should help with any race conditions.  I also added some output logging that every cycle (15 seconds) it will print the time, temp, and expected fan state.  That should be visible with systemctl status for the last few entries, or journalctl to view more.

```
import RPi.GPIO as GPIO
import os
import time

onTemp = 40 #Temperature in Celsius

fanIo = 9
fanOn = False

GPIO.setmode(GPIO.BCM)
GPIO.setwarnings(False)
GPIO.setup(fanIo,GPIO.OUT)
GPIO.output(fanIo,GPIO.LOW)

while True:
    cpu_temp = os.popen("vcgencmd measure_temp").readline()
    temp = float(cpu_temp.replace("temp=", "").replace("'C", ""))
    t = time.localtime()
    current_time = time.strftime("%H:%M:%S", t)
    if temp < (onTemp - 1):
        GPIO.output(fanIo,GPIO.LOW)
        if fanOn:
            fanOn = False
            print("Fan off")
    elif temp >= onTemp:
        GPIO.output(fanIo,GPIO.HIGH)
        if fanOn == False:
            fanOn = True
            print("Fan on")
    print(current_time + " " + str(temp) + " " + str(fanOn))
    time.sleep(15);
```

daved44, 2023-11-05:
I have tested over several reboots, pulling the plug, shutdown then pulling the plug. It looks like we have arrived. It is working every time.

I would like to display the output logging you added, but I can't find the systemctl options to make that happen. I did it yesterday, based on a tutorial I took online, but can't find that either.  Displaying the status of the service does not give any information aside from date and time it was started. Can you help?

skipfire, 2023-11-05:
I'm doing this from memory, but `systemctl status PintSizeFanManager`.

daved44, 2023-11-05:
Jon, sorry, I asked for info on the wrong command. Systemctl  is the status command I run all the time and, as I stated above, it gives me no additional information. 

I'm looking for the source and/or filtering options of the journalctl command to get to all of the logged info on the running PintSizeFanManager script. That, I believe, will give me all the info logged with the fan script (unless I was getting it through some other command structure yesterday...)

skipfire, 2023-11-06:
@daved44 I believe `journalctl -u PintSizeFanManager -e` should get it.

daved44, 2023-11-06:
If it matters... But it does, as I would like to monitor the temperature and fan on/off status when the unit is back in the generator. 

I can display the journalctl entries for the fan manager, but the additional  output logging that you added, including fan on/off entries, haven't appeared since early this morning before I rebooted. I changed nothing. Here's the last of the journalctl output.

The portion at top that displays the additional logging was the end of thousands of lines of output before the reboot. I ran it just a few minutes ago with:
journalctl -u PintSizeFanManager --since=2023-11-06.
The last entries were the result of me restarting the service.
As you can see, none of the additional output happened after a couple reboots.

. . . 
Nov 06 06:39:58 raspberrypi python[1267]: 06:37:58 34.0 False
Nov 06 06:39:58 raspberrypi python[1267]: Fan on
Nov 06 06:39:58 raspberrypi python[1267]: 06:38:13 35.0 True
Nov 06 06:39:58 raspberrypi python[1267]: 06:38:28 34.5 True
Nov 06 06:39:58 raspberrypi python[1267]: Fan off
Nov 06 06:39:58 raspberrypi python[1267]: 06:38:43 33.6 False
Nov 06 06:39:58 raspberrypi python[1267]: 06:38:58 34.5 False
Nov 06 06:39:58 raspberrypi python[1267]: Fan on
Nov 06 06:39:58 raspberrypi python[1267]: 06:39:13 35.5 True
Nov 06 06:39:58 raspberrypi python[1267]: 06:39:28 35.0 True
Nov 06 06:39:58 raspberrypi python[1267]: 06:39:43 34.0 True
Nov 06 07:45:46 raspberrypi systemd[1]: Stopping PintSizeFanManager.service - PintSize.Me Fan Service...
Nov 06 07:45:46 raspberrypi systemd[1]: PintSizeFanManager.service: Deactivated successfully.
Nov 06 07:45:46 raspberrypi systemd[1]: Stopped PintSizeFanManager.service - PintSize.Me Fan Service.
Nov 06 07:45:46 raspberrypi systemd[1]: PintSizeFanManager.service: Consumed 35.839s CPU time.
Nov 06 07:45:46 raspberrypi systemd[1]: Started PintSizeFanManager.service - PintSize.Me Fan Service.
-- Boot 0437c04da66e40008f97db78b177d538 --
Nov 06 07:55:28 raspberrypi systemd[1]: Started PintSizeFanManager.service - PintSize.Me Fan Service.
Nov 06 08:08:42 raspberrypi systemd[1]: Stopping PintSizeFanManager.service - PintSize.Me Fan Service...
Nov 06 08:08:42 raspberrypi systemd[1]: PintSizeFanManager.service: Deactivated successfully.
Nov 06 08:08:42 raspberrypi systemd[1]: Stopped PintSizeFanManager.service - PintSize.Me Fan Service.
-- Boot 590fb0650ce4482d9df7a47868aa2bfc --
Nov 06 08:08:58 raspberrypi systemd[1]: Started PintSizeFanManager.service - PintSize.Me Fan Service.
Nov 06 08:26:55 raspberrypi systemd[1]: Stopping PintSizeFanManager.service - PintSize.Me Fan Service...
Nov 06 08:26:55 raspberrypi systemd[1]: PintSizeFanManager.service: Deactivated successfully.
Nov 06 08:26:55 raspberrypi systemd[1]: Stopped PintSizeFanManager.service - PintSize.Me Fan Service.
Nov 06 08:26:55 raspberrypi systemd[1]: Started PintSizeFanManager.service - PintSize.Me Fan Service.

skipfire, 2023-11-06:
I was going to recommend removing the logging so that you don't have it eating up write cycles on your SD card.  Is it functioning even though it isn't logging?  I don't know why the behavior would be changing with reboots if there were no other changes.

daved44, 2023-11-06:
That's probably a good idea, although I'm not too concerned about the health of the SD card currently running the OS. It's an old slow one and I will be updating. I have a couple options, including running a script to poll the temperature from time to time, with a longer sleep delay between polls. I suppose I could have the script generate an error if the temp exceeds a certain value. Got any ideas? I'm using the Pushover add-on in Genmon and if I can get the script to generate an error, Pushover would send an error notification to my phone. Genmon may even send an email?

I could also increase the delay in your fan manager script to a minute or so and remark out the additional logging; the only detrimental effect this longer delay would have is that the temperature would cycle at higher and lower temps than the set temp in the script. 

The unit is back in the generator and running. It can stay there, as the hardware is solid and any changes can be made from a terminal on my Windows desktop. And the logging output started again. Who knows? In any case, this thing is nice. Thank you for all your help.

skipfire, 2023-11-06:
I don't think the timing interval makes much difference, commenting out the last print line is what I'd suggest.  I have a very custom setup for temp monitoring along with everything else, so I'm not sure there.  I built my own home automation system 20 years ago (before IOT was a thing) and I've just been revisioning, so I'm not familiar with other options that may be in place for collecting that data.  It would be nifty to be able to have genmon alert on it when it goes past defined thresholds.