jgyates/genmon #968: Enhanced exercise would not shut down automatically

T-C123, 2023-11-02:
### Expected Behavior

Enhanced exercise would run and shut down automatically.

### Actual Behavior

Did not shut down.

### Steps to Reproduce (including precondition)

System was running in enhanced exercise. I had to reboot the Pi for unrelated reasons. At the end of the exercise period, the generator kept running. I had to manually stop it from the "Maintenance" tab. I understand that this probably happened because Genmon restarted, not knowing that enhanced exercise was running, and had no reason to stop the generator. Just wondering if there is a fix for this since it could unknowingly cause the generator to run continuously if a power glitch caused the Pi to reboot while exercising.

Genmon would have to sense on restart that the generator had been left in exercise mode and then shut down but, I wouldn't want it to shut down if it was running in emergency operation.

Thanks!

### Your Environment

- Generator Model: Generac 18 kW
- Genmon Version: V1.18.18

jgyates, 2023-11-02:
I have some changes planned for the enhanced exercise so I will revisit this at the same time. If memory serves the issue is that if the enhanced add on is restarted it can not tell if it was started by the add on or some other reason if you are using a full (non low speed exercise).

https://github.com/jgyates/genmon/blob/master/feature_requests.md

closed 2023-11-02 by jgyates

T-C123, 2023-11-02:
@jgyates  If it makes a difference, the generator was exercising at reduced speed (1800 RPM) when this happened.
Thanks for your response!

jgyates, 2023-11-06:
I added code to issue a stop of an exercise cycle is occurring when the program starts. Let me know if you have any issues:

https://github.com/jgyates/genmon/blob/master/changelog.md

In theory this could stop the standard exercise cycle if is occurring, but that is a low probability. there is also the possiblity that this will not stop a controller initiated exercise cycle as the controller may only allow a remote stop command to stop another remote start or remote exercise command.