jgyates/genmon #969: activation hours and run time during outages

hacs-home-assistant-plugin-github, 2023-11-06:
I see that genmon understands register 0054 (Hours since generator activation (hours of protection), but I can't find the MQTT topic that shows this value.  Is it available and I'm just missing it?  If not, can it be added?

I'd like to have an idea of how many of the total run hours were actually used when providing power due to an outage.  The hours since activation can get me close, given a standard 5 min exercise weekly.

ie. run_time_during_outages=total_run_hours-((activation hours / 168) *5 / 60)

In my case, this would work out to:
16258 hours / 168 hours per week = 96.77 weeks of activation
96.77 * 5 mins = 483.87 mins of exercise time
483.87 mins / 60 = 8.06 hours of exercise time
47 hours total run hours - 8.06 hours exercise time = 38.94 run time during outages
which is pretty close to my actual manually recorded times.

It would be really nice if genmon could do this math and provide the result as an MQTT topic as well.  Maybe the parameters could be configurable for those who might have different exercise schedules.

Thank you

jgyates, 2023-11-06:
The hours since activation is only available over modbus on liquid cooled evolution controllers. Unfortunately that data is not accessable to genmon on air cooled models. On liquid cooled models it is exported via MQTT. In short, if you can see the data in the web interface it should be available on MQTT as genmon exports data in JSON format for both of these sources of data.

the exercise duration is also only available on the liquid coolerd model over modbus. 

Let me know if you have any other queastions.

closed 2023-11-06 by jgyates

hacs-home-assistant-plugin-github, 2023-11-06:
Maybe I'm looking at the wrong thing, but in the little tool icon in the upper-right corner where it takes you to the page with all of the registers on it, I see Hours since activation, register 0054, and it is being updated appropriately on the web page.  See attached picture.  This is the value I'd like to see a topic for if possible.

[image]

jgyates, 2023-11-06:
I am making an assumption that you have an Evolution 2.0 controller. Is that correct?

this register (0054) only exported zeros on all models but the liquid cooled controller, when I initially wrote that code (hence the EvoLC in the description), but if you can confirm that it is changing once an hour on an Evolution 2.0 I can add that register as an output for Evo2.0.

the way the code works is that for all evolution and nexus controllers it reads the same set of registers, some of them contain valid data on Evolution air cooled 1.0, some on Evolution air cooled 2.0, some on Evolution liquid cooled, etc.. There are about  8 different controllers that are in the evolution/nexus group so I will need you to validate your controller before I add this to the code.

Regarding performing the calculation, that might be possible for some models, not all models have the exercise duration available over modbus.

hacs-home-assistant-plugin-github, 2023-11-06:
I do indeed have the Evolution 2.0 controller:
Controller Detected : Evolution 2.0, Air Cooled
Hardware Version : V1.01
Firmware Version : V1.11
and I see the value in 0054 changing once/hour.

For models that don't have the exercise duration on modbus, maybe it could be a configurable item that could be manually set?

jgyates, 2023-11-06:
I went ahead and tested this with a set of registers from Evo2.0 and checked in an update. You can update from the About page to get this update. The hours of protection would appear on the Maintenance page under Maintenance -> Controller Settings -> Hours of protection so the MQTT would be /generator/Maintenance/Controller Settings/Hours of Protection  

The software detects the controller and exports functionality based on the controller type so no need for an additional setting. genmon now reports Hours of protection for Evo Liquid Cooled and Evo 2.0 Air Cooled models.

The exercise cycle duration does not appear to be available over modbus (register 023e) for Evo 2.0 (only Evo LC).

Let me know if you have any questions.

hacs-home-assistant-plugin-github, 2023-11-07:
Perfect.  I now see the item on the GUI as well as the MQTT feed.  Thank you!