jgyates/genmon #970: New Feature Request

mstackler, 2023-11-06:
New Feature Request - Kiosk Mode

I have been using the SSH modes admin = full access and monitor = view only

Was wondering if it were possible to have a 3rd option to view only the gauges.  Was thinking of installing a tablet on the wall to replace the status panel with the led lights.  Would like to only see the gauges to make it easier for an office person or security guard.

I can see in this 3rd level having
Battery Voltage
Voltage avg
Current avg
Frequency
RPM
Coolant Temp
Power Output
Fuel Level
Oil Temp
Key Switch status off/auto/run
Runtime calculator gauge that can update automatically based on load and fuel level
and maybe even the genmon generator symbol that changes green to red when generator is running

At the bottom of the gauges maybe something simple like 
Building on Utility - if everything is operating normally
Building on Generator - outage
Generator in maintenance - key turned to run
Generator offline - key on off

mark

skipfire, 2023-11-06:
Do you have other dashboards at your site?  I have a Grafana dashboard that includes environmental displays, door/window warnings, etc, and has a section for generator data.

jgyates, 2023-11-06:
I have grafana on my list of things to try. Once that happens (after my pi5 arrives and testing occurs) I may post some notes about genmon integration or possibly some automated import. In the interim I will add this to the feature list, but a grahana based solution may be the path that gives the most flexibility in displaying the data.

https://github.com/jgyates/genmon/blob/master/feature_requests.md

I will update this thread when any progress is made.

closed 2023-11-06 by jgyates

mstackler, 2023-11-07:
I was thinking of using grafana to create a dashboard to see all of my
equipment at once.  This would be something on the wall in the building
lobby.

mark

snakeeyes5, 2023-11-07:
You can use Telegraf to ingest the Genmon JSON into Influx. Then it’s easy to create Grafana dashboards from there.