jgyates/genmon #971: stupid me, i descided on updating to latest version on my genpi, now nothing works

grsthegreat, 2023-11-08:
i looked thru previous issues, and followed the instructions
i went thru putty and logged into my pi via ssh

typed 

cd genmon
sudo ./genmonmaint.sh -u

entered Y when asked

then typed

./startgenmon.sh restart

still not working

all i get is a clocking icon and the message LOADING GENERATOR MONITOR

i have no idea what to do now.

skipfire, 2023-11-09:
I'm sure @jgyates will reply, but until he does, you may want to try `sudo ./genmonmaint.sh -i` to ensure all the libraries are on it and then `sudo ./startgenmon.sh restart` again.

dennis48755, 2023-11-09:
Same with me.

jgyates, 2023-11-09:
I don't suppose you know what version of genmon you were upgrading or if you can post what messages were output to the screen when updating and restarting genmon? That info would help.

Depending what version you were upgrading from this note may apply to you:

https://github.com/jgyates/genmon/issues/920

@skipfire is right about the commands he suggested. Let me know how that works. If it does not be sure to include the output of any commands you run as this helps me determine what is working for you and what is not. 

Do you have a file named /etc/genmon/update.txt? Type this command to see and post the output:

    sudo cat /etc/update.txt

If you have this file it will let me know how long ago your last update was. 

Also what is the output of this command:

     grep 'version =' /etc/genmon/genmon.conf 

The output of the file /var/log/genmon.log may help also.

    cat /var/log/genmon.log

jgyates, 2023-11-09:
also, check if a reboot of the pi will get you running again.

dennis48755, 2023-11-09:
I was at v1.18.18 although I believe you made an update or two within that version since I previously updated.

I did a sudo reboot. I thought I could no longer log in with putty, but while I was typing this comment I see the password prompt finally popped up. Very slowly.

Response to grep 'version =' /etc/genmon/genmon.conf is...
version = V1.18.18

Response to cat /var/log/genmon.log is attached.
[2023-02-10 111142 Genmon Log.txt](https://github.com/jgyates/genmon/files/13302544/2023-02-10.111142.Genmon.Log.txt)

Genmon GUI is not working.

Dennis

dennis48755, 2023-11-09:
Putty seems to be responding faster now.

I tried /home/pi/genmon/startgenmon.sh start
and get the response... 
pi@raspberrypi:~ $ /home/pi/genmon/startgenmon.sh start
Starting genmon python scripts

This program requires the modules located in the genmonlib directory in the gith        ub repository.

Please see the project documentation at https://github.com/jgyates/genmon.

Error: invalid syntax (mycommon.py, line 41)

Dennis

jgyates, 2023-11-09:
@grsthegreat 

Let me know the above when you have the time.

@dennis48755 

Your genmon.log has a couple of things that are problematic.

1) Genmon checks if a software update is avaialbe every 24 hours. The code that checks this is putting this error message in your log: 

     Error checking for software update: <urlopen error get         random() initialization failed. (_ssl.c:720)> : genmon.py:1393

This is the network library failing to create a random number. This error has occured *every day* since July 10, 2023.

2) The 2nd issue I see is this error:

     Error in GetRaspberryPiTemp: [Errno 12] Cannot allocat         e memory : myplatform.py:174

This is the genmon getting the data to display in the web interface and the operating system is failing to allocate memory. This error occured on the 11/5/2023 and not other logs entries have occurred since then. 

Both of these are very fundamental errors that to me indicate a probem with the pi. My guess is that:

  a) another installed app is using too much memory (doubtful but possible)
  b) your pi is malfunctioning
  c) your SD card is corrupt or failing

The last error that you are reporting on load is genmon not even loading (hence not errors from today in your log). This would only occur if your system or file system is corrupt.

When you load genmon and get the error in your last reply (Error: invalid syntax), this indicates that genmon will not even run due to a problem with the source file, which backs up the theory about your SD card failing. You can run a file system test on your SD card. Typically you have to do this on a reboot (https://forums.raspberrypi.com/viewtopic.php?t=8451) or remove the SD card and plug it into another linux system and run fsck (https://linux.die.net/man/8/fsck). You can not run this on a file system that you are running the operating system as the file system needs to be read only.

The short answer is this: based on your log it looks like you pi started having issues around 7/10/2023, you restarted your software on 9/23 but errors still occured daily. The last update the genmon directory is not able to load anything. At this point is is probably easer for you to just save the files in /etc/genmon/ (if they are not corrupt) and start with a fresh SD card and if you still have an issue then replace the pi.

dennis48755, 2023-11-09:
Since I was an early adopter of your genmon (Feb. 2018), a faulty SD card seems likely to me.

I have been thinking of asking about how best to create a new system card or copy my system to a new card.

I guess this is a good time to ask for your suggestions.

Dennis

grsthegreat, 2023-11-09:
now when i type anything into putty, i get command not found
[image]

grsthegreat, 2023-11-09:
i tried rebooting my genmon unit, still not working

grsthegreat, 2023-11-09:
ok, i closed and reopened putty, redid the sudo ./genmonmainy.sh -u

this time it worked. when i went to reboot, i got this message.

by the way, i also updated from 18.18

[image]

jgyates, 2023-11-09:
@dennis48755 

My recomenations are use the raspberry pi imager to write the new SD card (https://www.raspberrypi.com/software/). 

the wiki has a page on tips and tricks that has useful info (https://github.com/jgyates/genmon/wiki/Appendix-M-Tips-and-Tricks-useful-for-Genmon). I would highly recommend using log2ram (1st one on the list) as it extends the life if your SD card. If you are OK with the command line / ssh then use the lite version of the operating system with a headless (no graphical use interface or keyboard required). You can use raspberry pi imager's advanced options to setup and default user and password and enable ssh so you can just flash the SD card, plug it in and log in over the network.

grsthegreat, 2023-11-09:
ill have to look into this when i have more time and patients. i wish i never updated this thing

jgyates, 2023-11-09:
I think I found an issue that my be common to both of you.

It looks like the code I added for the new version of the Raspberry Pi OS abound a month ago is not compatible with python 3.5

If you type this:

     python3 -V

what is your output?

@grsthegreat 

The file /var/log/genmon.log may provide some history on your system as to why you are seeing these issues. Can you show the output of this file? or attached it to this thread?

    cat /var/log/genmon.log

will display the log contents.

In your first screen cap it does not look like you are in the genmon directory. I think

dennis48755, 2023-11-09:
My output is; Python 3.5.3

Dennis

jgyates, 2023-11-09:
ok. I checked in a file that fixes the issue with python 3.5

in the genmon folder, update the software with:

      ./genmonmain.sh -u

then restart

      ./startgenmon.sh restart

Let me know how it goes.

@dennis48755 this may fix your problem, but you still have some issues with your system from the log file, but check your log in a day or so if you are still getting issues after a reboot. If you dont see any new errors after 48 hours you may be OK.

grsthegreat, 2023-11-09:
Mine is version 3.5

grsthegreat, 2023-11-09:
i tried adding ./genmonmain.sh -u

says, no such file

[image]

jgyates, 2023-11-09:
sorry, typo:

      ./genmonmaint.sh -u

jgyates, 2023-11-09:
The change I checked in works for me with Python 3.5. Before the change I got the same results when testing with this version of python. 

As a point of reference. Python 3.5 was installed with Raspbian stretch (v9) which is from 2018. The raspberry pi foundation does not produce any bug fixes for Raspbian stretch so even if this fixes the issue for you, it is probably a good idea to backup your genmon conf files via the web interface, grab a new SD card and reinstall the OS with the latest version (bookwork v12). With open source software, keeping up with the latest os updates often ensures things don't break when upgrading other software. No a requirement, but just my two cents of advice. You can keep that version until it does not work and upgrade then also.

I keep one pi around with that version of the OS on it for testing, but I will admit that it is tested last and not every time as most of the testing goes into the newer versions of the operating system.

grsthegreat, 2023-11-09:
THAT FIXED IT

jgyates, 2023-11-09:
great! thanks for bringing up this issue. I am going to close this thread but feel free to post any follow up.

@dennis48755 let me know if you have any issues in this thread as well.

closed 2023-11-09 by jgyates

grsthegreat, 2023-11-09:
I noticed a message about not being able to do further updates with python 3.5 in future. to update the python, is a new pi required, or is this just a software issue?

and thank you for your help

dennis48755, 2023-11-09:
Mine is still updating.

Dennis

jgyates, 2023-11-09:
To update, you would need to install the operating system again, you can use the same pi. While not required, a new SD card may be helpful in the long run. If you have been running this one for 5 years it may be worth the hassle to just get a new SD card, keep the old one for a backup. You can just install the new version of raspberry pi OS on the new SD card, then pop it in the pi and let it boot, install genmon , then copy the backup files to /etc/genmon

This page describes how to save and restore your config files to a new pi: https://github.com/jgyates/genmon/wiki/Appendix-M-Tips-and-Tricks-useful-for-Genmon

genmon will still work with python 3.5. the operating system updates will not be issues as bugs are fixed. 

@dennis48755 if you  update is taking that long you may have two issues, 1 being fixed by the software update and the 2nd is a SD card failure. you had two real issues in your log file so you may not be able to resolve this until you install a fresh version of the OS on a new SD card. Let me know if you have any questions.

grsthegreat, 2023-11-09:
thank you for your help with this issue

jgyates, 2023-11-09:
to clarify, it is just a software issues, no need to buy a new pi. The operating system you are running is 5 years old and the people who maintain it do not supply updates any more for that version. The latest version is 12, you are running 9.

dennis48755, 2023-11-09:
Below is the output from my putty session.
[image]
I still can't get to the GUI but I did receive emails that the monitor is starting.

Tomorrow, I think I will put my effort into replacing the SD card.

Dennis

jgyates, 2023-11-09:
all of that looks good.

If you like you can submit your log files this way with the ClientInterface.py program and the "generator: sendlogfiles" command. Here is the out put:

     pi@genmon:~ $ cd genmon
     pi@genmon:~/genmon $ python3 ClientInterface.py 
     OK : Auto, Off - Ready
     >generator: sendlogfiles
     Log files submitted
     >exit
     pi@genmon:~/genmon $

dennis48755, 2023-11-09:
I just submitted the logs. My system appears very slow, took a minute between prompts.

jgyates, 2023-11-09:
Some of your other log files have errors that are similar to the ones listed above, also your serial log has some errors from 11/5/2023 and earlier that indicate some issues:

Resetting SerialDevice:ReadThread Error: /dev/serial0:read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?) : myserial.py:166

I would continue with the reinstall on a fresh SD card. That is your easiest path back to normal at this point. My guess is that you have a failing SD card and the operating system is trying to make the best of it but the retries and errors are causing the slow down.,

dennis48755, 2023-11-09:
Thanks Jason, I'm working on a new SD card now. I'll open a new thread if/when I need help.

Dennis

capindc, 2023-11-09:
it looks like I am have a problem with V19 I was on 1.18.18 
Python3 is Python 3.10.7
python3 ClientInterface.py  Genmon is not running
can you walk me trough where the log would be. I have check /Var/log the logs in there are old

Thank you again

jgyates, 2023-11-09:
@capindc 

You will need to give more details on the error you are seeing. You said it is not working but did show any output or error message.The main log is in /var/log/genmon.log

capindc, 2023-11-09:
There is no errors on start. every looks fine but nothing at the IP address and when I run ClientInterface.py I get genmon is not running 
[genmon.log](https://github.com/jgyates/genmon/files/13312304/genmon.log)

jgyates, 2023-11-09:
I don't have enough info to help you. I will need the commands you are running to start genmon and ClientInterface.py. also need the output of your /var/log/genmon.log file

Copy and paste the commands with the output or a screen capture would give more details of what you are doing.

Do you get errors when genmon is restarting? From the genmon folder type:

     ./startgenmon.sh restart

capindc, 2023-11-09:
genmon@generator:~/genmon$ ./startgenmon.sh restart
Restarting genmon python scripts
[sudo] password for genmon: 
Stopping....
Stopping genemail2sms.py
kill: (2139): No such process
Stopping gentankutil.py
kill: (2136): No such process
Stopping genserv.py
kill: (2134): No such process
Stopping genmon.py
kill: (1831): No such process
Starting....
Starting /home/genmon/genmon/genmon.py 
Starting /home/genmon/genmon/genserv.py 
Starting /home/genmon/genmon/addon/gentankutil.py 
Starting /home/genmon/genmon/addon/genemail2sms.py 

genmon@generator:~/genmon$ python3 ClientInterface.py
Genmon not loaded.
genmon@generator:~/genmon$ 
[genmon.log](https://github.com/jgyates/genmon/files/13313430/genmon.log)

jgyates, 2023-11-09:
@capindc 

the log file helped a lot. I found the issue. This was related to a change I made for the handleing of serial parity on custom controllers. I corrected the bug in the code and it should work now so you can update from the command line in the genmon folder like this:

      ./genmonmaint.sh -u

Then select 'y' for yes when prompted. Once this is complete you can start genmon as you normally would:

     ./startgenmon.sh restart

Let me know how it goes.

capindc, 2023-11-09:
Wow its up I did the update like you said above it looked good. Thanks again.

jgyates, 2023-11-09:
Great. I am glad it is working for you now. Thanks for reporting this issue.

dennis48755, 2023-11-10:
I am up and running on a new SD card.

Thanks Jason for all your help and suggestions.

I did install the Log2Ram and the backup / restore saved the day. It even turned on the couple Add-Ons I use (External Temp & Current Sensors).

Dennis

jgyates, 2023-11-11:
Glad to hear it. Thanks for the update.