jgyates/genmon #972: unable to upgrade to latest version, need help

mac1-131, 2023-11-10:
don't seem to be able to upgrade to latest version

about, upgrade to latest version seems to go through the motions, says it is downloading, and then restarts and makes me log back in, but still on old version

try to do it manually, fails with this message:

fatal: not a git repository (or any of the parent directories): .git

thanks, any help is welcome. old version still working fine so not critical.

jgyates, 2023-11-10:
What version are you trying to upgrade from? 

Can you post the exact command and output when you run manually? 

Have you upgraded using this method before? Have you moved any directories or files?

This message is from the 'git' command and it indicates that the files used by git to update the github project are not found.

Sending your logs may help also.

jgyates, 2023-11-10:
I received your log files. I think I know the source of the problem. 

accourding to your logs this command is run on your system to update the software:

     /bin/bash /home/pi/genmon-master/genmonmaint.sh -u -n -p 3

This looks like you have installed the project by downloading a zip file instead of cloning the repository via git since you are using a 'genmon-master' folder instead of just 'genmon'. This will work however you will not be able to update your software unless you use the 'git clone' method. This page has more details one how to setup.

https://github.com/jgyates/genmon/wiki/3.3--Setup-genmon-software

If you want to transition to the recommended setup then do the following:

1) from your /home/pi folder type the following:

      git clone https://github.com/jgyates/genmon.git

This will download the repository along with the needed git folders into the /home/pi/genmon folder
 2) modify your crontab entry to use the genmon folder version instead of the genmon-master version. to do this type:

      sudo crontab -e

You may be prompted to select and editor if you have not run crontab befre. If you do select the 'nano' editor, or one of your liking.

Once the crontab file comes up for editing change this line:

   @reboot sleep 30 && /bin/bash /home/pi/genmon-master/startgenmon.sh start

to this:

   @reboot sleep 30 && /bin/bash /home/pi/genmon/startgenmon.sh start

3) then restart genmon by executing this from the genmon folder.

     cd /home/pi/genmon
     ./startgenmon.sh restart

All of your setting should remain as they are stored in /etc/genmon

The above will put you on the latest version and allow you to update future versions.

jgyates, 2023-11-10:
I am going to close this issue but feel free to post any follow up status or questions to this thread.

closed 2023-11-10 by jgyates

mac1-131, 2023-11-10:
OK thanks I will work on your suggestions, I thought it might be something like that. thanks

mac1-131, 2023-11-12:
this seemed to work fine, thanks. Is it OK to delete the now unused (I think) genmon-master folder?

thanks again

Brian

jgyates, 2023-11-12:
yes, if you changed your crontab you can delete that with no issues.

mac1-131, 2023-11-12:
yes I did, and it came right up on new version, so will delete the other folder

thanks