jgyates/genmon #973: I broke Genmon...updated ubuntu to 23.04 and Python got updated to 3.11.4

capindc, 2023-11-10:
Sorry for braking it
I was having issues with my VNC connection a known problem with Trunas VM and Ubuntu I read that if you up grade to 23.04 fixes it. Yes my VNC is fixed but my Genmon is broke
how screwed am I?

[genmon.log](https://github.com/jgyates/genmon/files/13322069/genmon.log)

 
Python is now 3.11.4
genmon@generator:~/genmon$ ./startgenmon.sh restart
using binary: /home/genmon/genmon/genenv/bin/python
./startgenmon.sh: line 20: /home/genmon/genmon/genenv/bin/activate: No such file or directory
Restarting genmon python scripts
[sudo] password for genmon: 
Warning: required library flask not installed. Attempting to install....
Error in ExecuteCommandList  : b'/home/genmon/genmon/genenv/bin/python: No module named pip\n'
Installing python3-pip
Error in InstallLibrary using pip : ['/home/genmon/genmon/genenv/bin/python', '-m', 'pip', 'install', 'flask']: b'/home/genmon/genmon/genenv/bin/python: No module named pip\n'
Warning: required library pyserial not installed. Attempting to install....
Error in InstallLibrary using pip : ['/home/genmon/genmon/genenv/bin/python', '-m', 'pip', 'install', 'pyserial']: b'/home/genmon/genmon/genenv/bin/python: No module named pip\n'
Warning: required library crcmod not installed. Attempting to install....
Error in InstallLibrary using pip : ['/home/genmon/genmon/genenv/bin/python', '-m', 'pip', 'install', 'crcmod']: b'/home/genmon/genmon/genenv/bin/python: No module named pip\n'
Warning: required library pyowm not installed. Attempting to install....
Error in InstallLibrary using pip : ['/home/genmon/genmon/genenv/bin/python', '-m', 'pip', 'install', 'pyowm==2.10.0']: b'/home/genmon/genmon/genenv/bin/python: No module named pip\n'
Warning: required library pytz not installed. Attempting to install....
Error in InstallLibrary using pip : ['/home/genmon/genmon/genenv/bin/python', '-m', 'pip', 'install', 'pytz']: b'/home/genmon/genmon/genenv/bin/python: No module named pip\n'
Warning: required library pysnmp not installed. Attempting to install....
Error in InstallLibrary using pip : ['/home/genmon/genmon/genenv/bin/python', '-m', 'pip', 'install', 'pysnmp']: b'/home/genmon/genmon/genenv/bin/python: No module named pip\n'
Warning: required library ldap3 not installed. Attempting to install....
Error in InstallLibrary using pip : ['/home/genmon/genmon/genenv/bin/python', '-m', 'pip', 'install', 'ldap3']: b'/home/genmon/genmon/genenv/bin/python: No module named pip\n'
Error in InstallLibrary using pip : ['/home/genmon/genmon/genenv/bin/python', '-m', 'pip', 'install', 'pyasn1', '-U']: b'/home/genmon/genmon/genenv/bin/python: No module named pip\n'
Warning: required library smbus not installed. Attempting to install....
Error in InstallLibrary using pip : ['/home/genmon/genmon/genenv/bin/python', '-m', 'pip', 'install', 'smbus']: b'/home/genmon/genmon/genenv/bin/python: No module named pip\n'
Warning: required library pyotp not installed. Attempting to install....
Error in InstallLibrary using pip : ['/home/genmon/genmon/genenv/bin/python', '-m', 'pip', 'install', 'pyotp==2.3.0']: b'/home/genmon/genmon/genenv/bin/python: No module named pip\n'
Warning: required library psutil not installed. Attempting to install....
Error in InstallLibrary using pip : ['/home/genmon/genmon/genenv/bin/python', '-m', 'pip', 'install', 'psutil']: b'/home/genmon/genmon/genenv/bin/python: No module named pip\n'
Warning: required library chump not installed. Attempting to install....
Error in InstallLibrary using pip : ['/home/genmon/genmon/genenv/bin/python', '-m', 'pip', 'install', 'chump']: b'/home/genmon/genmon/genenv/bin/python: No module named pip\n'
Warning: required library twilio not installed. Attempting to install....
Error in InstallLibrary using pip : ['/home/genmon/genmon/genenv/bin/python', '-m', 'pip', 'install', 'twilio']: b'/home/genmon/genmon/genenv/bin/python: No module named pip\n'
Warning: required library paho-mqtt not installed. Attempting to install....
Error in InstallLibrary using pip : ['/home/genmon/genmon/genenv/bin/python', '-m', 'pip', 'install', 'paho-mqtt']: b'/home/genmon/genmon/genenv/bin/python: No module named pip\n'
Warning: required library pyopenssl not installed. Attempting to install....
Error in InstallLibrary using pip : ['/home/genmon/genmon/genenv/bin/python', '-m', 'pip', 'install', 'pyopenssl']: b'/home/genmon/genmon/genenv/bin/python: No module named pip\n'
Warning: required library spidev not installed. Attempting to install....
Error in InstallLibrary using pip : ['/home/genmon/genmon/genenv/bin/python', '-m', 'pip', 'install', 'spidev']: b'/home/genmon/genmon/genenv/bin/python: No module named pip\n'
Warning: required library mopeka_pro_check not installed. Attempting to install....
Error in InstallLibrary using pip : ['/home/genmon/genmon/genenv/bin/python', '-m', 'pip', 'install', 'mopeka_pro_check']: b'/home/genmon/genmon/genenv/bin/python: No module named pip\n'
Warning: required library voipms not installed. Attempting to install....
Error in InstallLibrary using pip : ['/home/genmon/genmon/genenv/bin/python', '-m', 'pip', 'install', 'voipms']: b'/home/genmon/genmon/genenv/bin/python: No module named pip\n'
Stopping....
Stopping genemail2sms.py
kill: (2123): No such process
Stopping gentankutil.py
kill: (2122): No such process
Stopping genserv.py
kill: (2121): No such process
Stopping genmon.py
kill: (2119): No such process
Starting....
Starting /home/genmon/genmon/genmon.py 
Starting /home/genmon/genmon/genserv.py 
Starting /home/genmon/genmon/addon/gentankutil.py 
Starting /home/genmon/genmon/addon/genemail2sms.py 
./startgenmon.sh: line 26: deactivate: command not found
genmon@generator:~/genmon$ 
genmon@generator:~/genmon$ ./startgenmon.sh restart
using binary: /home/genmon/genmon/genenv/bin/python
./startgenmon.sh: line 20: /home/genmon/genmon/genenv/bin/activate: No such file or directory
Restarting genmon python scripts
[... 28 lines cut ...]
Error in InstallLibrary using pip : ['/home/genmon/genmon/genenv/bin/python', '-m', 'pip', 'install', 'paho-mqtt']: b'/home/genmon/genmon/genenv/bin/python: No module named pip\n'
Warning: required library pyopenssl not installed. Attempting to install....
Error in InstallLibrary using pip : ['/home/genmon/genmon/genenv/bin/python', '-m', 'pip', 'install', 'pyopenssl']: b'/home/genmon/genmon/genenv/bin/python: No module named pip\n'
Warning: required library spidev not installed. Attempting to install....
Error in InstallLibrary using pip : ['/home/genmon/genmon/genenv/bin/python', '-m', 'pip', 'install', 'spidev']: b'/home/genmon/genmon/genenv/bin/python: No module named pip\n'
Warning: required library mopeka_pro_check not installed. Attempting to install....
Error in InstallLibrary using pip : ['/home/genmon/genmon/genenv/bin/python', '-m', 'pip', 'install', 'mopeka_pro_check']: b'/home/genmon/genmon/genenv/bin/python: No module named pip\n'
Warning: required library voipms not installed. Attempting to install....
Error in InstallLibrary using pip : ['/home/genmon/genmon/genenv/bin/python', '-m', 'pip', 'install', 'voipms']: b'/home/genmon/genmon/genenv/bin/python: No module named pip\n'
Stopping....
Stopping genemail2sms.py
kill: (3170): No such process
Stopping gentankutil.py
kill: (3169): No such process
Stopping genserv.py
kill: (3168): No such process
Stopping genmon.py
kill: (3167): No such process
Starting....
Starting /home/genmon/genmon/genmon.py 
Starting /home/genmon/genmon/genserv.py 
Starting /home/genmon/genmon/addon/gentankutil.py 
Starting /home/genmon/genmon/addon/genemail2sms.py 
./startgenmon.sh: line 26: deactivate: command not found
genmon@generator:~/genmon$

jgyates, 2023-11-10:
Hi @capindc 

I believe all debian bookworm based operating systems like the latest Rasbian and Ununtu 23 use a managed version of python 3.11 so the changes I made for Raspbian Bookworm should help you out. The managed version of python requires that you install python libraries in a virtual environment. the good news is that the recent updates to the project should do all of this for you.

First check your crontab to make sure that genmon is still set to load on boot, if it is great, if it is not we can fix this later in the install script.

     sudo crontab -e

make a note if it is set to run on install. We will use that info later.

Next get into your genmon folder and update to the latest software:

      cd genmon
      ./genmonmaint.sh -u

once this is complete then run the install script which will install the virtual python environment and install the needed libraries into the virtual environment.

        ./genmonmaint.sh -i

When prompted to install the conf files select 'n' for no as your conf files should already be in the /etc/genmom folder from your previous install.
When prompted to setup genmon on boot, you can select no if that was setup before (when you checked "sudo crontab -e"), otherwise select yes
When prompted to setup the serial port. Select No. I am not sure how this would work on Ubuntu, if you are using serial over TCP then select no.

Now just restart genmon:

      ./startgenmon.sh restart

Let me know if you have any issues or questions.

jgyates, 2023-11-11:
I am going to close this issue but feel free to post any related questions or status to this thread.

closed 2023-11-11 by jgyates

capindc, 2023-11-11:
sudo crontab -e is all good
./genmonmaint.sh -u no errors
./genmonmaint.sh -i does this look right
This script will install libraries needed by genmon. This script assumes you have already downloaded the genmon project via 'git'. This script will optionally copy the config files to the configuration directory. This script will not test the serial port. This script requires internet access to download the needed libraries. Press any key to continue.  
Managed system packages found, installing python virtual environment
E: Package 'python3-venv' has no installation candidate
Creating virtual python environmnet for genmon
The virtual environment was not created successfully because ensurepip is not
available.  On Debian/Ubuntu systems, you need to install the python3-venv
package using the following command.

    apt install python3.11-venv

You may need to use sudo with that command.  After installing the python3-venv
package, recreate your virtual environment.

Failing command: /root/genmon/genenv/bin/python3

Installing genmon package requirements....
/root/genmon/genenv/bin/python: No module named pip
Installing rpirtscts...
~/genmon ~/genmon
fatal: destination path 'rpirtscts' already exists and is not an empty directory.
make: 'rpirtscts' is up to date.
Finished installing rpirtscts.
~/genmon
Copy configuration files to /etc/genmon/? (y/n)?n
Not copying *.conf to /etc/genmon/
Setup the raspberry pi onboard serial port? (y/n)?n
Not setting up serial port
Done.
Start genmon on boot? (y/n)?y
Updating crontab....
./genmonmaint.sh: line 184: /tmp/gmtemp: Permission denied
Crontab has an incorrect configuration, updating:
@reboot sleep 30 && /bin/bash /home/genmon/genmon/startgenmon.sh start
to
@reboot sleep 30 && /bin/bash /root/genmon/startgenmon.sh start

 ./startgenmon.sh restart the very end does not look right
Stopping....
Stopping genserv.py
kill: (6428): No such process
Stopping genmon.py
kill: (6427): No such process
Starting....
Starting /root/genmon/genmon.py 
Starting /root/genmon/genserv.py 
./startgenmon.sh: line 26: deactivate: command not found
root@generator:~/genmon#

jgyates, 2023-11-11:
you can try the suggestion:

   sudo apt-get install python3.11-venv

to see if that fixes your problem. The underlying issue is that the install script tried to run this comand:

     sudo apt-get -yqq install python3-venv

and it failed, probably due to an issue with your apt sources in /etc/apt/sources.list. From what I understand that can have issues sometime on a upgrade of the distro. 

this thread shows a similar issues from a couple of years back (upgrading Ubuntu then failing to install python virutal environments. It offers a couple of suggestions.

https://askubuntu.com/questions/1285650/cannot-install-python-venv-on-ubuntu-20-04-after-upgrading-from-bionic

here is another one:
https://discuss.python.org/t/venv-trouble-with-python3-on-ubuntu-22-04-jammy/25890/12

If any of these correct the issue then you will need to run the install script again, letting the install script install the virtual environment:

      ./genmonmaint.sh -i 

an alternative solution is to remove the managed python library restriction by rename the file /usr/lib/python3.11/EXTERNALLY-MANAGED

     sudo mv /usr/lib/python3.11/EXTERNALLY-MANAGED /usr/lib/python3.11/EXTERNALLY-MANAGED.bak

This will make using virtual python environments not a requirement and it will act more like earlier versions of python where the system shares all python libraries installed instead of an app getting its own version of installed python libraries. If you go this route then you will need to run the install script again:

      ./genmonmaint.sh -i

Let me know if you have any questions.

capindc, 2023-11-11:
I ended up deleting EXTERNALLY-MANAGED.bak
then ./genmonmaint.sh -i
add-on's & optional outbound setting not saved
So far all good
Thank you thank you