jgyates/genmon #974: No alert sent on alarm

dsolodow, 2023-11-13:
Outbound email is configured, and tested successfully to the alert recipients. Notifications are set for outage, error,  warn. The alarm log shows a shutdown due to ruptured basin (high) but not email notification was sent.

### Screenshot or Pictures relating to the problem (if possible)
[image]

{Please write here}

### Your Environment

- Generator Model: Generac H-100 Industrial Generator
- Genmon Logs: Submit via the About page
- Genmon Version: 1.19.00; was 1.18.18 at time of alarm

jgyates, 2023-11-13:
I received your log files. Thanks. I assume that this alarm is not currently active on your controller. Has it been acknowledged on the controller? or did the condition not require an acknowledgement?

First I would check your spam folder to make sure that the email was not filtered. You can also check the sent folder on the email account used for sending to make sure the email was actually sent.

You logs indicate that there was an identical alarm on 8/23/2023. Did you receive and email for this event?

On the H-100, the only way that genmon determines an alarm is active is if one or more of these conditions is active:

1) The Alarm ACK count register (modbus register 012e) reads a non zero value
2) The Active Alarm count register (modbus register 012f) reads a non zero value

The design of genmon relies on the controller as much as possible to keep up with active alarms.

I can only guess that one of thee following occurred:

1) The email was sent but due to some filtering it was not received.
2) The controller did not set both of these two registers due do some other setting on your controller?
3) The alarm condition on the controller was momentary and was cleared before genmon could cycle through reading the other registers. I am not sure why this would happen as alarms are suppose to be acknowledged but I put it here as this technically could occur.

jgyates, 2023-11-14:
I am going to close this issue for now but please response to any questions or with an updated status within this thread.

closed 2023-11-14 by jgyates