jgyates/genmon #975: Bug / Feature request

mstackler, 2023-11-17:
I had one of our generators run its monthly test today.  It is programed to run with load thru the transfer switch.  I received emails from genmon and was hoping we could tweak the email send feature.

What i received is below.  Was thinking could the email be delayed by 30 seconds or so?  That way it would populate all the data with usable values.  RPM, Frequency, ect

mark

Time: 07:04:32 AM
Date: Fri 17-Nov-2023

Status : 

    Engine : 
        Engine State : Running from 2-wire 
        Generator Status : Cranking            
        Switch State : Auto
        Output Power : 0.00 kW
        Power Factor : 1.00 
        RPM : 0 
        Frequency : 0.00 Hz
        Throttle Position : 0 Stp
        Coolant Temp : 118 F
        Coolant Level : 792 Stp
        Oil Pressure : 0 psi
        Oil Temp : 104 F
        Fuel Level : 85 
        Oxygen Sensor : 0 
        Current Phase A : 0 A
        Current Phase B : 0 A
        Current Phase C : 0 A
        Average Current : 0 A
        Voltage A-B : 0 V
        Voltage B-C : 0 V
        Voltage C-A : 0 V
        Average Voltage : 0 V
        Air Fuel Duty Cycle : 0.00 

    Alarms : 
        Number of Active Alarms : 0 
        Number of Acknowledged Alarms : 1 

        Alarm List : 
            Emergency Stop

    Battery : 
        Battery Voltage : 26.94 V
        Battery Charger Current : 0.30 A

    Line State : 
        Transfer Switch State : Utility

    Time : 
        Monitor Time : Friday November 17, 2023 07:04:30
        Generator Time : Friday November 17, 2023 07:04:21

jgyates, 2023-11-17:
Based on the output this looks like an H-100, but correct me if I am wrong. 

The emails are sent out based on a change in the engine state. The evolution controller sends out emails when the engine state, switch state or generator state change. Presently the H100 only sends notices on a change in the engine state. Changing the H100 to send emails on all three states (switch, engine, generator) as well would likely give you what you want to see however you would get an email for every change (cranking, warming up, running, etc). This change would be more maintainable and provide more info so I will opt to make this change instead of a delayed email. Apparently 2 wire mode on the H100 has fewer engine states to display vs non 2 wire mode.

I have made this change to the H100 code. Update to the latest on the about page and let me know if you have any issues.

closed 2023-11-17 by jgyates

mstackler, 2023-11-17:
Yes I am using the h-100 panel. I will update the software Monday and test it. Thank you for the feedback back and updates. MarkSent from my Mark's  iPhone