jgyates/genmon #976: snmp communication

gaylordw, 2023-11-17:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

snmpwalk used to poll genmon every 5 minutes and worked flawlessly until the V1.19 update

### Actual Behavior

snmpwalk no longer can connect to genmom

### Steps to Reproduce (including precondition)

genmon starts and runs normally but cannot communicate with snmp.

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: Eveloution 2.0
- Genmon Logs: Submit via the About page
- Genmon Version: 1..9

jgyates, 2023-11-17:
I got your log files. Can you perform this test:

1) log into your pi via ssh and edit the file /etc/genmon/gensnmp.conf. Add the like

       debug=True

save the file
2) restart genmon. From the genmon folder type

      ./startgenmon.sh restart

3) wait 1 minute then send your logs on the About page

4) edit the file again and remove the debug=True line (or set it to False instead of True
5) restart genmon.

This will give me more info so I can resolve this issue.

jgyates, 2023-11-17:
also, what is the output of this command on your pi:

     grep 'controllertype' /etc/genmon/genmon.conf

jgyates, 2023-11-17:
Based on your logs I am assuming the issues is that your "controllertype" entry in /etc/genmon/genmon.conf is either misspelled, not present or invalid. I checked in a change that will compensate for this however I would check this to make sure that your setting looks like this:

      controllertype = generac_evo_nexus

This is the default setting. If you set this value to h_100 it will use the H100 controller. If it is not present then genmon will default to generac_evo_nexus. The SNMP addon (gensnmp) was modified to detect the controller used and load a config file that contains the SNMP definitions for that controller. In your case the gensnmp add on could not determine the controller type due to the invalid setting in the /etc/genmon/genmon.conf file. The change I checked in made the gensnmp code a bit more resilient to invalid settings. Normally this setting is not changed or it is is modified via the web app. I am guessing this was modified manually an this may be the reason for this invalid setting. Anyway, If you update form the About page this *should* fix your issue if I have deduced the problem correctly from what I have. It f it does not fix your issues then send the requested info above and we can go from there. 

I am going to close this issue but please let me know via this thread if this corrects your problem or not.

closed 2023-11-17 by jgyates

gaylordw, 2023-11-21:
I just got around to looking at this. I reinstalled the version from the about page and I am now able to see the snmp data.  The controller type entry in the conf file is not set.  Should this be changed or leave it as is?  Thanks for the help.Sent from my Galaxy