jgyates/genmon #977: New Evolution 2.0 not reporting total run hours

calabresemic, 2023-11-21:
### Expected Behavior

Report total run hours in Service section of Maintenance Page

### Actual Behavior

Reports 0.00 h Total Run Hours

### Steps to Reproduce (including precondition)

Connected an opengenset to the generator and it looks like it's reporting all the information except total run hours. I can get Run Hours in the last 30 days and last year but not total hours.

### Screenshot or Pictures relating to the problem (if possible)

[image]

[image]

### Your Environment

- Generator Model: G0071710
- Genmon Logs: Submit via the About page
- Genmon Version: V1.19.00

calabresemic, 2023-11-21:
It seems that it only reports full hours. My appologies

closed 2023-11-21 by calabresemic

gm58, 2023-11-21:
Did you read somewhere that it only reports fulls hours? I'm wondering why the decimal if that was the case.

jgyates, 2023-11-21:
Evolution Liquid Cooled only reports a whole number for run hours over modbus. The Liquid Cooled model supports reading the engine run minutes, which genmon converts to hours with a decimal remainder.

Run Hours in the last 30 days and last year are calculated by genmon from the power log (not read over modbus from the controller) and are floating point with a decimal value.

The run hours reporting a floating point number was chosen for maintainability as this value is passed to MQTT and SNMP add ons and it is easier to have a consistent data type across the different controllers. The result of this is that on Evolution Air cooled and Nexus controller there will be two trailing zeros displayed.

calabresemic, 2023-11-21:
After it ran for long enough it populated that field with an hour. Since it's in the service history section I'm guessing it's purposely lower resolution.