jgyates/genmon #978: is the Generac PM-SC supported?

mstackler, 2023-11-21:
I have multiple sites running generators in parallel with a  Generac PM-SC.  Can i connect Genmon to the RS-232 port on it and get the data from all generators with the multiple genmon feature?

mark

jgyates, 2023-11-21:
Hi Mark,

the short answer is, I don't know, but probably not in its current state. I have never tried that before and I am not currently familiar with how the multiple generators are addressed. I would guess each as a separate modbus address but that is just a guess at this point. 

Here is the best shot at it working in it's current state that I can think of:

**If** you addressed each generator with a separate modbus address and **if** you had a serial to TCP converter that accepted multiple connections, in theory you could run multiple instances of genmon, each with a different modbus address for each connected generator, but all instances communicating with the same serial to TCP converter. 

If that does not work we can run some tests and see if we can come up with a solution. Let me know what you think.

mstackler, 2023-11-21:
I will look into it.  I will try the modbus dump on it and see what it gives me. It is hard to find documentation on these systems.  I have some time around the holidays to spend on this.  Just wanted to see if it has been tried before.

mark

jgyates, 2023-11-22:
sounds good. let me know if you have any questions that I can help with. I am going to close this thread but post here with any followup.

closed 2023-11-22 by jgyates