jgyates/genmon #979: Web update killed GenMon

howardbuck, 2023-11-21:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Last WEB update did not complete and killed GenMon  I then did the bash ./genmonmaint.sh which completed OK  with errors. I then did /bin/bash /home/pi/genmon/startgenmon.sh start and got the following -

### Actual Behavior

This program requires the modules located in the genmonlib directory in the github repository.
Please see the project documentation at https://github.com/jgyates/genmon.
Error: invalid syntax (mycommon.py, line 41)

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: Honeywell 17KW
- Genmon Logs: Submit via the About page
- Genmon Version: Updating to 19

howardbuck, 2023-11-21:
Here is the output from genmonmaint.sh -
pi@genpi [~/genmon] bash ./genmonmaint.sh -i
This script will install libraries needed by genmon. This script assumes you have already downloaded the genmon project via 'git'. This script will optionally copy the config files to the configuration directory. This script will not test the serial port. This script requires internet access to download the needed libraries. Press any key to continue.
Installing genmon package requirements....
E: The repository 'http://raspbian.raspberrypi.org/raspbian stretch Release' does no longer have a Release file.
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/h/http-parser/libhttp-parser2.1_2.1-2_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/o/openssl/libssl-dev_1.1.0l-1~deb9u6_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/o/openssl/libssl1.1_1.1.0l-1~deb9u6_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/r/rustc/libstd-rust-1.24_1.24.1+dfsg1-1~deb9u2+rpi1_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/r/rustc/libstd-rust-dev_1.24.1+dfsg1-1~deb9u2+rpi1_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/r/rustc/rust-gdb_1.24.1+dfsg1-1~deb9u2+rpi1_all.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/r/rustc/rustc_1.24.1+dfsg1-1~deb9u2+rpi1_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/c/cargo/cargo_0.15.0~dev-1+rpi1_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
/usr/local/lib/python3.5/dist-packages/OpenSSL/crypto.py:14: CryptographyDeprecationWarning: Python 3.5 support will be dropped in the next release of cryptography. Please upgrade your Python.
  from cryptography import utils, x509
Requirement already satisfied: Flask in /usr/local/lib/python3.5/dist-packages (from -r requirements.txt (line 1))
Collecting configparser (from -r requirements.txt (line 2))
Exception:
Traceback (most recent call last):
  File "/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 594, in urlopen
    chunked=chunked)
  File "/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 391, in _make_request
    six.raise_from(e, None)
  File "<string>", line 2, in raise_from
  File "/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 387, in _make_request
    httplib_response = conn.getresponse()
  File "/usr/lib/python3.5/http/client.py", line 1256, in getresponse
    response.begin()
  File "/usr/lib/python3.5/http/client.py", line 320, in begin
    version, status, reason = self._read_status()
  File "/usr/lib/python3.5/http/client.py", line 289, in _read_status
    raise RemoteDisconnected("Remote end closed connection without"
http.client.RemoteDisconnected: Remote end closed connection without response

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/usr/lib/python3/dist-packages/pip/commands/install.py", line 353, in run
    wb.build(autobuilding=True)
  File "/usr/lib/python3/dist-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/usr/lib/python3/dist-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/usr/lib/python3/dist-packages/pip/req/req_set.py", line 554, in _prepare_file
    require_hashes
  File "/usr/lib/python3/dist-packages/pip/req/req_install.py", line 278, in populate_link
    self.link = finder.find_requirement(self, upgrade)
  File "/usr/lib/python3/dist-packages/pip/index.py", line 465, in find_requirement
    all_candidates = self.find_all_candidates(req.name)
  File "/usr/lib/python3/dist-packages/pip/index.py", line 423, in find_all_candidates
    for page in self._get_pages(url_locations, project_name):
  File "/usr/lib/python3/dist-packages/pip/index.py", line 568, in _get_pages
    page = self._get_page(location)
  File "/usr/lib/python3/dist-packages/pip/index.py", line 683, in _get_page
    return HTMLPage.get_page(link, session=self.session)
  File "/usr/lib/python3/dist-packages/pip/index.py", line 792, in get_page
    "Cache-Control": "max-age=600",
  File "/usr/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 501, in get
    return self.request('GET', url, **kwargs)
  File "/usr/lib/python3/dist-packages/pip/download.py", line 386, in request
    return super(PipSession, self).request(method, url, *args, **kwargs)
  File "/usr/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 488, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 609, in send
    r = adapter.send(request, **kwargs)
[... cut ...]
Backup file already exist: /boot/cmdline.txt.bak1

Disable BT: Success
Enable UART: Success
Change BT serial port: Success
Disable BT service service: Success
Disable serial console: Success
Disable serial console service: Success

Serial port settings are OK. A reboot is needed if changes were made.

Finished setting up the serial port.
~/genmon
Done.
Start genmon on boot? (y/n)?y
Updating crontab....
Crontab already contains genmon start script:
@reboot sleep 30 && /bin/bash /home/pi/genmon/startgenmon.sh start
pi@genpi [~/genmon]

jgyates, 2023-11-21:
@howardbuck 

I am not sure what version your were upgrading from.

Here are the observations from the data provided so far:

1) You appear to be running Raspbian stretch, which is not supported by the Raspberry pi foundation so you can't do any updates via apt, which the installation script performs when you try to install. That is not a show stopper for an existing installation but the errors in you are seeing in the installation script are due to trying to update an old raspbian operating system. Raspbian Stretch is version 9 from 2020, the current veriosn is Raspbian bookworm version 12 from 2023

2) The error "Error: invalid syntax (mycommon.py, line 41)" was fixed about a week ago. This issue was specific to python 3.5, which is included Raspbian stretch. The latest is V3.11. 

Since you are running an older version of the operating system, rather than re-installing, just try upgrading the genmon sources manually. To do so run this command from the terminal:

     cd genmon
     ./genmonmain.sh -u

You will need to type 'y' to confirm the update. If you get any errors, post them here.

once you do this, then restart genmon

      ./startgenmon.sh restart

Let me know if you have any issues or questions and let me know if this works for you. 

I am going to close this thread but please post any follow up here.

howardbuck, 2023-11-21:
Thank you very much.  I was able to get GenMon working again. It is on V1.19.00 now.
That leads to several questions- 
Is there a way for me to upgrade to Raspbian bookworm version 12?
To upgrade in the future do I use the same commands?
      cd genmon
     ./genmonmaint.sh -u
     ./startgenmon.sh restart

Here is the log from what I did -

pi@genpi [~] cd genmon
pi@genpi [~/genmon] ./genmonmaint.sh -u
This script will update genmon to the latest version from the github repository. This script requires internet access. If you have modified any files in the genmon directory, they will be overwritten. Configuration files in the configuration directory will not be overritten.  Continue? (y/n)  y
Updating genmon...
remote: Enumerating objects: 70, done.
remote: Counting objects: 100% (70/70), done.
remote: Compressing objects: 100% (32/32), done.
remote: Total 70 (delta 47), reused 61 (delta 38), pack-reused 0
Unpacking objects: 100% (70/70), done.
From https://github.com/jgyates/genmon
   885eee7..2256121  master     -> origin/master
 * [new tag]         V1.19.00   -> V1.19.00
Checking out files: 100% (269/269), done.
HEAD is now at 2256121 V1.19.00
Done.
pi@genpi [~/genmon] ./startgenmon.sh restart
Restarting genmon python scripts
/usr/local/lib/python3.5/dist-packages/OpenSSL/crypto.py:14: CryptographyDeprecationWarning: Python 3.5 support will be dropped in the next release of cryptography. Please upgrade your Python.
  from cryptography import utils, x509
Running one time maintenance check
Error in ExecuteCommandList  : b'/usr/local/lib/python3.5/dist-packages/OpenSSL/crypto.py:14: CryptographyDeprecationWarning: Python 3.5 support will be dropped in the next release of cryptography. Please upgrade your Python.\n  from cryptography import utils, x509\n'
Installing python3-pip
Error in ExecuteCommandList  : b'E: Command line option --allow-releaseinfo-change is not understood in combination with the other options\n'
Error: Unable to run apt-get update. Retrying..
Error in ExecuteCommandList  : b"E: The repository 'http://raspbian.raspberrypi.org/raspbian stretch Release' does no longer have a Release file.\n"
Error: Unable to run apt-get update. Retrying..
Error in GetLibararyVersion using pip : pyowm: b'/usr/local/lib/python3.5/dist-packages/OpenSSL/crypto.py:14: CryptographyDeprecationWarning: Python 3.5 support will be dropped in the next release of cryptography. Please upgrade your Python.\n  from cryptography import utils, x509\n'
Stopping....
Starting....
Starting /home/pi/genmon/genmon.py
Starting /home/pi/genmon/genserv.py
pi@genpi [~/genmon]

jgyates, 2023-11-21:
you should be able to use the web interface to update in the future. I am guessing that you may have been coming from a version before 1.18.18. There have been a lot of changes so if you get to far out on an upgrade there could be some issues since I don't test every combination of upgrades (typically I just test the last release to the latest). If for some reason you do have an issue you can just use the steps above to correct it. 

One thing to consider is getting a new SD card and install the latest version of Raspberry pi os. Not a requirement but it would likely keep your genmon system running longer. You can either plan an upgrade, or run genmon as it until you can't with the older OS, or just stop upgrading. Of course if you don't upgrade, then if you do have an issue it is not as likely I can help out as I only support the latest version. [This page](https://github.com/jgyates/genmon/wiki/Appendix-M-Tips-and-Tricks-useful-for-Genmon#restoring-settings-from-a-backup) describes how to save and restore your genmon settings. Once you save them just run the setup on the new system, [described here](https://github.com/jgyates/genmon/wiki/3.3--Setup-genmon-software), then restore you settings. 

I am glad it is working for you.

closed 2023-11-21 by jgyates