jgyates/genmon #980: i had to replace my defective Netgear router last week, now i cant see generator

grsthegreat, 2023-11-23:
I log the generator to my network as i did previously. i reboot the RaspberryPi unit and wait until i see GenPi on the phone available wifi signals.

i then log into that wifi signal, then open wifisetup.pi:1880/GenPi page on browder. i assign the available wifi name and password. the system reboots. 

but i dont receive an email that i have connected like i used to, and when i run fing, i dont see the generator. 

i have installed a TP-Link AX3000 router that my internet provider supplied me. 

router is set to WPA2-PSK(aes) security.

anyone have any idea what setting i need to enable for it to get hooked nto my network?

ive tried installing it 6 times now. Evert other wifi device has been able to connect.

jgyates, 2023-11-23:
If your pi setup for static or dynamic IP addressing? If the wifisetup page loads then that show the pi is working. Based on the description it sounds like your pi is not connecting to your wfi network.

you may want to validate the Wifi band the router is using. Often fast wifi (like wifi 6) routers will disable the slower wifi bands (b or g bands) as they slow the the performance if your network does not have any of the older devices. I would:

1) check the band your pi is using
2) make sure that band is enabled on your router

Other things to check:

Your router should have the ability to show a list of connected wifi clients and another list of devices that have been allocated a DHCP address.

grsthegreat, 2023-11-23:
My old router was static ip address, so is the new one. I would guess the pi was set for static as it used to work.

i cant find it on putty, as i dont know the ip address its issuing. Using fing i tried to install every listed ip address in web browser, none work to bring up genmon screen

grsthegreat, 2023-11-23:
The new router is set to find all 802.11 b/g/n devices

grsthegreat, 2023-11-23:
I take that back, i can log into the unit with putty. Not sure what to do now

jgyates, 2023-11-23:
if you can long into the pi via ssh your wifi is working.

you mentioned installing multiple times. Did you overwrite your config files? If so they this would explain why you are not getting emails. It would also reset the use of secure HTTP so you will need to validate this.

You can try to reload genmon manually to see if there are any issues:

     cd genmon
     ./startgenmon.sh restart

Once you do this post the output of the above command here. Also post the output of this command after you restart:

     cat /var/log/genserv.log

I have seen some routers block web sites on the local network so you may want to look at the security settings and disable any "AI" type security setting. 

Also, include the address you are using the browser to access genmon so I can compare it to the output of your genserv.log file. You may be looking for genmon at the wrong address if you reinstalled and wiped your settings.

grsthegreat, 2023-11-23:
Ok, i restarted as described. Not sure how to paste putty results here, as it a huge screen of run on wording. I have no link to send logs as i cannot enter genmon app, only putty. 
[image]

grsthegreat, 2023-11-23:
Is there a way , using putty, to see what ip address genmon is using, so i can add this address to my router

skipfire, 2023-11-23:
`ip address` will list out all the network details.

grsthegreat, 2023-11-23:
Ah…..that did it. I found the ip address. Now its working

grsthegreat, 2023-11-23:
Thanks . And have a happy thanksgiving

grsthegreat, 2023-11-23:
Happy thanksgiving

closed 2023-11-23 by grsthegreat