jgyates/genmon #981: Need help modbus id on h-100

RickNDozer, 2023-11-24:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here}
 Hi I am going out on a limb here I do know know what I am doing on this website, it’s the only info I have found on the internet regarding generac and modbus communication, I am working on a h-100 panel trying to get comms to a Maximus controller with a modbus card I can not find the modbus ID on the h-100 panel. Could someone please email at naberhausnick@gmail.com with any help thank you

jgyates, 2023-11-24:
if you are looking for the modbus ID of a H-100, it is configurable in the menus, but I think the default is 100 decimal (64 hex)

RickNDozer, 2023-11-24:
Thank you for the reply. Is it configurable through the menu on the h-100 panel or only through genlink? I talked to generac tech support on the phone the couldn’t tell me anything but guess what it was and told me I would have to have a generac industrial certified tech come out and connect to genlink to view or chance the ID number. I appreciate the help I have been lost this my first time doing anything with modbus.  We are trying to connect the generator to the controller of a turkey barn so they can view amps, alarms m, etc on the controller remotely so when they weather gets bad they have a piece of mind staying at home

jgyates, 2023-11-24:
I don't know the answer as I don't own a H100. Someone else may be able to help with that questions.
You can also ask in one of these two forums:

https://www.zillerstore.com/categories
https://gentekpower.com/forums/

closed 2023-11-24 by jgyates

liltux, 2023-11-24:
@RickNDozer  you can view modbus info via the display, Menu, Diagnostic, page thru Inputs, Outputs, Analogs, then you will find Comms.  Comm 0 is the main port on front of controller DB9, should be RS232. Comms 1 is RS485 for annunciator panels.  On the RS232 page it will display the settings and ID.

RickNDozer, 2023-11-25:
Right now we are using the RS485 port. It’s running to what I believe is called a modbus bridge possibly correct me if I’m wrong on that, we are just trying to read the data from the generator amps, volts, faults should we be using the RS232 instead?

liltux, 2023-11-25:
@RickNDozer that is a problem.  These are two separate interfaces.  If you were able to get to the diagnostic screen you would see that by default the RD485 port is configured as a master.  To use the RS485 port you will have to have an Industrial Technician on site to change that to Modbus Slave so it will work correctly.  Also it’s settings are different as well, 4800, 8, 2 stop bits.  All of that has to be configured with GenLink and a certified tech.

RickNDozer, 2023-11-25:
Would it be possible to do it through the RS232 port?

liltux, 2023-11-25:
@RickNDozer yes. The default settings for the RS232 are: Modbus Slave, 9600, 8, 1, none.  That port does not require any intervention to use, but it is RS232.  So you may need to transition from that to 485 or event over Ethernet using a converter.