jgyates/genmon #983: Having trouble with initial setup - serialconfig

eBoon123, 2023-11-28:
Setting up a new Genmon for a friend and I'm having some problems with the initial setup. Have followed everything for the setup of a brand new Pi 3B+ from section 3.2 of the Wiki , but I am getting an error from serialconfig - "Is BT overlay disabled: Fail".

/genmon/OtherApps $ sudo python3 serialconfig.py -c

Checking : Is enable UART in boot config: OK
Checking : Is BT overlay disabled: Fail
Checking : Serial console command line removed: OK
Checking : Serial console service disabled: OK
Checking : BT service disabled: OK

Here's what the bottom of my /boot/config.txt looks like:

[all]
enable_uart=1
dtoverlay=pi3-disable-bt

I have rebooted (multiple times).

What am I missing?

Thank you for your help!

jgyates, 2023-11-28:
I think you can safely ignore this failure message. The back story is when the pi3 came out the name of the setting was 'pi3-disable-bt' but they changed it in later raspberry pi kernels to 'disable-bt'. I updated serialconfig.py to use check for the presence of 'disable-bt' instead of 'pi3-disable-bt'. I think the kernel will allow either one. Looks like I forgot to update the wiki. If you enable the serial port with via serialconfig.py instead of editing config.txt then the program will show success instead of failure. Thanks for pointing this out. I will update the wiki to match what serailconfig.py is setting / checking.

jgyates, 2023-11-28:
I updated the wiki. I am going to close this issue but if you have any follow up questions feel free to post them to this thread.

In short, you can either ignore the message, or change config.txt to 'dtoverlay=disable-bt'

closed 2023-11-28 by jgyates

eBoon123, 2023-11-28:
Thank you! I can confirm that when I used 'disable-bt' that serialconfig is now happy.