jgyates/genmon #988: Mopeka Pro+ Config

philruel, 2023-12-11:
### Actual Behavior
I'm trying to configure Genmon to install a Mopeka Pro + sensor
I did:
`sudo python3 serialconfig.py -e -b`
Reboot
`sudo python3 serialconfig.py -c -b`
All ok
`sudo bluetoothctl`
**White cursor and nothing show up**
So Ctrl C
Restore  your serialconfig.py settings
`sudo python3 serialconfig.py -r`
`sudo python3 serialconfig.py -e -b`
Reboot

Now, I cannot connect to Raspberry Pi 4 via VNC:  Cannot currently show the desktop
Genmon is running but Add-Ons dials are not displayed but are On:  External Temperature Sensors and External CT Sensors.
Did try to reboot the Pi via Genmon but results are the same.

### Your Environment

- Generator Model: 0065520
- Genmon Logs: Submit via the About page
- Genmon Version: 1.19.00

jgyates, 2023-12-11:
I received your log files.  
Based on your description it does not look like the BT controller is enabled on your pi.

Based on your description above I do not see where you reset your serial config before enabling bluetooth. You logs indicate that you have been running genmon for some time.  The initial genmon BT setup utilizes serial and disables BT.  If you already had serial working and want to enable BT at a later time you need to reset your serial config. Once that is done you can use serialconfig.py to enable both serial and BT.

I would reset your serial config:

     sudo python3 serialconfig.py -r

reboot your pi

     sudo reboot

After the reboot type:

    cd genmon
    cd OtherApps
    sudo python3 serialconfig.py -e -b
   sudo reboot

When the pi comes up, validate that your serial communications are still working. If they are then proceed to the next step. If not show the output of this command and the serial port name in your genmon setting:

      ls -la /dev/ser*

If your serial comms are working then type this:

     sudo bluetoothctl

If you do not get the MAC address showing something like this (you will get a different controller address, or not address if your controller is not enabled):

     Agent registered
    [CHG] Controller DC:A6:32:96:B0:03 Pairable: yes
    [bluetooth]# 

then type this command at the [bluetooth] prompt:

      power on

Then type
     
      exit

to exit the bluetoothctl program

jgyates, 2023-12-11:
also, you should have these in your /boot/config.txt file:

     enable_uart=1
     dtoverlay=miniuart-bt
     force_turbo=1

It is possible that you may have other program that have modified the /boot/config.txt file (like a graphical desktop that tries to enable BT) that may not be compatible with the serialconfig.py program (just a guess).

jgyates, 2023-12-11:
You can post any status or questions to this thread. I am going to close this thread for now.

closed 2023-12-11 by jgyates

philruel, 2023-12-12:
Did
`cd genmon`
`cd OtherApps`
`sudo python3 serialconfig.py -r`
Restoring...
Restored /boot/config.txt.bak1 to /boot/config.txt
Restored /boot/cmdline.txt.bak1 to /boot/cmdline.txt
Restore Complete.

` sudo reboot`
Rebooted

Did
`cd genmon`
`cd OtherApps`
`sudo python3 serialconfig.py -e -b`
Backup file already exist: /boot/config.txt.bak1
Backup file already exist: /boot/cmdline.txt.bak1

Enable UART: Success
Change BT serial port: Success
Enable Turbo clocking: Success
Disable BT Disable Overlay: Success
Disable serial console: Success
Disable serial console service: Success
Enable BT service service: Success

Serial port settings are OK. A reboot is needed if changes were made.

` sudo reboot`
Rebooted

Did
`cd genmon`
`cd OtherApps`
`sudo python3 serialconfig.py -c`

Checking : Is enable UART in boot config: OK
Checking : Is BT overlay disabled: Fail
Checking : Serial console command line removed: OK
Checking : Serial console service disabled: OK
Checking : BT service disabled: Fail

Serial port may not work as expected. Not all required settings changes were detected.

Is this how to check if the serial communications are still working?  
My Pi is not plugged to the generator at the moment because I cannot connect to the Raspberry Pi 4 via VNC: Cannot currently show the desktop.

Did
`ls -la /dev/ser*`
lrwxrwxrwx 1 root root 7 Dec 12 08:37 /dev/serial0 -> ttyAMA0
lrwxrwxrwx 1 root root 5 Dec 12 08:37 /dev/serial1 -> ttyS0

In genmon, I have:  /dev/serial0
Is it okay?

Did
`sudo bluetoothctl`
Agent registered
[bluetooth]# 

No MAC address.

How do I check you my  /boot/config.txt file?

Thank you!

jgyates, 2023-12-12:
the file /boot/config.txt is a text file so you can view it with a text editor in the graphical user interface or you can use a command line text editor like nano.

      sudo nano /boot/config.txt

To exit the editor type Ctrl+X. Here is a full list of commands: https://www.unomaha.edu/college-of-information-science-and-technology/computer-science-learning-center/_files/resources/CSLC-Helpdocs-Nano.pdf

your genmon serial setup looks good. You just don't have a BT controller enabled. 

Make sure your /boot/config.txt **does not** have the following line:

     dtoverlay=disable-bt

and it should have these lines:

     enable_uart=1
     dtoverlay=miniuart-bt
     force_turbo=1

if you make any changes to the file in nano, type Ctrl-X to exit, then 'y' for yes to save, then Enter to save the file. A reboot is required if any changes to /boot/config.txt are made.

philruel, 2023-12-12:
/boot/config.txt does not contain
`dtoverlay=disable-bt`

And it does contain 
` enable_uart=1`
` dtoverlay=miniuart-bt`
` force_turbo=1`

jgyates, 2023-12-12:
also, what is the output of this command:

     systemctl status hciuart.service

Mine is:
[image]

jgyates, 2023-12-12:
and this command:

     cat /lib/systemd/system/hciuart.service

mine is this:
[image]

jgyates, 2023-12-12:
your could try updating your system software: https://github.com/jgyates/genmon/wiki/Appendix-M-Tips-and-Tricks-useful-for-Genmon#updating-your-pi-system-software

philruel, 2023-12-12:
> what is the output of this command:
> 
> ```
>  grep 'bt' /boot/config.txt 
> ```
> 
> in my system it is this:
> 
> ```
>  dtoverlay=miniuart-bt
> ```

Same as yours

philruel, 2023-12-12:
> also, what is the output of this command:
> 
> ```
>  systemctl status hciuart.service
> ```
> 
> Mine is: <img alt="Screenshot 2023-12-12 at 10 23 50 AM" width="817" src="https://private-user-images.githubusercontent.com/4513286/289912331-7fe2f479-7240-49b8-8183-dcabaaef6113.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDIzOTU1NzIsIm5iZiI6MTcwMjM5NTI3MiwicGF0aCI6Ii80NTEzMjg2LzI4OTkxMjMzMS03ZmUyZjQ3OS03MjQwLTQ5YjgtODE4My1kY2FiYWFlZjYxMTMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIxMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMTJUMTUzNDMyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9M2EzMGI4YTdkNzU0YjcyZGVmNDRmYTY5NDNjN2JhOGM5YzVjMGY4YWUyMTRkOGZiNzExYTAzYTQ2MzJhMDY5MCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.DzOISRi5e-MKPN__aQqZIBeDT_boW7Gd71t0D3IDono">

● hciuart.service - Configure Bluetooth Modems connected by UART
   Loaded: loaded (/lib/systemd/system/hciuart.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2023-12-12 10:22:44 EST; 9min ago
  Process: 432 ExecStart=/usr/bin/btuart (code=exited, status=0/SUCCESS)
 Main PID: 675 (hciattach)
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/hciuart.service
           └─675 /usr/bin/hciattach /dev/serial1 bcm43xx 460800 noflow -

Dec 12 10:22:37 raspberrypi systemd[1]: Starting Configure Bluetooth Modems connected by UART...
Dec 12 10:22:44 raspberrypi btuart[432]: bcm43xx_init
Dec 12 10:22:44 raspberrypi btuart[432]: Flash firmware /lib/firmware/brcm/BCM4345C0.hcd
Dec 12 10:22:44 raspberrypi btuart[432]: Set Controller UART speed to 460800 bit/s
Dec 12 10:22:44 raspberrypi btuart[432]: Device setup complete
Dec 12 10:22:44 raspberrypi systemd[1]: Started Configure Bluetooth Modems connected by UART.

philruel, 2023-12-12:
> and this command:
> 
> ```
>  cat /lib/systemd/system/hciuart.service
> ```
> 
> mine is this: <img alt="Screenshot 2023-12-12 at 10 30 19 AM" width="619" src="https://private-user-images.githubusercontent.com/4513286/289914410-9e519b41-96a0-456b-baf7-d8038dc0b719.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDIzOTU1NzIsIm5iZiI6MTcwMjM5NTI3MiwicGF0aCI6Ii80NTEzMjg2LzI4OTkxNDQxMC05ZTUxOWI0MS05NmEwLTQ1NmItYmFmNy1kODAzOGRjMGI3MTkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIxMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMTJUMTUzNDMyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NjM2YWJmNmQ1MDBjY2E3ZGM4OTkxMjBiZjhmYWM4MDNkZmM4NGZjOGEyMzY0OTk4ZTgxZjcyNWFhMmNkN2FjYyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.vXn5TSgvQqvFvDhC2kMkNu78b4xOkZ4ayqkiNCfHQ-U">

pi@raspberrypi:~ $ cat /lib/systemd/system/hciuart.service
[Unit]
Description=Configure Bluetooth Modems connected by UART
ConditionFileNotEmpty=/proc/device-tree/soc/gpio@7e200000/bt_pins/brcm,pins
After=dev-serial1.device

[Service]
Type=forking
ExecStart=/usr/bin/btuart

[Install]
WantedBy=multi-user.target

jgyates, 2023-12-12:
so far nothing looks out of the ordinary. According to your logs you are running the OS version bullseye (version 10), the latest is bookwork (version 11). 

Here are the things I would try:

1) update your system software
2) try running the provision program while holding the sync button on the mopeka sensor, your BT may be working but just not showing the controller MAC address.

     cd genmon
     cd OtherApps
     sudo python3 mopeka_utility.py 

If your BT system is working you should see your sensor in the output if you hold the sync button while running the program. If it does not you can post the output here.

3) if that does not work then I would save your genmon settings and reinstall the operating system.

philruel, 2023-12-12:
The Mopeka sensor doesn't get discovered.  I know it works because it's synch with my phone.

> pi@raspberrypi:~/genmon/OtherApps $ sudo python3 mopeka_utility.py
> 
> NOTE: This program will look for Mopeka Pro Sensors. The SYNC button must be pressed and held for the discovery process to work.
> 
> Starting Discovery....
> Discovery Stats ReadStats ( Ignored Ad Count: 216, Processed Ad Count: 0, Error Ad Count: 0), Zero Data Ad Count: 0)
> 
> Finished Discovery.  Found 0 new sensor(s):

I will install a new OS.

Thank you!

jgyates, 2023-12-12:
if you have saved your genmon settings external to the pi, you can restore the settings with these instructions: 
https://github.com/jgyates/genmon/wiki/Appendix-M-Tips-and-Tricks-useful-for-Genmon#restoring-settings-from-a-backup

philruel, 2023-12-12:
I installed Raspberry OS 12 and the genmon software.

Now I'm turning around in circles...  How do you get the bleson library installed?

> To use the mopeka pro sensor with genmon the python bleson library (this is installed with the install script) must be installed and your system must have a compatible Bluetooth controller.

> admin@raspberrypi:~/genmon/OtherApps $ sudo python3 mopeka_utility.py
> The required library bleson is not installed.

jgyates, 2023-12-12:
This is an anomaly of the new python requirements in OS12 (a managed python environment. Try this:

 First start out in the genmon directory:

     cd genmon

then enabled the virtual python environment:

     source ./genenv/bin/activate

Then run the mopeka_utility.py program with this command:

     sudo ./genenv/bin/python3 ./OtherApps/mopeka_utility.py

There are pros (less library version conflicst between python apps) and cons (more things to type) of the new managed python environment requirements. This page has more info: https://github.com/jgyates/genmon/wiki/Appendix-S---Working-in-a-Managed-Python-Environment

I will update the warning message in mopeka_utility.py to be more descriptive of this error.

jgyates, 2023-12-12:
I updated mopeka_utility.py to print a the URL listed above if it it can not find the library in a managed environment. I had done this for serialtest.py and modbusdump.py but I forgot to put this in mopeka_utility.py. Sorry, the managed environment stuff is relatively new (couple of months) so there are still a few things like this that pop up that let me know where the holes are in pointing people in the right direction.

philruel, 2023-12-13:
Awesome help!
It works!  Added my Mopeka Pro+.
Will try to restore my backup next.

jgyates, 2023-12-13:
FYI, If you overwrite the file /etc/genmon/genmopeka.conf you will have to add the sensor MAC address again.