jgyates/genmon #989: PowerZone 410 with MQTT integration not outputting all parameters as a /value when JSON for numerics is active.

Evadrzz, 2023-12-11:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Under the add on of MQTT integration I have JSON for numerics turned on, and my PowerZone 410 should be outputting a /value for all parameters that are being output.

### Actual Behavior

Only a few of the values are coming through as a /value, but the majority are do not have a corresponding /value parameter.

### Steps to Reproduce (including precondition)

Connect to a PowerZone 410, activate MQTT integration, turn on JSON for numerics.

### Your Environment

- Generator Model: Generac SD50
- Genmon Logs: Submit via the About page
- Genmon Version: 1.19.0

jgyates, 2023-12-11:
I have not received your log files. You must have outbound email setup and working to submit you log files. Not all values are compatible with a non string value. Can you be more specific regarding the MQTT element you are expecting to be a numeric value? It is possible that more could be exported as an integer, but I would need more info.

jgyates, 2023-12-11:
After looking at the code for the custom controller it may be be exporting data for all integer and float values. I don't have a lot of test data on the Powerzone 410 so have your logs and registers would allow me to test this an make any changes needed to get more data exported via the Numerics option. If you want to send them directly (not via the about page) let me know and I can tell you how to export them and email them outside of genmon.

jgyates, 2023-12-12:
I think I found the issues that you are describing. I checked in an update (version number is still the same) so just update from the About page and see if that corrects your problem. If you get a chance still send your logs as I only have one other Powerzone 410 data set to test with. The logs let me see if things are running as expected and the registers allow me to test against a 410 set of registers in my simulator.

I am going to close this issue but post any follow up status or questions to this thread.

closed 2023-12-12 by jgyates

Evadrzz, 2023-12-20:
I apologize for not responding sooner, as I just checked this today. I did the update, and that solved the issue I was having. Thank you for your response and the fix, I appreciate it. I would be more than happy to send you my log files, if you could explain how to get them to you I will do that as soon as possible.

jgyates, 2023-12-20:
If your outbound email is configured in genmon you just go to the About page in the web interface and click the "Submit Logs" button. If you don't have email setup you can do this:

1) type this address in your browser: http://192.168.11.10:8000/cmd/getlogs

change the above command "http://192.168.11.10:8000" to match your genmon URL. this will download the file "genmon_logs.tar.gz" to your downloads folder on your PC (your broswer may try to block it as insecure but it is just a zip file of log files). 

then  issue this command in your browser: http://192.168.11.14:8000/cmd/support_data_json

Copy and past this info in an email and attached the file genmon_logs.tar.gz to the email and send the email to generatormonitor.software@gmail.com.

Thanks!