jgyates/genmon #990: Adding SmartGen MGC120 controller

llaforest, 2023-12-11:
Hello,

I was wondering if I would be able to add a configuration for SmartGen controller given this modbus communication document.

First step for me is to connect this controller to my Champion 12kW gasoline generator. The controller has a step motor output for my air flap (choke) which is not common in genset controllers because theuy mostly address diesel models and not gasoline models.

Then if all works fine, I might invest some time to have a valid config in genmon so that I can remotely monitor and configure the controller, do monthly startups, etc.

So if someone could check and let me know if the document attached is sufficient to create a valid configuration in the project.
[MGC120_Protocol_en.pdf](https://github.com/jgyates/genmon/files/13640438/MGC120_Protocol_en.pdf)

Thanks for this project! Looks amazing and I would love to manage it remotely without needing the cloud from vendors which is usually horrible and expensive.

jgyates, 2023-12-11:
@llaforest 

yes, the attached document does appear to be what is needed. The next step is to get a dump of the modbus registers. This is described [here](https://github.com/jgyates/genmon/wiki/Appendix-N-Genmon-Supporting-Other-Controller-Types#know-the-modbus-addresses) and [here](https://github.com/jgyates/genmon/wiki/1----Software-Overview#otherappsmodbusdumppy)

In short there are 4 steps:

1) get the register definitions (completed by you)
2) get a dump of the modbus resiters (to be compelted by you)
3) Create a JSON file describing the registers (I can to this based on the data from steps 1 and 2)
4) Test (do be done by you)

Let me know if you have any questions regarding how to get a dump of the registers.

llaforest, 2023-12-11:
Ok it won't be for now, I did not even buy the thing yet. Will buy, install, make sure it works. Now they sell the communication module that goes in the LINK port and outputs rs485. Or maybe I can connect directly in the LINK port and get modbus data stream from there, what is your experience regarding this?

jgyates, 2023-12-11:
I don't have any experience with this controller, however genmon has been used with serial RS-232, serial RS485/RS422 and serial over TCP (wireless and Ethernet). 

The question I have is can you use the SG72A in Linux? I am guessing you can based on the doc as it uses the FTDI driver however additional drivers may be required, but that is a guess. I would ask your dealer about this.

The raspberry pi has a serial port already. It may be compatible with the LINK connector described in this doc:

https://www.smartgen-america.com/upload/iblock/b1e/6tmi62blbyyrn3p7713cyy73lwuibvqu/SMARTGEN-SG72A_V1.0_en-User-manual.pdf

The doc describes it as a TTL level port so this may be abled to be tied directly to the PI, if we knew the pinout of the LINK connector. If not then I guess we would connect the contoler to the SG72A and then connect the SG72A to the pi, which would appear to the pi as a USB serial port (genmon can communicate with USB serial ports)

jgyates, 2023-12-12:
@llaforest 

I am going to close this thread but please continue to post related status or questions regarding the SmartGen to this thread.

closed 2023-12-12 by jgyates