jgyates/genmon #991: Extended Exercise Post Controller Issue?

eBoon123, 2023-12-14:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Using the Enhanced Exercise add-on with Post-Controller selected. Worked perfectly after the normal exercise period, but generator seemed to run an "extra" post controller cycle the following day.

### Actual Behavior

Potential additional Post-Controller exercise period

### Steps to Reproduce (including precondition)

Use Post-Controller Setting

### Screenshot or Pictures relating to the problem (if possible)

See Attached

### Your Environment

- Generator Model: G0070391
- Genmon Logs:  Submitted
- Genmon Version: V1.19.00

Not sure if this is actually a problem with the Extended Exercise or not. Noticed an additional run of the generator in the middle of the night for extended exercise minutes. No actual outage occurred.

/var/log $ cat /var/log/messages
Dec 10 00:10:22 genmon-saf rsyslogd: [origin software="rsyslogd" swVersion="8.2102.0" x-pid="423" x-info="https://www.rsyslog.com"] rsyslogd was HUPed
Dec 10 10:00:52 genmon-saf genmon: Generator Exercising  <-----Normal 5 minute exercise
Dec 10 10:05:57 genmon-saf genmon: Generator Ready
Dec 10 10:06:06 genmon-saf genmon: Generator Running  <------12 minute Extended Exercise
Dec 10 10:18:00 genmon-saf genmon: Generator Ready
Dec 12 03:19:31 genmon-saf genmon: Generator Running  <-----Not sure what is going on here. Seems like about 12 minutes again??
Dec 12 03:30:43 genmon-saf genmon: Generator Ready
Dec 12 03:30:47 genmon-saf genmon: Generator Running
Dec 12 03:31:29 genmon-saf genmon: Generator Ready
Dec 14 08:46:11 genmon-saf genmon: Utility Service is Up
Dec 14 08:46:11 genmon-saf genmon: System Health :  Not receiving data.
Dec 14 08:46:11 genmon-saf genmon: Software Is Up To Date
Dec 14 08:46:11 genmon-saf genmon: Pi Health : OK
Dec 14 08:46:11 genmon-saf genmon: Generator Ready
Dec 14 08:46:15 genmon-saf genmon: System Health : OK

[image]

Thanks for any insights you can provide!

jgyates, 2023-12-15:
That is odd. I checked in a minor tweak to the logic in genexercise.py. It may corerect the problem, but at the moment my best guess is this:

The evolution 2.0 does spit out some garbage data from time to time, [see Known issues number 15 for details](https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues). If this garbage data appears on the right regiters (register 0001) and the right combination of bits, it can make genmon think that the status changed on the controller. This false reading only last for a moment (1-5 seconds) as the next reading of the register will again read the correct value. It is possible that a false condition is picked up by the genexercise addon. It is possible that this just happened to line up with the right register, and bits to make the software think an exercise cycle was active. If that is the case then it will not likely happen soon as this is just random happenstance.

I suggest we try this:

1) One the Advanced Settings page, enable "Ignore Unknown Values" and save your settings at the bottom of the page. This will reduce the chances of the above happening. See if this correct the problem. If it does not then go to step 2.
2) If you notice that this occurs again, then we can add the line 'debug=True' in /etc/genmon/genexercise.conf. This will put more info in the log file. If it happens again after the additional logging is enabled then send your logs as soon as you can after seeing the problem occur.

Let me know how it goes.

jgyates, 2023-12-15:
I am going to close this issue but please update this thread with any follow up status or questions.

closed 2023-12-15 by jgyates