jgyates/genmon #992: [Feature] Current and Latest Version in JSON/MQTT

snakeeyes5, 2023-12-16:
Love the plug-in, been using it for a long time now.

Could you add the running Genmon version and the latest available version to the status_json and MQTT?

I think the running Genmon verison is available in MQTT, but neither are in the JSON.

Thank you

jgyates, 2023-12-16:
The running version is in monitor_json. status_json, maint_json, outage_json and monitor_json are exported  in the MQTT add. The presence of a new version is also in monitor_json (and in MQTT) however it is just "Update Available" with Yes or No as the value. The version of the new version number is not displayed anywhere except the web interface. Let me know if this is sufficient or if you think the new version number would be useful.

I am going to close this for now but feel free to update this thread with questions or feedback.

jgyates, 2023-12-16:
Also, the web interface uses status_json, maint_json, outage_json and monitor_json to display the data so you can look at the web interface for the values / format.

closed 2023-12-16 by jgyates

snakeeyes5, 2023-12-18:
Thank you, learned something new. No changes needed, thanks for the details.