jgyates/genmon #994: my Genpi unit keeps shutting down

grsthegreat, 2023-12-19:
my unit has the latest update. 

i just had to reboot the Genpi unit about 3 hours ago. it than worked ok.

i just checked it again and it is off line. i cannot even find it in Putty.

any ideas why this could be happening?

jgyates, 2023-12-19:
Do you have any USB peripherals connected?

What variant of pi are you using (pi3, pi4, etc)? If it is a pi4 does it have a heat sink?

Can try moving the power cables and see if that causes the power to cut out temporarily? You could have a marginal power cable.

Here is a list of things to look at: https://www.theelectronics.co.in/2023/01/how-to-fix-raspberry-pi-power-issue.html

grsthegreat, 2023-12-19:
i just went and checked again, and its back up on its own. can i use putty to find version.

also, ive been thinking of replacing the SD card for some time as its about 5 years old. can i just copy and paste the card to a new card and reinsert it?

jgyates, 2023-12-19:
To get type type of pi, if you are running the raspberry pi os type this: 

      cat /proc/device-tree/model

No, on copy and paste the contents of the SD card. There are boot sectors that need to be setup at a level lower than the file system and you have to make sure the operating system permissions are setup properly and a copy and past will typically not do that uniformly.

You are better off just saving a back up of genmon and restoring it on a new pi with the latest OS version. The project wiki has info on restoring the genmon settings (see the tips and tricks wiki page)

jgyates, 2023-12-19:
I am going to mark this thread as closed but please continue to post status or questions to this thread.

closed 2023-12-19 by jgyates

grsthegreat, 2023-12-19:
its a rasp Pie 3 model B

ill go study that link. thanks