jgyates/genmon #996: Router replacement

RoyRomeo, 2023-12-23:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Connectivity

### Actual Behavior

Router was replaced by Verizon.  Since I can't puddy in, is there an easy way to update router info for the Genmon or do I have to start from scratch.

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here}

jgyates, 2023-12-23:
@RoyRomeo 

I don't know how your pi is configured for TCP/IP (static or dynamic settings). I am guessing since you changed your router you have a static IP address, but that is a guess. If that is the case then you can setup another PC on the same subnet address group that the pi is on (e.g. 192.168.1.xx) and then log into the pi via ssh. At this point the PC and the PI will be on the same subnet and you can log into the pi form the PC. Once you log in you can change the IP address of the pi to the subnet of your router.

Alternatively, you can take the pi and plug it into a monitor and keyboard and log into the pi from the keyboard. Then change the IP settings to match your new router.

Let me know if you have any questions.

RoyRomeo, 2023-12-23:
Thanks very much for your help/info.

closed 2023-12-23 by jgyates