jgyates/genmon #997: Cummins/ Onan PowerCommand

HGupta2018, 2023-12-24:
I have a big 50kW Cummins diesel generator that uses PowerCommand 2100 behind my detached garage and the PowerCommand [Annunciator](https://cdn.chipkin.com/files/solutions/cummins/AnnunciatorManual.pdf) in the garage.    The GenSet self checks and alarms go off in the detached garage when something isn't working, but because the garage is far from the house I don't hear the alarms.  For example, I had no idea that the battery was dead for weeks.      I would love to have GenMon running and eventually integrated with my HomeAssistant setup, but I don't know much about generators and I don't want to mess it up.   

 How do I get started for getting this up and running?  
[image]
[image]

I don't mind buying additional parts or paying for support/guidance if that helps!

jgyates, 2023-12-24:
Hi @HGupta2018 

The process for supporting a new generator is described here:

https://github.com/jgyates/genmon/wiki/Appendix-N-Genmon-Supporting-Other-Controller-Types#how-to-create-support-for-additional-controllers

Your controller must support modbus communications and you need to have documentation of the modbus registers. If you have met these two requirements then we can use the project modbusdump.py utility to read the modbus registers. With this data and the modbus registers documentation I can create the JSON file needed to support your generator. If you are capable of setting up a raspberry pi, creating a serial cable and running commands on the pi while connected to the controller, I can handle the additions to the project.

Also, I have this document already, but I am not sure if your controller is one that it supports:
[image]

Let me know if you have any questions or you can contact me directly at generatormonitor.software@gmail.com

HGupta2018, 2023-12-24:
Thanks for the quick response!  Doing some further digging, I think that I would need to purchase [a ModLon II Gateway](https://www.cummins.com/sites/default/files/s-1471.pdf)    Of course, the purchasing aspect is the easy part!   Installing it on my existing system is the scary part.   I do have a couple of extra  raspberry pis and have made serial cables before, so I am comfortable there.

jgyates, 2023-12-27:
Let me know if you have any questions going forward. I am going to close this issue for now but feel free to contact me at the email above or in this thread with questions on this topic.

closed 2023-12-27 by jgyates