jgyates/genmon #998: 30kW Generac with Evolution is not communicating with Genmon

Evadrzz, 2023-12-27:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

The generator in question is a Generac 30kW Generator with a liquid cooled 2.2L Diesel engine equipped with an Evolution controller. This Evolution looks different than the air cooled one from the 20kW but seems to have all of the same connections and functions. The Genmon was plugged into the accessory port on the back of the controller. As far as I know, the expected behavior should be normal Genmon operation. Configured to an Evolution controller, and setup for normal operation.

### Actual Behavior

The Genmon is not receiving data, and will not connect to the generator. I tested using a Pi that was functioning on a Generac 20kW air cooled unit, and it did not connect to the 30kW Evolution controiller. 

### Steps to Reproduce (including precondition)

Configure a Genmon for normal Evolution operation, connect to a 30kW liquid cooled Generac with an Evolution controller

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: Generac SD030 2.2L Diesel liquid cooled with Evolution controller
- Genmon Logs: Submit via the About page
- Genmon Version: 1.18

jgyates, 2023-12-27:
@Evadrzz 

A few comments and questions:

1) Are you getting timeout errors on the Monitor page? Are there any other communication errors? More details on the communication failure may shed more light on this issue.
2) You mentioned that you are using genmon version 1.18. I would recommend using the latest version which is currently 1.19.00. If this is a new model it would include support for the new 2.4L model, if it is an older model then it would allow any errors in the log to be more useful as the code line numbers are included in the log.
3) The model you mentioned in SD030. The RD030 is know to work. Are you sure this unit has an evolution controller? Can you take a picture of the controller and the name plate (that should have the model number and date of manufacture). This may help identify the controller and set a base line of if this controller is supported.

jgyates, 2023-12-27:
[This link](https://legacy.genconnect.generac.com/Media/vwDoc.axd?d=992027b5-3d60-41ff-9c6d-682a0c734c85) shows the SD030 has what looks like an H100 controller

liltux, 2023-12-27:
@jgyates Generac did have industrial units with the Evolution controller, specifically the Verizon Telecom units.  If this is one it has different firmware than the commercial RD series and that maybe causing the issue? 🤷🏼‍♂️ They were made specifically for TMobile and not listed.

jgyates, 2023-12-28:
Possibly, we need to see what specific errors are present in the serial stats. Of course we can try other things like a different modbus address or data rate, but if it is just a question of the model register is different then we should not be seeing timeout errors and the UI should show some error regarding an unknown controller type.

jgyates, 2024-01-02:
@Evadrzz 

I am going to close this issue for now but feel free to post any follow up questions or status here. There are a few things we can try, like examining the communication stats and using the modbusdump.py program to try different data rates, assuming the unit is not damaged and can communicate via the serial interface. Let us know via this thread if you want to troubleshoot this.

closed 2024-01-02 by jgyates