jgyates/genmon #999: GenMon Access Over Dial-Up Modem

redchairlabs, 2024-01-02:
I remotely monitor a generator and live some distance from it's location. In almost all cases, a power outage at the location also results in the cable broadband internet dropping out too. Internet service is usualy not re-established until well after the power is restored. The POTS line though, is far more resiliant. It almost never fails. I can establish a dial-up modem connection to the linux server that GenMon is running on and  have a rather functional CLI interface to it.

My question is, is there any way to poll the registers or write a script that will pull basic system data from GenMon? Better, would you have advice as how best to interface with the data that GenMon manages, over a CLI?

I am aware of using cell service but that is the sketchiest of options in my location. As I am not interested in a GUI interface, it's not something I am considering.

My apologies if this was addressed earlier, I wasn't able to find any references in the search of old posts.

skipfire, 2024-01-02:
There are endpoints that give you the JSON and endpoints that can trigger the commands, not exactly CLI, but you could build a local CLI tool that just hits those lightweight endpoints so that the data transfer is minimal.  Doing that you could even have a rich local UI if you wanted one.  The genmon UI already uses those endpoints so it may be lowish effort to have a local flask running the UI and pointing to the existing endpoints, just have to handle CORS.

jgyates, 2024-01-02:
There are a few options when interfacing to genmon.

This wiki page has info on how to access the data:
https://github.com/jgyates/genmon/wiki/Appendix-C--Interfacing-Generator-Monitor-to-External-Applications

One of the methods is using the [ClientInterface.py](https://github.com/jgyates/genmon/wiki/1----Software-Overview#clientinterfacepy) command line program. You can display the data on the CLI with this app. Or you can write your own program that will pull the data. ClientInterface.py is just a wrapper that does exactly this.

Let me know if you have any questions.

redchairlabs, 2024-01-02:
> Let me know if you have any questions.

If I am reading this correctly, while remotely connected to the local server (vial dial-up and CLI), I can make a call using ClientInterface.py using the commands referenced and it will return the respective data in JSON format?

I should predicate all of this with stating that I am an absolute beginner at coding.

jgyates, 2024-01-02:
Yes, ClientInterface.py is a python program so you will have to use it like this:
[image]

If you want to export the data to a file or parse the JSON you would need to redirect the python output (stdout and stderr) to a file. 

probably an easier way would be to use the curl program that is included iwth linux like this:

     curl http://localhost:8000/cmd/status_json

This will output the same data. The ClientInterface.py gets the data direclty from genmon.py. Using curl would go through the local web server (genserv.py) to get the same data.

redchairlabs, 2024-01-03:
Ok, this should do nicely. Thank you for the example, exactly whay i am looking for.

closed 2024-01-03 by redchairlabs